Eomys/pyleecan #2: .xls file for automated code generation?

magnetron, 2019-02-20:
Hello team,

Do you plan to share the .xls spreadsheet that is used as an input for the automated code generator?

PS: Great project, thanks! Looking forward to getting involved.

BonneelP, 2019-02-20:
Hello,

In fact now we use .csv files for the code generation (which is better to track the changes on git). All theses files are available in the folder Generator/ClassesRef. Note that we plan (one day) to move from csv files to a database. Where did you find the mention of the .xls files (to update it ?).

Thank you for your interest in the project ! By the way we are currently working on the coupling with FEMM as the fist Magnetic model and a website generated with sphinx is incoming. Once all that available it will be easier to discuss about how to use and contribute to the project. 

Best regards

magnetron, 2019-02-20:
Great, thanks for clarifying! I came across the reference to the .xls files in the pyleecan architecture description document (rev A): E.g. page 15.

Makes sense to use .csv files to be able to track changes. If I'm not mistaken, using commas to separate rather than semicolons would allow github to display them nicely in tables...

closed 2019-02-20 by magnetron