Eomys/pyleecan #4: categorization of holes

SebGue, 2019-04-02:
Hello Pierre,

in the hole's build_geometry method, hole's labels are "Air", while there is no FEMM air group nor an explicit "Air" material in create_FEMM_materials or an "Air" mesh setting in get_mesh_param.
I would like to add the missing parts, but before I need to know if holes should get their own group, etc. or if they should belong to "Ventilation".

Best Regards,
Sebastian

BonneelP, 2019-04-02:
Hello Sebastian,

I think that the best would be to set the label as "Hole". Then in the function create_FEMM_materials we create the "Air" material only if we find some surfaces with the label "Ventilation" or "Hole". I think that we should differentiate "Ventilation" and "Hole" to be able to set different mesh properties for instance. 

In fact we can be even more generic by defining the actual material of the ventilation or the holes. 99% of the time it will be Air but one day it can be water or something else. In this case, we need to make sure that if the material is not defined, by default, we set Air (to avoid constraining the user to set the material). But this is not the priority I think. 

By the way, I'm working on a SIPMSM validation case then I will work on the prius IPMSM machine. 

Thank you for your help,
Best regards,
Pierre

SebGue, 2019-04-04:
Hello Pierre,

within my buxfixes/LamHole- branch, I added "Holes" to FEMM related parts of pyleecan. You can have a look.
Therefore I also changed the magnets label. Is there a documented default magnet label? What does the letters Rx_Ty_Sz stand for?
I also replaced "label[index] == Something", since this is error prone in my opinion because the magnet label can have different length (for example if Tx exceeds 9).
Is there a reason for renumbering magnets labels within "LamHole.build_geometry()"? 
I think it would also be good idea to replace "magnet_0, magnet_1, ..." with list of magnets (i.e. magnet[idx] as it is in LamSlotMag).

Best regards,
Sebastian

BonneelP, 2019-04-04:
Hello Sebastian,

The Rx_Ty_Sz is the coordinates of the magnet or winding. For instance a Winding_R1_T0_S8 is the winding area of the 8th Slot (in trigo order, 0 is on the X axis) 2nd radial part and 1st tangential part.
We propose to name the surfaces that way to be able to differentiate them. For instance we can image to set a different magnetization to a particular magnet using this. We also use that to set the correct phases for the winding in the correct surfaces (cf create_FEMM_circuit_material)

The renumbering of the labels in LamHole has the same purpose: it enables (one day) to set different property to a particular magnet that can be used to simulate faults or test different material for each slot why not. 

We had a discussion internally about whereas using a list or magnet_X for Holes. Basically we chose the magnet_X convention for the following reasons:
- LamSlotMag uses a list because we don't know in advance how many magnets there are (we can use several magnet to define a pole with the list). For the HoleM50 we know that there are (for instance) only two magnets and it will always be. 
- If we use a list for the HoleMag, it would require the user to set the list with the correct length every single time, when most of the time only the magnet material matters. With the magnet_X convention, the init of the HoleMag automatically creates the correct number of Magnet object.
- With the magnet_X we can easily set magnet_X = None to remove a magnet. With a list it would look like [Magnet1, None, Magnet2] and there can be some index errors (which magnet is at the first index ?). magnet_0 has a docstring that indicates which magnet it is related to.

We are open to debate if you think that we should change any of the pyleecan convention. For the beginning of the project we set the convention that suits us the most but we need to find the one that will be the more coherent for everyone. By the way if you think that some properties of some objects are odd, we can also discuss about how to improve them (in particular, the winding definition is the trickiest one)

Best regards,
Pierre Bonneel

BonneelP, 2019-04-04:
Regarding your commit, I seems Ok. Here are some advices/comments:
- In assign_FEMM_surface, normally you shouldn't need to add the ventilation or Hole check, it should be handled in create_FEMM_materials to be able to find the property in the prop_dict in draw_FEMM (in fact it seems that it is already the case)
- For comp_FEMM_dict, I think that we can add a "get_height_magnet" method to the HoleMX classes.
- In create_FEMM_magnet, I agree with your todo
- In draw_FEMM, I have contacted David Meeker regarding smartmesh command, he will add mi_smartmesh  in the next release (it was a mistake). By waiting we can use callfemm('mi_smartmesh(0)');
- In build_geometry of HoleMX, great initiative to change from "R" to "Rotor" for instance. In will be easier to scan the labels and more coherent. 
- If it is not the case, can you use black before your commits ? (pip install back and then python -m black path/to/pyleecan). It will update all the code to match PEP8 (there are some files that doesn't end with an empty line plot_B_space for instance)

Thank you for your help,
Best regards,
Pierre Bonneel

BonneelP, 2019-04-12:
Closed by pull request #7

closed 2019-04-12 by BonneelP