Eomys/pyleecan #15: Simplification of HoleMx

SebGue, 2019-04-15:
Due to the possibility to set arbitrary magnets of HoleMx-Class to None, the magnets label is not always the same, i.e. "T0" may be magnet_0, magnet_1 or magnet_2 with HoleM51 for example.
This causes the code of related methods and functions to have a lot of conditionals.
To simplify the code it would be more convienent to have all magnets first (with labels "T0" = magnet_0, "T1" = magnet_1, ... all the time) and set the "None" property later, for example with an additional magnetization "None".

BonneelP, 2019-04-15:
Hello Sebastian, 

I think that we can set T0 to always be magnet_0, T1 for magnet_1, T2 for magnet_2... That way in build_geometry we create only the needed surfaces with the correct label (for instance only T2). Then in the FEMM function if we find a magnet label with a T2 we know that we need to set the parameters of magnet_2 (regardless of the HoleMX type since all HoleMX follow the magnet_X convention).

Also regarding your latest pull request, I think that we should also differentiate the labels of magnet for LamSlotMag and for LamHole (to remove the if type(rotor) == LamHole). For instance _Magnet_ and _HoleMagnet_. That way we know if a "TX" means magnet[X] (for LamSlotMag) or magnet_X (for LamHole).

What do you think of these conventions ? 

Best regards,
Pierre

SebGue, 2019-04-17:
Hello Pierre,

I am okay with these conventions and updated my pull request therefore. 
I didn't changed the Hole names, but I think thats not a problem for now.
The only thing to be careful with this new magnet names is the use of something like this: 
if "Hole" in label: ...
One has to exclude "Magnet" explicitly now.

Best regards,
Sebastian

BonneelP, 2019-04-17:
I didn't though about the Hole issue. I think that we should look for "_Hole_" and "_HoleMagnet_" instead of "Hole" and it should solve the problem.

BonneelP, 2019-04-17:
I mean "Hole\_" instead of "Hole". I just commit something related to that. In general I think that we should separate all the label part (HoleMagnet_Rotor_Parallel...) with "_" to be able to use the split method if needed. 
Do you think that we can close this issue or is there something else ?

SebGue, 2019-04-17:
Yes, I think the issue can be closed.

closed 2019-04-17 by SebGue