Eomys/pyleecan #17: Machine GUI

BonneelP, 2019-04-16:
To simplify the machine definition it would be great to have a GUI. Here are the main idea for the GUI:
- We plan to use PyQt5 since the Qt designer simplify a lot the definition of a GUI
- The definition will be done step by step with some check, intermediate output computation and maybe some default meaningful values
- The tooltip of the fields will be automatically added according to the csv class documentation
- For the slot (and any other part that can have several version of the same class) we need to find a way to automatically detect how many slot widget are available to simplify the process to add a new slot to the GUI for instance. 
- The machine GUI need to enable to select the materials from a library
- The GUI will load and save the machine json file (with the save/load function of the machine object), the simulation will then be done in a separated python script (by waiting the simulation GUI)

BonneelP, 2019-12-11:
The machine GUI is finally available (a bit early for Christmas) ! This GUI enables to create a machine from scratch and contains all the schematics to ease the design. This GUI is based on the MANATEE one, so it may still have some MANATEE related code/comments that pass though my code review. If you find some, please notify me and I will update it.
In particular, this GUI use some link to the eomys website for the extended documentation. My next step will be to update the pyleecan website to be able to update these links. 

For now I let this issue open to get the feedback on the GUI.

SebGue, 2019-12-12:
Hello Pierre,

thank you very much for this commit.
As allways I've got something to complain. I hope you don't mind. :-)

I think it would be good idea to add pyleecan to the search path temporarily. Otherwise no one is able to start it "out of the box".
```
from os.path import abspath, dirname, join, realpath, split
from sys import argv, exit, path

path.append(split(dirname(realpath(__file__)))[0])
```
Maybe you could also add a list of packages needed to the readme, so everyone interessted can start quite easily with pyleecan.

Best regards,
Sebastian

Superomeg4, 2019-12-12:
Hello everyone,

About the list of package, a good practice is to have a specific "requirements.txt" file which contains all necessary related to software/packages  and their version number.

Best Regards,

Raphaël

BonneelP, 2019-12-12:
Hello everyone,
I think that the point of open source is to get feedback on the code. So of course I don't mind and I'm always happy to find new ways to improve the code :) 

In fact I have an equivalent code at the beginning of my main_generator and run_GUI script. I just didn't commit it to avoid modifying the user python path. But now that you mention it, I realize that it is better to have it on Github so I will commit it.

As I said, the next step for me is the documentation to enable anyone to use pyleecan easily. The requirements are indeed a good place to start.

Best regards,
Pierre

SebGue, 2019-12-16:
Hello Pierre,

I had a look at the GUI and since it is the first release, there are still some things to do.
What do you think, should we collect all bugs, feature requests, ... here or will it be reasonable to start a new issue every time? 
Maybe you could also add subtasks (see http://sassyhtml.com/snippets/workflow/creating-github-subtasks-in-issues/) to your first comment?

Best regards,
Sebastian

BonneelP, 2019-12-16:
Hello Sebastian,

My idea was to gather the discussion around the release of the Machine GUI here. It was more about small thing like script renaming, wrong data naming, remaining code from MANATEE... Once the release of the GUI "validated" we would have closed this issue and carry on with new issues. But if you have already some feature requests feel free to open some new issues. 

Best regards,
Pierre

SebGue, 2019-12-17:
Hello Pierre,

I know some of the following things are worth new issues, I first list them here and we can decide.

- axial vent. duct preview window is inaccessible as long as axial vent. duct input dialog is opened
- slot pitch calculation result should have deicmal digits
- size of right side input panels should be fixed since it "jumps" with Drsh result length for instance
- switching between some inputs (i.e. machine type, slot type, shaft option, ...) discards all previous input values -> these values should be stored for reuse instead
- picture size or at least aspect ratio should be fixed
- output values should update immediately when spinner is used
- the last step "save" button should be "save & quit"
- default load and save dir should be the same
- recent dir should be remembered
- rotor preview should be one or two poles only
- input of only one magnet for HoleMagnet with multiple magnets possible
- input of BH curve not possible yet
- it would be nice to have a live preview of the geometry inputs instead of pictures
- every parameter should have a short description in addition to the full descrition (as a tooltip), i.e. Ext. Lamimation Diameter Rext [INUPT] mm for instance
- there could be a complete machine preview button next to load & save in addition to the singe part previews
- there should be a winding pattern editior
- "desc" of new materials is defaulted to "COPPER WINDING"

Is there a reason why the inputs have to be step by step?
In general we should check all class default values to make sense and won't produce unexpected behaviour.

Since there are a lot of hand coded files, what do you think about an object attribute editor instead?
I don't actually know if that will work as expected, but it would track user class modifications easily without the need to adapt the GUI.
A quick search got me the following link which seems promising. http://trevorius.com/scrapbook/python/attribute-editor-in-pyqt/ 

Best regards,
Sebastian

BonneelP, 2019-12-17:
Hello Sebastian, 

Thank you very much for your feedback. This is really helpful to improve the GUI.

To answer/discuss your questions in order:

> - axial vent. duct preview window is inaccessible as long as axial vent. duct input dialog is opened

=> The issue there is that we wanted to validate the ventilations before updating the actual object (which is a different behaviour from the rest of the GUI). So we "block" the "normal" GUI while the axial vent GUI is opened. But there were lot's of changes since the creation of the GUI so maybe it is possible to update this behaviour. 

> - slot pitch calculation result should have deicmal digits

=> Corrected is commit 0c6921e7fe7a6f07bfd1dea4f15b3644669a9b5c

> - size of right side input panels should be fixed since it "jumps" with Drsh result length for instance

=> We agree, we just need to find a correct size for all screens. If possible this should not be set in the ui files but rather as a general parameter somewhere in the code. 

> - switching between some inputs (i.e. machine type, slot type, shaft option, ...) discards all previous input values -> these values should be stored for reuse instead

=> It is already the case for the slots (SWSlot with previous_slot). I'm not convince yet that it would be useful for machine type, but Ok for shaft and other things. 

> - picture size or at least aspect ratio should be fixed

=> We agree, we need to find a correct size for all screen (but most images, can be replaced by a matplotlib widget)

> - output values should update immediately when spinner is used

=> In the GUI we mostly use the "editingFinished" signal because we wanted to update the value only when meaningful. But for some widget "valueChanged" can be more appropriate. 

> - the last step "save" button should be "save & quit"

=> Done in commit 524d443d9d377b91b16d8a9f1cdaabc8834857b2

> - default load and save dir should be the same

=> For me it was already the case ? Where did you find this issue ?

> - recent dir should be remembered

=> Done in commit 2615f075e4a6a0cd6034756686dbf2926276f7fd

> - rotor preview should be one or two poles only

=> Then we can do the same with the stator ? 

> - input of only one magnet for HoleMagnet with multiple magnets possible

=> The idea of the GUI was to address the most used configuration but it is something that can be updated easily (and the possibility to remove some magnet). Same for several magnet for a pole 

> - input of BH curve not possible yet

=> We can open an issue on that or discuss it on #55 

> - it would be nice to have a live preview of the geometry inputs instead of pictures

=> Can be done with matplotlib but it require to have meaningful default parameter for everything. We can open an issue on this one. 

> - every parameter should have a short description in addition to the full descrition (as a tooltip), i.e. Ext. Lamimation Diameter Rext [INUPT] mm for instance

=> For me we can define a "verbose name" in the csv files and then a GUI option to switch from the "normal" name (Zs) to the verbose name (Slot number) but we may have some issue with inconstant text length. 

> - there could be a complete machine preview button next to load & save in addition to the singe part previews

=> Agree, we just need to check if everything is correctly set in the machine (which should be already possible)

> - there should be a winding pattern editior

=> definitely yes. We have few ideas on this one, we can open an issue

> - "desc" of new materials is defaulted to "COPPER WINDING"

=> "desc" default is in fact "Lamination M400-50A" (from the csv), but in the GUI we create Material by copping existing materials (and their descriptions). I think that we just need to add a widget to display/edit the "desc".

> Is there a reason why the inputs have to be step by step?

=> We made the GUI step by step because some parameters makes sense only if some other are set. For instance setting the slot before the lamination dimensions doesn't make sense. Changing the machine type change the lamination type and everything that need to be set. 

> In general we should check all class default values to make sense and won't produce unexpected behaviour.

=> At the beginning the idea of the GUI was "no default value" (that's why the "_set_None" method was introduced). But if we go to dynamic geometry preview (which I think we should) we will have to change this philosophy. 

> Since there are a lot of hand coded files, what do you think about an object attribute editor instead? I don't actually know if that will work as expected, but it would track user class modifications easily without the need to adapt the GUI.

Why not. I'm just not sure how it would work with the current GUI. The structure of the GUI enables to easily add new slots, magnets, conductor... which is mostly the new objects that contributor should add. But as we have the csv doc, I think that we should be able to generate automatically such GUI. So it is worth investigating. 

Best regards,
Pierre

SebGue, 2019-12-17:
Hello Pierre,

I've edited my comment to have a task list and try to update it until all points are finalized.

> > * size of right side input panels should be fixed since it "jumps" with Drsh result length for instance
> 
> => We agree, we just need to find a correct size for all screens. If possible this should not be set in the ui files but rather as a general parameter somewhere in the code.

=> I think, I saw a limitation of digits for LineEdit-Input somewhere, so we could use monospaced font and fixed input panel size. 

> => It is already the case for the slots (SWSlot with previous_slot). I'm not convince yet that it would be useful for machine type, but Ok for shaft and other things.

=> It will also be usefull for machine type. For instance you could keep/load an "old" stator for a different type of machine. And it will also be less frustating if one changes machine type by accident.

> > * default load and save dir should be the same
> 
> => For me it was already the case ? Where did you find this issue ?

=> I couldn't reproduce this, so maybe I was mistaken.

> > * rotor preview should be one or two poles only
> 
> => Then we can do the same with the stator ?

=> Another possibility could be to have both preview options, full part and only sym. part.

> > * it would be nice to have a live preview of the geometry inputs instead of pictures
> 
> => Can be done with matplotlib but it require to have meaningful default parameter for everything. We can open an issue on this one.

=> I will open a new issue.

> => For me we can define a "verbose name" in the csv files and then a GUI option to switch from the "normal" name (Zs) to the verbose name (Slot number) but we may have some issue with inconstant [...]

=> I think there should be both verbose and normal name with max. number of allowed characters and monospaced font in the GUI.

> > * there should be a winding pattern editior
> 
> => definitely yes. We have few ideas on this one, we can open an issue

=> I will also open a new issue on that.

> Why not. I'm just not sure how it would work with the current GUI. The structure of the GUI enables to easily add new slots, magnets, conductor... which is mostly the new objects that contributor [...]

=> I will open an issue on that as well.

Best regards,
Sebastian

BonneelP, 2019-12-18:
Hello Sebastian,

I've removed the closed/new issue topics for more clarity

> > > * size of right side input panels should be fixed since it "jumps" with Drsh result length for instance
>>>
> > => We agree, we just need to find a correct size for all screens. If possible this should not be set in the ui files but rather as a general parameter somewhere in the code.
>>
> => I think, I saw a limitation of digits for LineEdit-Input somewhere, so we could use monospaced font and fixed input panel size.
>
> => I think there should be both verbose and normal name with max. number of allowed characters and monospaced font in the GUI.
>
=> That would be great. We should also fix the size of the label (according to the maximum verbose name length) before the FloatEdit/Spinbox to be consistent in the complete GUI. The default size of both, the fond, and the number of digit should be defined in the __init__ file of the GUI folder (for instance) and applied in the "Gen_" files for instance. 

> => It will also be usefull for machine type. For instance you could keep/load an "old" stator for a different type of machine. And it will also be less frustating if one changes machine type by [...]
=> So when changing the machine type we can do two things: store the current machine as we currently do with the slot, and if the GUI doesn't have any machine of the new type, we create one by using as much parameter from the current machine as possible. In particular we should always be able to reuse the stator since all machine have LamSlotWind for Stator (for now). 

> > > * rotor preview should be one or two poles only
>>>
> > => Then we can do the same with the stator ?
>>
> => Another possibility could be to have both preview options, full part and only sym. part.
>
=> Nice idea. Do you think that it would be possible to design a "Plot" button that enables to select sym or full without taking too much place ? 

Best regards,
Pierre

SebGue, 2019-12-18:
Hello Pierre,

regarding "Plot" button (and some other parts of the GUI), I would go for icons. Maybe I will "design" some with the standard plot functions in my spare time.
An alternative could also be a tabbed output with full and sym.

Best regards, 
Sebastian