Eomys/pyleecan #22: Improve assign_FEMM_surface for HoleMagnet

BonneelP, 2019-04-18:
There are two things to improve in assign_FEMM_surface regarding the HoleMagnet:
- For now the magnet are always set according to hole[0] but in the case of several holes it would require to read the label to select hole[ii]. The convention could be that in build_geometry of LamHole we update the hole label to have RX with X the position of the hole in the hole list of the lamination. 
- We need to add a method comp_angle_magnetization to compute the angle of the magnet for parallel magnetization. It would take as argument a bool is_north and return a tuple corresponding to the magnetization of each magnet (None if the magnet is not there).

jnddong, 2019-05-23:
For the second point: is it necessary to add an option to define your own magnetization angle?

BonneelP, 2019-05-23:
Hello,

Regarding the second point the idea is to find a convenient way to get the correct angle for parallel magnetization because it depend on the Hole orientation/geometry. For instance for Hole 50:
[image]
it would be the angle to have the magnetization like the H2 arrows. We already compute that in Functions/FEMM/assign_FEMM_surface (line 63 to 92). The idea is to provide a method in the HoleMX classes to get directly the correct angles to replace the:
`if type(lam.hole[0]) == HoleM50
....
if type(lam.hole[0]) == HoleM51
....`
by something like:
`angle_list = lam.hole[0].comp_angle_magnetization()`
Which is more coherent with the object oriented approach and easier to adapt when len(lam.hole)>1.

So for parallel magnetization there is no need to add a new parameter (otherwise the user will have to define this value to set the correct magnetization). However, once we have the comp_angle_magnetization method, it could be interesting to add a parameter in the Magnet object to enforce the returned values. That way it would be possible to enforce easily the magnetization of any magnet in any direction for research purpose. 

Best regards,
Pierre

BonneelP, 2019-05-27:
As a side note, I'm only referring to parallel magnetization. So maybe the method name should be comp_angle_parallel_mag(is_north=True). 
For now we only have parallel and polar magnetization but there will be some other type and it would also be interesting to adapt assign_FEMM_surface to avoid the 
```
if "Parallel" in label
...
if "Radial" in label
...
```

BonneelP, 2020-11-05:
Hello all,

I'm digging the issue up because now the fix (if type(hole) is HoleM50) can't work any more. We use now HoleUD that have a dict of magnets defined as SurfLine objects. This new kind of Hole is very handy when importing a topology from a DXF (cf webinar 6th November 2020). So for these kind of slot we can either add a meta-data to indicate the axis (I prefer we don't but it can be a quick fix by waiting). Either we need to find a clever way of finding the orientation of any surface. Maybe we can start by assuming that all magnets are "mostly rectangular" and if not use the user-defined value.

So maybe we can introduce comp_angle_parallel_mag(magnet_id, is_north=True) as a method of Hole. For HoleMX, we can define it to compute the proper value for each Hole. Then for HoleUD we can add a property magnetization_dict to store the proper angle for each magnet. 

Does any one knows how we could compute/define this angle for any surface ?
Best regards,
Pierre

BonneelP, 2020-11-08:
Hello all,

To solve this issue here is my proposal: in build_geometry of Hole daughters for the magnet / active surface, we add a label "bottom_line". Then we add a method in Hole get_magnetization_dict(is_north=True) that would return a dictionary ["magnet_X"] = alpha (rad). This method would call build_geometry, then for all active surfaces get the bottom line, get the middle of the line and find the direction of the normal (by looking for the ref point). 

If latter we have issue with particular shapes, get_magentization_dict can either be defined in the corresponding Hole daughter or a property magnetization_dict can be directly added. 

Finally, for HoleUD one will have to provide the bottom line. In particular we will have to adapt the DXF_Hole GUI to select this line. 

What do you think about this solution ? 

Best regards,
Pierre

BonneelP, 2021-02-12:
Hello all,

I will now focus on this feature to be able to import a HoleUD from DXF and simulate it in FEMM. At the end, I won't use a label for the lines but directly the method get_magnetization_dict(is_north=True) that would return a dictionary ["magnet_X"] = alpha (rad) and a "magnetization_dict_enforced" property for Hole. Then the labels can be used for something else (like Elmer) and we have the possibility to enforce the magnetization_dict for all Hole (to study mismatching magnetization direction for instance). In the get_magnetization_dict of the HoleMX I will check that magnetization_dict_enforced is not None and compute it. For Hole without magnet the dictionary will just be empty. 

I will also take the opportunity to introduce a new method for Li+ne comp_normal(return_type=0) that would return
- return_type = 0: (Zbegin+Zend)/2 (even for arc) and the normal point "on top" (begin=O and end on Ox)
- return_type = 1: Z2 - Z1 (with Z1 and Z2 the points from return_type 0)
- return_type = 2: the angle of the vector of return_type 1 according to Ox

I may also add an arrow on the corresponding plot so I can check the computed angle (with an option to active/deactivate it)

Best regards,
Pierre

BonneelP, 2021-03-15:
This issue was solved in PR #323. Now we use the result from comp_magnetization_dict and the surface label to know which direction to use for each magnet.

closed 2021-03-15 by BonneelP