Eomys/pyleecan #31: Electrical module: first workflow and equivalent circuit

BonneelP, 2019-06-17:
The first module of a full PYLEECAN simulation is the electrical one. [The architecture proposal document details some idea on this module (p37)](https://eomys.com/produits/article/pyleecan-open-source-project-for-electrical-machine-design?lang=en)

To summarize, the module would be organized around 3 objects with several versions:
- Electrical: To select the workflow and the Operating Point
- Drive: Generate the power signal of the machine
- EEC: Compute the parameters of the Equivalent Electrical Circuit (EEC)

To begin with this module, the idea is to choose a particular simple case to create the first version of each object and set all their interactions. Then other versions of these objects will be easier to develop to add new hypothesis or workflow.
We suggest to begin with the following case:
A SIPMSM at a single speed, powered by a sinusoidal tension drive with the following EEC:
 
[image]

P180 from “Advanced Electrical Drives, analysis, modeling, control” Rik de doncker, Duco W.J. Pulle, André Veltman, Springer edition

The advantage of this EEC is that we will be able to use it for surface, inset and interior permanent magnet synchronous machine. 

To develop this module several independent steps are required:
- Create a DriveImport object that would contains an Import object and have a gen_signal() method. For the previous case, we can use a ImportGenMatrixSin to define the signal.
- Add a comp_resistance_wind to the LamWind object
- Create an IndMagANL object (included in the EEC object) to compute the magnetization inductance along d and q axis analytically (for instance)
- Create a FluxLinkANL object (included in the EEC object) to compute the flux linkage analytically (for instance)
- Provide some validation cases to check that the computation of each EEC parameter is correct

Note that inductance and flux linkage has their own object so that it will be possible to easily add new model to compute them. The winding resistance is more straight forward and can be a “machine” method. 

Once all this available, it would be interesting to compute the current (as a function of time) without assuming that the tension is a sinusoidal to be able to change the Drive for other cases.

awindhorn, 2019-06-26:
There should be no problem computing current for any waveform by converting it to D-Q frame or using harmonic series of sinusoidal waveforms (for linear approximations).

The winding resistance would be a property of a Winding, which would incorporate Coil(s) and Hookup objects, and the Coil would incorporate a WireBundle, containing Wires, and be dependent on the Slot object of the Lamination.  These would be common to several types of machines.

Eomys, 2019-07-04:
Hello,

For now we have a LamSlotWind object for Lamination with slots containing winding (which can be used for several type of machine). LamSlotWind contains two other objects: Slot and Winding. The Winding object define everything from the pattern (which phase in which slot) and the circuit (Connexion type, number of parallel circuits...). The pattern is selected by the object type (WindingCW1L is for single layer non-overlapping 'concentrated' tooth winding 'alternate teeth wound') that enables to adapt the methods to the pattern. Note that to generate the pattern we need to know the number of slot (Zs) which is defined in the Slot object. We can access Zs though [the parent property](https://github.com/Eomys/pyleecan/blob/master/Methods/Machine/WindingCW1L/comp_connection_mat.py).

The Winding object contains also a Conductor object with several version  CondType11 for preformed rectangular conductor, CondType12 for random winding round wires.
 [image]
[image]

All the winding convention and definition are available p31 from the [architecture document](https://eomys.com/IMG/pdf/pyleecan_architecture.pdf). 

To summarize, we gather everything in two objects kind Winding and Conductor. Do you see something to improve in these definitions ? In particular we are interested to discuss about how to compute the end winding length according to the winding pattern. 
Best regards,
Pierre

BonneelP, 2019-07-04:
Hello,

For now we have a LamSlotWind object for Lamination with slots containing winding (which can be used for several type of machine). LamSlotWind contains two other objects: Slot and Winding. The Winding object define everything from the pattern (which phase in which slot) and the circuit (Connexion type, number of parallel circuits...). The pattern is selected by the object type (WindingCW1L is for single layer non-overlapping 'concentrated' tooth winding 'alternate teeth wound') that enables to adapt the methods to the pattern. Note that to generate the pattern we need to know the number of slot (Zs) which is defined in the Slot object. We can access Zs though [the parent property](https://github.com/Eomys/pyleecan/blob/master/Methods/Machine/WindingCW1L/comp_connection_mat.py).

The Winding object contains also a Conductor object with several version  CondType11 for preformed rectangular conductor, CondType12 for random winding round wires.
[image]
[image]

All the winding convention and definition are available p31 from the [architecture document](https://eomys.com/IMG/pdf/pyleecan_architecture.pdf). 

To summarize, we gather everything in two objects kind: Winding and Conductor. Do you see something to improve in these definitions ? In particular we are interested to discuss about how to compute the end winding length according to the winding pattern.
Best regards,
Pierre

BonneelP, 2020-05-26:
Hello Sebastian,

We are currently working a lot on the electrical module and I would like to talk about DQ axis computation. For the computation of the Flux linkage and inductance of the machine we have introduced some new methods "comp_angle_d_axis" and "comp_angle_q_axis" on some Lamination daughters. For the LamSlotWind we compute the d axis angle according to the maximum of the unit mmf fft. For InputCurrentDQ you introduced the method "comp_initial_angle" for Synchronous machines to compute the "angle between the d-axis and the alpha-axis of the machine". For the stator angle you used the method "comp_phasor_angle". Here is the result of the corresponding methods:
[image]
I think that we should officially set the convention for DQ axis to be able to unify our methods by updating "comp_initial_angle" to use the "comp_angle_d_axis" methods. Can you check if our methods fit with what you need for InputCurrentDQ ? I saw at least one difference between our two methods: We assume that there is a tooth on X axis to be consistent with the coupling with FEMM. Are you Ok with this convention ? 
Another question: would you mind if we rename "comp_initial_angle" into "comp_angle_d_axis_diff" (or anything else) ?

We have also worked on the file "Functions/Electrical/coordinate_transformation" to extend some of the transformation to N phase.

Best regards,
Pierre

SebGue, 2020-06-02:
Hello Pierre,

it's a little confusing to me that all laminations have dq-axis (methods), since I'm used to have alpha-beta for stator and dq for rotor. But since it's not defined if a lamination is stationary or rotating, I'm okay with your suggestions.
For the new name of "comp_initial_angle" I think there should be something like "initial" or "model_offset" in it for some reason.
The assert error seems to result due to the different possible (mechanical) angles depending on the number of poles, i.e. for your example above is 90° since it's a 4 pole pair machine.

(I haven't completely understood your code yet, especially the indexed search for the final angle.)

Best regards,
Sebastian

BonneelP, 2020-06-02:
Hello Sebastian,

Thank you for your answer. I think that we can as well introduce a comp_angle_axis_alpha for the lamination. To know if a lamination is stationary or rotating we can use the property "is_stator". We defined the D axis for the stator because even if the lamination is stationary, the flux is rotating.

To respect pyleecan naming convention and according to your proposition I would suggest "comp_angle_offset_initial".

The purpose of pyleecan is to enable to compare several model to simulate the same physics, so I think that we should enable the possibility to choose the convention for these angles. What do you think about introducing a parameter for "comp_angle_offset_initial" to select the convention (0 default: alpha stator - D rotor, 1: D stator - D rotor) ? We can then modify the existing code to use the corresponding axis methods. Since there can be many models / conventions, it becomes necessary to indicate in the docstring from which reference it comes from. If there is no reference we should describe why we are using this convention.

We would also need to introduce a property "offset_initial_convention" (0 or 1) in the Simulation object. InputCurrentDQ, Electrical and Magnetic module would get the value from the parent or enforce a particular convention (with the proper docstring/check/warning) if the model is strongly linked to a particular one. It would be better to put the convention directly in the InputCurrentDQ, Electrical and Magnetic objects but as this convention needs to be the same for each of these object, Simulation can centralize the definition.

Regarding our comp_angle_axis_d for LamSlotWind, we compute the 1D fft of the MMF over space along the airgap, then we find the angle for which fundamental of stator MMF is maximum.

Best regards,
Pierre

SebGue, 2020-06-08:
Hello Pierre,

I think the "comp_initial_angle" method is only used with InputCurrentDQ-Class.
In my opinion the most simple approch would be to remove/replace this method. So there is no need to have different conventions and further documentation.
Only the dq- and alpha/beta-axis angles as well as the rotor angle need a documented convention, that is needed anyway. The convention could be, that the respective axis angle is always the angle between the axis and the x-axis.

What do you think about that? Would that also satisfy your needs and wouldn't that be a much cleaner solution?

Best regards,
Sebastian

BonneelP, 2020-06-10:
Hello Sebastian,

Thank you for your answer. In this case I will rename comp_initial_angle to comp_angle_offset_initial, change it to the D stator - D rotor conventions and update the website documentation to address these conventions. I have added recently some [Geometrical conventions](https://pyleecan.org/coding.convention.html) for slot and winding on the website, I will add a new article about the axis. I will also say that "axis angle is always the angle between the axis and the x-axis".

Best regards,
Pierre

BonneelP, 2020-06-23:
Hello Sebastian,

I have pushed the code corresponding to my latest message on the branch "offset_angle". The corresponding documentation is available on https://github.com/Eomys/pyleecan-doc/blob/master/axis.convention.rst

Do you agree with my modifications and definitions? 

Best regards,
Pierre

SebGue, 2020-06-24:
Hello Pierre,

I agree with your modification. I also did some small additions. :-)
We also should do some further detailing, i.e. add the position of the first phase or the first slot and the underlaying assumptions regarding the mmf calculation (I think there are some, e.g. even distributed slots, ...).

Best regards, Sebastian

BonneelP, 2020-07-08:
Hello Sebastian, 

I have updated the documentation according to your last comment: 
https://github.com/Eomys/pyleecan-doc/blob/master/axis.convention.rst
https://github.com/Eomys/pyleecan-doc/blob/master/winding.convention.rst

Do you think that further details should be added or can I merge the branch offset_angle ?

Best regards,
Pierre

SebGue, 2020-07-09:
Hello Pierre,

I think you should merge that branch. If someone need some further details, there is some realy good starting point to contribute with your documentation.

Best regards, Sebastian

BonneelP, 2020-08-05:
Hello everyone,

We were working a lot lately on the Electrical module and the definition of the Operating Point (OP). Our aim was to be able to define multi-simulation on several OP (and draw the corresponding plot). Here are few realization / reorganization / architecture optimization that impose themselves when we were experimenting:
- Each Simu1 object correspond to a single OP and VarLoad object enables to define "variable speed" simulation (or multi-simulation) on several OP
- The purpose of the Electrical module is to compute the OP that matches the Equivalent Electrical Circuit (with resistance, inductors, voltage sources etc) solving constraints
- Then the object to enforce the OP already exist: InputCurrent
- The OP is defined in OutElec with: a fixed speed N0, Id and Iq (even for asynchronous machines) and a reference theoretical torque (the calculated torque is still defined in OutMag). We provided a method to set Id, Iq in InCurrent according to I0, Phi0
- InCurrent and InCurrentDQ have been merged: By defining Id, Iq and not Is, get_Is will generate the current corresponding to Id Iq.
- VarLoadCurrent was defined to generate a multi-simulation on several OP
- Now that the speed is a known float, the time vector can be computed by defining the number of point and the number of rotation. 
The corresponding code is available at: https://github.com/EOMYS-Public/pyleecan/tree/elec
What do you think of these modifications ? 

We are currently validating the electrical module and finishing few modifications on the Optimization module and then we should finally be able to do the first release of pyleecan (Hopefully by the end of the month). 

Best regards,
Pierre

BonneelP, 2020-09-28:
The first electrical module was validated and included with PR #141. It match the description of the previous post.
I close this issue, any further work on the Electrical module should have its own issue.

closed 2020-09-28 by BonneelP