Eomys/pyleecan #32: Coil geometry

awindhorn, 2019-06-26:
Do you have a coil geometry calculator defined for lap form coils yet?  (I couldn't see one in the code, but I haven't absorbed the program structure yet.)  If not, I may be able to assist with that.  Concentric and other oddball windings maybe not so much.

Regards,
Allen

BonneelP, 2019-07-04:
Hello,

If you are referring to the end winding length computation we don’t have that yet and it is important to add a model. 

To do that we can for instance define a method Lew = comp_length_end(Zs) to compute the length of an end winding from one slot to another with:
-	Zs the number of slot
That way the winding complete length would be (0.5*2*Zs*Lew+Zs*L)*Nlay*Ntcoil with: 
-	L the lamination length
-	Nlay the number of winding layer
-	Ntcoil the number of coil turn.
This method would be an abstract method of Winding, then each Winding daughter will then have a comp_length_end method to adapt the computation according the winding pattern. 
We can also add some properties to the Winding object to define the end winding shape when it is meaningful. For instance: 
[image]
[image]

What do you find of this convential proposal ? 

Best regards,
Pierre

SebGue, 2021-03-15:
Hello Pierre,

despite the proposal of one 'comp_length_end' method per winding class, I would go for a dedicated EndWinding class.
Then Winding class will have an end_winding property of that new class. 
The EndWinding classes will have a 'comp_length_endwinding' method that will return the respective length.
Further every Winding class can have its own suitable default for end_winding prop.
The more or less abstract class EndWinding will return 0 endwinding length, which will result in the same winding lenght as the current implementation does. 

If you like that idea, I will PR this feature ( I already have it in my internals). So we could also close this old issue :-)

Best regards, Sebastian

BonneelP, 2021-05-17:
#378 introduces a new object for EndWinding computation. I think we can close this issue, we can create a new issue if we need to extend the current implementation.

closed 2021-05-17 by BonneelP