Eomys/pyleecan #36: Rotor Notches / Rotor Surface Shape

SebGue, 2019-08-23:
Hello Pierre,

do you plan to include your MANATEE's rotor notches functionality into pyleecan in the near future?

Otherwise I would try to include some similiar functionality, i.e. a notch or (rotor) bore surface shaper class. That class can be a property of all concentric bore laminations (e.g. LamHole-type) similar to LamHole.holes list.
The class will then be used (if not None) to shape the bore surface in a substractive manner to get rectangular notches or other shapes. Since it will be substractive one could also combine multiple shapes.

I think that will be a good addition to pyleecan. What do you think?

Best regards,
Sebastian

BonneelP, 2019-08-23:
Hello Sebastian,

Indeed notches are in the development plan. In the architecture, we designed the build_geometry methods of the Lamination the notches in mind. I would more refer as "Lamination Notches" rather than "Rotor Notches" since we use the same object for the stator and rotor. 

For LamSlot object, the idea of the build_geometry method is to get the slot lines with self.slot.build_geometry then add the "bore lines" with self. get_bore_line. For now this method only return the tooth arc, the idea would be to modified it to include the notches (if there is any).

Then there are several interesting options to keep in mind when designing the notches related method:
- We want to be able to have several notches on a same tooth
- We want to be able to have some teeth with notches and other without
- We want to be able to have several shape for the notches (and maybe to combine several kind on the same machine). 
- There can be some issue with FEMM when using symmetries (since we cut the lamination in the middle of a tooth) (Can be solve with issue #24)
- Some slot (like [slot 17](https://eomys.com/produits/manatee/howtos/article/how-to-set-up-slot-type-17)) have flat teeth. For now in pyleecan we handle them by assuming that there is no bore line to add. 

For me a first idea could be to use the Slot objects to define the shape of the notches. That way we will be able to reuse all the work of the Slot (comp_surface, comp_angle_opening...) and the shape that we add for the notches can be used for the slot. we already have SlotMFlat that define what could be the rectangular notch (with magnet = list( ) ). The idea would be to use a build_geometry method to draw the lines of the notches. 

Then there is the lamination shape that can be uneven (in particular for LamHoles)
[image]
With the question: Would it be possible to combine an uneven lamination and notches ? 

Note that for now the build geometry of the LamHoles directly creates a Circle (or an arc with symmetry). It would be interesting to modify it so that it also use the "get_bore_line" method. That way we would use the same method for all the Lamination object. As a side note, at the end it would be great to have the Lamination related surface (including notches and uneven shape) created in the build_geometry of "Lamination" and then build_geometry of "LamHoles" should only add the holes. 

Here are lot's of though on how to include notches in pyleecan. I tried to be as exhaustive as possible but I think that we can start working with something simple (without considering the FEMM symmetry or the uneven lamination). I would be happy to discuss with you about the best way to implement this feature. How would you organize a "Notch" object ? 

Best regards,
Pierre

SebGue, 2019-08-27:
Hello Pierre,

indeed this is a lot of thought and you are right. Notches should be part of Lamination.
With all the points you made, I think we are more or less in common.
The first thing we need in my opinion, is a (list) property within the Lamination object similar to "axial_ven" that will contain the notches. 
What we need in addition to the information given by the slot object is the angular placement of the notch, i.e. alpha0.
I think there are three possible implementations:
- We have alpha0 and the Slot Object object within the notch property. 
- We have a new Notch class that is derived from Slot class to include alpha0. (Not so good since there are a lot of new classes only because of one parameter.)
- We could add alpha0 to the desired Slot classes. Classes without alpha0 may be told to be incompatible by a warning.

Regarding uneven lamination and notches, I think that would be interesting too. Therefore maybe the first of the above possibilities could be interesting, since we could have additional control parameters. For example if the notch is placed on the uneven structure or on the virtual bore circle. Also a angle_delta parameter could be interesting to have groups of notches (e.g. for multiple damping bars on slot_60 rotors ) defined at once.

At first I will start with the "get_bore_line" method, which is needed in any case.

What do you think on the proposals above? Do you have other implementations in mind?

Best regards,
Sebastian

BonneelP, 2019-08-28:
Hello,

I agree with your idea to do something similar to axial_vent and having the position of the notch as a parameter. There is something I forgot: the "parent" constraint for the Slot. Therefore if we put the Slot object within a Notch object the parent will change and build_geometry won't work. 

Because of that I think that the simplest solution would be to add a "notches_list" (type list of Slot) and a "notches_alpha" (array or list of float) as properties of the "Lamination" object. Both have the same length and notches_alpha[ii] correspond to the angular position of the middle of notches_list[ii] (i.e. the angle to rotate the default build_geometry notch which is on the X axis). "notches_alpha" would have ordered value between 0 and 2*pi. As a reminder, in pyleecan, the first slot is on the X axis. The problem with this solution is that as we don't have "Notch" object to inherit and adapt from. 

This solution is flexible enough to match all the constraints (different kind of notches, tooth with/without notch, several notches on a tooth...) but the problem is that it is "hard" to define simple configurations. For instance having the same rectangular notch at the center of all the tooth would require to duplicate Zs time the notch and to create the linspace for the tooth centre. What we can do is to add a method init_notches ("Lamination" object) to initialize the usual notches configuration. 
This method would have three arguments: notch (the Slot object to duplicate), N_notch_per_tooth=1 (the number of notch to evenly distribute on each tooth), skip_tooth=[ ] (list of tooth index between 0 and Zs -1 to skip). This method will "simply" set the notches_list and notches_alpha properties accordingly.

Finally the get_bore_line method takes as argument the begin and end angle of the bore lines to return. We need to check if there are notches between these two angles and add the corresponding lines. As a first implementation, we can raise an error if "the position of notch + half its angular width" is too big to match the angles of the bore line. I don't know how but it would also be interesting to detect notches that are defined in the angular part of the slots. 

For uneven Lamination, I think that we will need to create a new Family of object "LamShape": one daughter class for each shape. Then the corresponding objects will provide alternative to the get_bore_line method. We will need to adapt these methods to notches since all the lamination shape will have different constraints. 

For the Slot 60, maybe we can define a Lamination shape with flat bore lines and adapt the notch to this. We can use the "delta" parameter of build_geometry to move the notch on the tooth (but then there is the question what would means notches_alpha ?). With this hypothesis we would need to remove the tooth lines from the current slot. 

What do you think of this proposal ? 

Best regards,
Pierre

SebGue, 2019-08-28:
Hello Pierre,

just one thought on your first point (" ... build_geometry won't work").
What do you think if the notch object pass all the parent methods that are needed to build the slot geometry as it is done with e.g. get_Rbo (lamination/slot objects)?
That way we could have adaptable notch objects.

Best regards,
Sebastian

BonneelP, 2019-08-28:
Hello Sebastian,

This is a great idea ! Slot doesn't use the parent property but a method get_Rbo that calls the get_Rbo method of the parent. We just need to add a get_Rbo method to the Noch class that do the same. 

That way I would propose the following: an abstract Notch class (with the get_Rbo and is_outwards methods), with two daughters (for now):
- NotchTooth (I'm not satisfied with this name): for the same notch on all tooth. With as properties: notch (the Slot object to duplicate), N_notch_per_tooth=1 (the number of notch to evenly distribute on each tooth), skip_tooth=[ ] (list of tooth index between 0 and Zs -1 to skip).
- NotchList: with "notches_list" and "notches_alpha" for advanced configuration as explained above (but without the method init_notches)

Then the Lamination class would only have a notches property. I think that if we implement the NotchList object it won't be necessary to have a list of Notch for this property. I think that as the Slot object define all the Zs slot it is more convenient to have a unique object that gather all the notches. That way we can define some generic method like get_number, comp_surface, comp_height... 

What kind of notch topology do you want to study ? Would it convenient with these two objects or do you need something else ? 

What do you think of this new proposal ?

Best regards,
Pierre

SebGue, 2019-08-30:
Hello Pierre,

at the moment, I want to modify the rotor surface. So I am fine with your proposal.
Only I would like extend the notches object, so it can also be use to have uneven lamination, i.e. replace a LamShape object.

If you don't mind, I will start with the abstract Notch class and the get_bore_line method on LamHole objects.

Best regards,
Sebastian

BonneelP, 2019-08-30:
Hello Sebastian,

I think it is a good idea to merge "Notch" and "LamShape" object into one same class family. At the end both of them will change the behaviour of get_bore_line and for some LamShape it would require some specific constraints to introduce the notches. 

No problem for you to start the development. Anyway I think that during the development we will find new ideas and constraints on this feature. I can't wait to see your next pull request. On my side I'm working on the issue 3D mesh with GMSH #16.

Best regards,
Pierre

SebGue, 2019-09-05:
Hello Pierre,

I finished my first proposal on Notches (#41). 
To add notches eg to the prius machine you can use the following code
`# === Stator Notches ===
stator.notch.append(NotchEvenDist())
stator.notch[0].notch_shape.append(Slot19(
        Zs=8, 
        H0=0.5*mm, 
        W0=2.0*mm, 
        W1=1.0*mm,
        Wx_is_rad = False
        )
    )

stator.notch[0].alpha = array([0])*pi/180

# === Rotor Notches ===
rotor.notch.append(NotchEvenDist())
rotor.notch[0].notch_shape.append(Slot19(
        Zs=8, 
        H0=0.5*mm, 
        W0=4.0*mm, 
        W1=3.0*mm,
        Wx_is_rad = False
        )
    )

rotor.notch[0].notch_shape.append(Slot19(
        Zs=8, 
        H0=0.5*mm, 
        W0=4.0*mm, 
        W1=3.0*mm,
        Wx_is_rad = False
        )
    )

rotor.notch[0].alpha = array([12.5, 32.5])*pi/180
`

In the implementation I had an issue using lists of python default classes (ie. float, ...). How can I properly do this. The class generator seems only to support lists of pyleecan classes.
As a workaround I used numpy arrays (as you can see in the code above).
Offtopic: What does the "level" parameter in the class description do?

Next step in my opinion would be to implement intersections. This would resolve the actual limitations and may also help to resolve #24.

Best regards,
Sebastian

BonneelP, 2019-09-09:
Hello Sebastian,

Indeed for now there is only the list of pyleecan type available in the code generator but I can extend it to have a list of python type. How do you prefer to define it: "list" or "[float]" in the type filed ? 
The level column is a informative column to indicate if the parameter is important or not (liked to something advanced or a main parameter of the machine). You can leave it empty and I think I may remove it one day.

Best regards,
Pierre

BonneelP, 2019-12-17:
Hello Sebastian,

I not sure if I mentioned it but there is the possibility to use list in the class generator. You can refer to the SlotUD class (I used the "list" convention for more flexibility). Note that it is possible to enter [1, 2, 3] as a default value.

Best regards,
Pierre

BonneelP, 2020-02-18:
I have added  added a new bore shape for LamHole "BoreFlower" in commit e46668cb6e77af8cf27b0fdcbd7d37ce1dc70e20. For now as it doesn't handle properly the symmetry nor the Slot, bore is a property of LamHole and not Lamination. With issue #24 the symmetry should be easier to handle. When we will add this feature for all lamination, we should extend the "get_bore_desc" method from issue #75.

BonneelP, 2020-11-05:
Hello all,

By taking inspiration from the work on "DXF_Hole" one can easily define a "DXF_Surf" that would returns only one SurfLine object from a DXF. This surface can then be used to define Bore radius or Yoke shape. As for SlotUD, we can have a "BoreUD" class that contains a list of lines to draw the bore with a method to initialize it from a list of point to connect by Segments. 

To go further we should also enable to change the shape of the Yoke with the same object (then maybe we need to find another name that "BoreX"). Maybe we can directly use SurfLine objects as properties in Lamination since get_bore_line returns a closed list of line. 

I have renamed this issue since Notches are available (property of Lamination object). 

Best regards,
Pierre

BonneelP, 2021-06-02:
Hello all, 
The status of this issue is the following:
- We have notches on bore and yoke
- We have uneven bore shape but no uneven yoke shape although it should the same principal (we may need to rename the "Bore" classes then)
- We don't have symmetry for uneven bore shape but it should be possible by adding a property to Bore objects and solving issue #24 
- We are not able to have both uneven bore/yoke and notches because we still don't know what could be the rules to handle it (if you have a topology that uses both it could help us make up our mind).

Best regards,
Pierre