Eomys/pyleecan #43: Required to import FEMM

magnetron, 2019-11-05:
Thanks for all the work that has gone into this project so far. I have no intention to use FEMM yet when playing with the current code, I cannot 'assemble' a machine/geometry without having to import FEMM (some of the geometry methods try to call/import it). Is this intentional (or am I being silly)? Wouldn't it be possible to decouple this project and let the user choose what other modules to interact with?

BonneelP, 2019-11-05:
Hello,

Thank you for your interest in our project. 

Indeed this is something we have identified and that we want to improve. The issue is linked to the fact that some methods of the Line objects import FEMM to have a cleaner coupling (for line in surface, line.draw_FEMM( )). But as these objects are the base of all the geometry, you need to have FEMM installed even if you don't use it. If we use the same principal with other software (for instance GMSH) the user will have to install every single software that pyleecan have a coupling with. 

To avoid that I saw several solutions:
- Dynamic import of femm in the Line objects (fastest way to correct your issue by waiting a more general solution)
- Only some particular objects directly linked to the software import the corresponding packages. For instance the MagFEMM object would have some methods to draw the arcs instead of the Arc having a method to be drawn in FEMM. If you don't use a MagFEMM object in your simulation, you don't import it so you don't import femm.
- As the software will grow, we will have more and more dependency, some for only one model of one physic. It could be interesting to tag all the methods of the objects with particular dependency (femm, gmsh, panda...) in the csv files (by adding a new column beside the methods). Then we could adapt the code generator either to generate different version of the code with or without some object/methods (and so with or without the corresponding dependency), either to import the method in the class only if the corresponding package is installed.

What is your opinion on this topic ?
Best regards,
Pierre Bonneel

SebGue, 2019-11-05:
Hello Pierre,

I had some similar issue. I have some object methods, I don't want to commit to the project.
My workaround corresponds the most to your second proposal and is as follows:

I have a package file, that contains the additional method for some existing object.

```
"""@package Private Methods LamSlotMag\do_something"""

from pyleecan.Classes.LamSlotWind import LamSlotWind

def do_something(self):
    """ additional method for LamSlotWind object """
    ...
    return some_value

LamSlotWind.do_something = do_something

```

And I import this method as part of the package.

`from _Methods.Machine.LamSlotWind.do_something import do_something 
`

I don't know if this is best practice but it seems to work and may be a solution to the above issue.

Best regards,
Sebastian

BonneelP, 2019-11-05:
Hello Sebastian,

I'm not sure if I understood correctly your solution. When you need your new method in a script you add the line:
`from _Methods.Machine.LamSlotWind.do_something import do_something `
in the script and it add dynamically the corresponding method to the LamSlotWind object. Am I correct ? 

If so, it may create some confusions because you may have some part of the code where LamSlotWind doesn't have this method. What about creating a new object My_LamSlotWind that inherit from LamSlotWind ? Then your new method is imported only if you use a My_LamSlotWind object. As long as you don't commit anything related to My_LamSlotWind, everything is on your side with a clear separation between your code and the project one. Then you can easily change the code of LamSlotWind to contribute to the project (and get the latest modifications). 
If the issue is the inheritance with LamSquirrelCage maybe you can change it to inherit from My_LamSlotWind or create a My_LamSquirrelCage object.

Best regards,
Pierre Bonneel

SebGue, 2019-11-05:
Hello Pierre,

my suggestion was maybe a little bit misleading. (I also don't wanted to have inherited classes.)

If we want to avoid the FEMM import issue by having some seperation, my solution is to add some class methods only if the MagFEMM class for instance is imported. 
Adaped to your example it will be as follows.
We have a function line_draw_FEMM() somewhere within the MagFEMM class code and assign it to the line class only if MagFEMM is imported.

```
from pyleecan.Classes.Line import Line
from pyleecan.Methods.Simulation.MagFEMM.line_draw_FEMM import line_draw_FEMM

Line.draw_FEMM = line_draw_FEMM

```

I don't know where to place this code exactly, but I think this will be a clean solution and it may be easy to include into class generator. 

[define-a-method-outside-of-class-definition](https://stackoverflow.com/questions/9455111/define-a-method-outside-of-class-definition) 

Best Regards,
Sebastian

BonneelP, 2019-11-06:
Hello Sebastian,

I understand better now. Thank you for your clarification. In this case, I guess the best place may be the __init__ file of MagFEMM. If the class is imported (aka used) the code is called and Line is modified.

Regarding the class generator, we can easily do the following:
When we import all the methods of the objects we replace each method import:
`from pyleecan.Methods.Geometry.Arc.draw_FEMM import draw_FEMM`
by:
```
try:
    from pyleecan.Methods.Geometry.Arc.draw_FEMM import draw_FEMM
except ModuleNotFoundError:
    draw_FEMM = import_error_fct
```
Ideally when called the import_error_fct would raise an error like "Impossible to use Arc.draw_FEMM, femm package missing". But I have no idea yet how to create such function. There is also the issue of how to handle the arguments of such function (that should be compatible with the call of any method)
An intermediate step would be to set draw_FEMM at None in case of ModuleNotFoundError. But it would create some confusing error message. 

What do you think ? Have you some ideas on how to design the import_error_fct ?

Best regards,
Pierre

magnetron, 2019-11-07:
Pierre, Sebastian,

thank you both for the explanations & suggestions. What if, for the FEMM-related functions/methods, we included the following in the header:
`try:
     import femm
except ModuleNotFoundError as e:
    femm = e `

And then at the beginning of the function definition:
`if isinstance(femm, ModuleNotFoundError):
    raise femm`

Shouldn't that only throw the message if we tried to use the (optional) femm functionality?

BonneelP, 2019-11-07:
Hello all,

By combining all our discussions/ideas, I think I got the general solution for the class generator:
We change each function import from:
`from pyleecan.Methods.Import.ImportMatrixXls.get_data import get_data`
to:
```
try:
    from pyleecan.Methods.Import.ImportMatrixXls.get_data import get_data
except ImportError as error:
    get_data = error
```

Then we change the definition of the method in the object from:
`get_data = get_data`
to
```
    if isinstance(get_data, ImportError):
        get_data = property(
            fget=lambda x: raise_(
                ImportError(
                    "Can't use ImportMatrixXls method get_data: " + str(get_data)
                )
            )
        )
    else:
        get_data = get_data
```
With raise_ a dummy function (to be able to define the lambda) that would be imported in all classes:
```
def raise_(ex):
    raise ex
```

The trick is to define a getter for the method when there is an ImportError, the getter is called before the method itself and so there is always only one argument (self). 

I got the following result when running the test (without pandas installed):
[image]
The error occurs only in the tests that directly call get_data.
And when all the packages are installed everything works fine.

So normally it should work for all the classes, all the methods and all the missing packages. 

What do you think of this solution ? Do you see anything to improve ? Should I update the code generator ?

Best regards,
Pierre

magnetron, 2019-11-07:
Hi Pierre,

to me, it looks like a smart way of doing it - thanks!

Cheers,

Till

SebGue, 2019-11-07:
Hello Pierre,

I'm also fine with your solution. You should add it to the generator.

Best regards,
Sebastian

closed 2019-11-07 by BonneelP