Eomys/pyleecan #46: Code generator: adapt classes code for internal use

BonneelP, 2019-11-12:
As a follow up of issue #43, it would be interesting to be able to update the classes code for internal use. The point of pyleecan is to enable people to experiment new models and new topologies and we have chosen the Apache license to enable anyone to use it even for confidential purpose. So pyleecan should have a dedicated way to separate "open source code" and "internal code". 

My suggestion would be to use the code generator. The code generator is split in two parts:
- read all the csv files to generate the "class_dict"
- generate the code of all the classes present in the class_dict
The idea would be to add a new step before generating the code: updating the class_dict with internal modifications. 

I saw several useful modifications:
- Add a new method to an existing class
- Add a new property to an existing class
- Add new classes
- Replace an existing method by a confidential one
- Update the definition of a property of an existing class

All the modifications to apply would be gathered in csv files (like the existing one of the code generator). Theses csv files would be in a dedicated folder "Generator/Internal" (name to be discussed). If the folder doesn't exist, the step is skipped. We can imagine to store the internal methods of a class in the corresponding method folder/Internal.

There are two ways to defined these csv files and we need to chose one:
- The csv file define completely the internal class, we replace the data in the class_dict. Problem: any internal csv needs to follow the modifications of the corresponding csv. 
- The csv file defines only the modifications: we update the data in the class_dict. Problem: The definition of the class is split between two files. 

An option would also be added to skip the "internal" step to enable to chose between generating  the open source pyleecan or the internal one. 

Then by adding the "Internal" folder(s) to the git ignore list it would reduce the risk to commit internal code. The only possible error would be to commit a wrong class file which would not be an issue since there is no valuable code in it. This error can be avoided by following the guideline: "before any commit we need to run the code generator, black and the unittest".

What do you think of this new evolution ? 
Best regards,
Pierre

SebGue, 2019-12-10:
Hello Pierre,

I really like this new evolution. I would go for the update/overwirte approch, that should also include the ability to have new class definitions that don't exist in the "original" pyleecan.
Even though I haven't used the git ignore list yet, I think one top level folder "Internal" might be easier to handle. This folder then could contain the original folder structure of pyleecan.

As you already stated, the code generator should then check and print classes and methods that will be modified/overwriten/updated and then ask the user to generate the modified or the original pyleecan.

Best regards,
Sebastian

BonneelP, 2019-12-17:
Hello Sebastian,

At the end, I went for the first method:

> The csv file define completely the internal class, we replace the data in the class_dict. Problem: any internal csv needs to follow the modifications of the corresponding csv.

It was easier to implement (I just use the same read function and replace the corresponding value in the dict). This method enables to modify more things than the second one (deleting a property would has required to define a specific syntax) and for later work on internal classes, the doc is gather in one single file which avoid having to go from one file to another to know where to modify a property for instance. 

There can't be unittest of this method so here is how I checked that it worked:
- I have created the folder Generator/Internal with Lam2.csv and Lamination.csv in it
- I have introduced two new parameters in the class_dict: the csv path and "is_internal" to adapt the class header. The header will also serve as a warning for pull request.
- Here is the content on my internal Lamination csv (the modifications are in colour):
[image]
- Lam2.csv is there to check that you can introduce a new class, and add a property named "test1"
- When running the code generator with is_internal=True (and the proper path), the lamination file is modified as follow: 
[image]
The header and the mother import has changed
[image]
The plot method have been replaced by the one in the Methods.Machine.Lamination.Internal folder
and some method were deleted
[image]
notch_list property has been removed and the test1 property from Lam2 is there
[image]
The axial duct definition has been updated 

The Lam2 class file was created as expected. The only missing feature would be to delete all the class files before running the class generator to make sure that all new files has been deleted.

SebGue, 2020-01-01:
Hello Pierre,

great work, thank you very much. I already adopted this feature.
The only thing I changed is the folder, i.e. "_Internal" to have it on top of the "normal" dirs.

> The Lam2 class file was created as expected. The only missing feature would be to delete all the class files before running the class generator to make sure that all new files has been deleted.

What do you about having all permanent files in "Classes" folder an underscore in their name (_check.py, _frozen.py, ...). So you could just delete all files without underscore before writing class files.

Best regards,
Sebastian

BonneelP, 2020-01-06:
Hello Sebastian,

Great idea :) 

Best regards,
Pierre

closed 2020-01-16 by BonneelP