Eomys/pyleecan #49: [NF] 2D and 3D mesh/model with SALOME

Superomeg4, 2019-11-27:
By reusing the same principal as for the FEMM/GMSH coupling, we want to create a coupling with SALOME (https://salome-platform.org/).

The main step would be to:

- Reuse the build_geometry method to generate the surface to draw (most of the geometry work is already done)
- add draw_salome methods to the Geometry object
- create new functions (matching FEMM/GMSH ones) to generate the mesh
- For the mesh definition, we propose to assign a number of element for each line of each surface. It would be a comp_mesh_dict(element_size) method of machine that would call build_geometry and compute the length of each line to deduce the number of element on the line. It would return a dictionary[line label] = . That dictionary can be then updated (as the FEMM_dict one) to tune the mesh. We have started to work on this function.
- For 3D mesh we can then create a build_geometry_3D method for the machine and lamination to define (to begin) extrusion volume based on the build_geometry surfaces #

SebGue, 2019-11-27:
Hello Raphaël, hello Pierre,

this new feature sounds interesting, since Salome seems to be a powerful tool regarding geometry modeling and meshing.
But what are your expactations and what are the benefits in terms of automated geometry creation compared with GMSH?

Best regards,
Sebastian

Superomeg4, 2019-11-27:
Hello Sebastian,

Our expectations concern more the community than technical performance.. We know a lot of people (in particular from the L2EP laboratory, Lille, France) who are using SALOME, but who don't have any tool for automated electrical machine design. 
This feature would encourage them to participate to the Pyleecan project. 

Best regards,

Raphaël

Superomeg4, 2021-06-29:
No active contributor on this topic. We may re-open it later.

closed 2021-06-29 by Superomeg4