Eomys/pyleecan #55: Material Data

SebGue, 2019-12-17:
Hello everyone,

this issue is for discussion of a consistant and relieable way to store material data (i.e. BH curves) within the saved machine files (json).
Actually, if there is a BH curve imported, it is referenced by the file name i.e.
```
"mag": {
                "BH_curve": {
                    "__class__": "ImportMatrixXls",
                    "file_path": "C:\\...\\MaterialBH.xlsx",
```
Now, if the referenced file is deleted or moved for some reason, the machine can't be used any more and has to be fixed. 

To be able to use the machine file even though, I think it is benificial to also store the import data within the machine file. 
Then the program can compare the referenced and stored data, throw some warning and proceed with one of the datasets.

What do you think about that?

Best regards,
Sebastian

BonneelP, 2019-12-17:
Hello Sebastian,

Regarding this issue the general idea was to store how to get the data in the Machine though the Import objects (the B(H) can be either imported or generated) and then the "imported" data is stored in Output for later use.

I agree that this path issue is something that we need to address: even if we store the machine .json and the B(H) file in the same folder, if we move the folder the path is broken. 

I think that we should maintain the possibility not to have the B(H) loaded in the machine object (or any "Input" object) since it may create some unnecessary memory constraints when dealing with optimization.

A easy workaround is to use the object ImportMatrixVal which is an Import object that directly contains the data to be returned. Then the machine would be saved with the B(H) curve. 
To still be able to use the other Import object, we can imagine a method "convert_to_value" that would run the "import" method and replace the Import object by the corresponding ImportMatrixVal. We may also introduce an option to the save method select either to save the object with the original Import object or with only ImportMatrixVal converted ones. 

What do you think ?
Best regards,
Pierre

SebGue, 2020-01-07:
Hello Pierre,

wouldn't it be possible to override the _set_file_path method of ImportMatrixVal to store the data to a new attribute if required, i.e. store_data = True?
This in terms would need a specific order of initialisation (store_data be set first) and a desicion what will happen if store_data is changed later.

In general wouldn't it benificial one could override generated class methods and also add an additional initialisation mehtod, _init_ (single underscore) for instance?

Best regards,
Sebastian

SebGue, 2020-01-07:
Hello Pierre,

there is one more thing on material data.
Do you mind to change MatMagnetics to one general magnetics material class instead of having MatLamination and MatMagnet with different properties.
This would simplify things such as the MatLib. 
The reason is that also magnets can be laminated and users could be interested in having them BH curves. Also non magnetic materials could be set mat.mag = None or Br = 0, mur = 1.

What do you think about that simplification?
Best regards,
Sebastian

BonneelP, 2020-01-07:
Hello Sebastian,

Regarding the Import object, we can maybe do something more general. We can add a "data" property (default None) and a "is_store_data" property (default False) to the ImportMatrix class. In get_data if data is not None we directly return data (without calling edit_matrix), else we generate/import it and store it for later use if needed. The "ImportMatrixVal" can then be replaced by the ImportMatrix class. 
I propose to use these new properties only in ImportMatrix for now (not in Import) since we don't have a generic data type. 

This can be helpful if we need to use the same data which is long to import for several input or if we want to use the same generated value for several input.

Best regards,
Pierre

BonneelP, 2020-01-07:
Regarding the simplification of the MatMagnetics I also agree. It will be easier to set the material (no other class to import than Material), we just need to set meaningful material values. Maybe we can add a method "check_lamination" to check that every parameter is set to use this material for a lamination ?
Best regards,
Pierre

BonneelP, 2020-09-14:
Hello Sebastian,

There were two new updates on this issue:
- Now the GUI enables to choose between setting the B(H) curve as a Matrix or from an Excel file (and can do the conversion). All default materials in Data folder use the Matrix way to not rely on a file. 
[image]
- Now ImportMatrix object can directly be set with ndarray (automatically converted to the corresponding ImportMatrixVal object).

Should we still add the "convert_to_value" methods for ImportMatrix daughters to convert them to the corresponding ImportMatrixVal ? Is there anything else to do on this issue or can we close it ? 

Best regards,
Pierre

closed 2020-11-05 by BonneelP