Eomys/pyleecan #60: Class Generator: Taking into account ndarray stored in dict

Superomeg4, 2019-12-25:
Hello everyone,

I have encountered a limitation to the actual class generator. The method "save" output is not JSON serializable if some ndarray are stored into a dict. 

When an attribute is directly a ndarray, the method "save" turns it into an list (which is JSON serializable). But if this ndarray is stored into a dict, the method "save" does not perform this conversion.

Best regards,

Raphaël

SebGue, 2020-01-01:
Hello everyone,

since the following issue seems to be related to this issue, I will also place it here:

It would be great if the "save" method would be able to save python default types, i.e. especially dicts and lists. That way it will be possible to have "collections" of pyleecan data, for instance machine or material databases or multiple simulation outputs within one file.

Best regards,
Sebastian

BonneelP, 2020-01-06:
Hello everyone and best wishes for the new year !

Regarding Raphaël's issue: The method to work on is not "save" but "as_dict". It is working for pyleecan object containing ndarray because in the "as_dict" method we convert the ndarray to list (and in the __init__ method we convert it back to ndarray with "set_array"). I see three solutions: 
- We introduce a new type for the class generator "dict(ndarray)" that can contains only ndarray and that do the correct convertion in as_dict and __init__
- We don't use dictionary of ndarray but dictionary of Pyleecan object containing ndarray
- We extend the save method to another way to save a python object which is not json

Regarding Sebastian's issue:
save is a method of any pyleecan object (thanks to the class generator). To be able to save a list or dict of pyleecan type we would need to create a "save_list" or "save_dict" function and extend the load function to take this new possibility into account. These new functions shouldn't be that hard to create since we can reuse the "as_dict" method.

Best regards,
Pierre

SebGue, 2020-01-15:
Hello Raphaël, hello Pierre,

@Superomeg4 could you please post some code to reproduce your issue.

I think the issue could be related to the class generator in general (not only the as_dict method).
I would expect to be only permitted to set the defined type of dict elements. But with the following code it is possible to also set ndarray dict elements for instance.

```
from pyleecan.Classes.Mesh import Mesh
from pyleecan.Classes.NodeMat import NodeMat
from pyleecan.Classes.ElementMat import ElementMat
import numpy as np

mesh = Mesh()
mesh.element["Triangle3"] = ElementMat(nb_node_per_element=3)
mesh.element["ndarray"] = np.array([1, 2, 3])

```

On the other hand, if I try to set the dict directly, I will get an CheckTypeError (str given) even though the dict element has the expected type (i.e. Element).
```
elem_dict = {
    "Tri": ElementMat(nb_node_per_element=3),
}
other_mesh = Mesh()
other_mesh.element = elem_dict

```

Maybe we should fix this first before working on the above issue.

Best regards,
Sebastian

BonneelP, 2020-01-15:
Hello everyone,

I have corrected the second part in commit: 9309f6406dc79106e629d949112052cb757e44ef
The check method was testing the key type instead of the value type.

Regarding the first error, the reason is that we call the check function only when we directly set the property. So changing the value of the dict like in your example (or calling the built-in methods) won't call the check function. There is the same issue with the list of pyleecan type and the append method (for instance). That's why we update the parent property in the getter and not in the setter. 

The only solution that I can see would be to create or own dict and list class that would call check and update the parent when needed. We can then use the class generator to convert to "normal" list/dict to our own types to ease the use of pyleecan.

Best regards,
Pierre

closed 2020-02-07 by Superomeg4