Eomys/pyleecan #66: requirements.txt

BonneelP, 2020-01-15:
Hello everyone,

As we said before we need to add a requirements.txt file for pyleecan so it is more convenient to install/use. 
Two things are bothering me for the creation of this file:
- First there are some dependencies that are not mandatory. For instance, if you don't want to use gmsh, there is no need to install it. Either we say that you need to install everything (which is a radical constraint), either we will have to discuss about the "core" dependencies of pyleecan. Maybe we can create several requirement depending of the usage (core functions only, all dependencies, core + testing for developers...)
- Once we know which package, we need to know which version. Is there a way to detect the oldest version of a package that makes the project work or should we just put the current version number and update it if we find an actual version constraint ?

How do you think we should define our requirements.txt ?
Best regards,
Pierre

Superomeg4, 2020-03-11:
Hello,

I have found a project working on this topic:
https://github.com/bndr/pipreqs

This requirement.txt generation is interesting because it is compatible with pip and so the file can be use for automated testing when someone perform a pull-request. 

Best Regards,
Raphaël

BonneelP, 2020-03-27:
Requirements added in commit c610abb42fd223c60527fdd721bacd482e4c0afb
At the end, I just take all the dependencies and the current version of each module as reference.

closed 2020-03-27 by BonneelP