Eomys/pyleecan #69: [BC] backward compatibility

SebGue, 2020-01-22:
Hello Pierre,

I got some problem with (missing) backward compatibility of pyleecan.
I have some saved "old" output objects, I can not load with the actual version. 
This is because FEMM_dict attribute was removed from OutMag class (commit of 18 Nov 2019).

Do you see any chance to add backward compatibility by ignoring attributes that have been removed for instance?

Best regrads,
Sebastian

BonneelP, 2020-01-23:
Hello Sebastian,

As a first reply I would say that we just should simply add back the FEMM_dict. This input is still relevant if we want to go further with the coupling.

Then we can modify the class generator to remove the verification on the init_dict. The more we work on the project, the more we want to play with this feature and the check doesn't really prevent errors (since we use the init_dict only with "as_dict results"), it just lock some interesting features.

Best regards,
Pierre

SebGue, 2020-01-24:
Hello Pierre,
it would be fine if you could remove the verification.
Best regards,
Sebastian

closed 2020-03-27 by BonneelP