Eomys/pyleecan #75: get_bore_desc, Lamination with list of slot and notches

BonneelP, 2020-02-10:
Hello all,

I'm currently working on a publication for ICEM 2020 (and on how to upload Jupyter notebook to pyleecan.org for the documentation). In this publication, I will present the geometrical modeller of pyleecan (build_geometry) and how it can be used to create complex and exotic topologies. 

In particular I'm investigating a new class "LamSlotMulti" a "LamSlot" with a list of Slot (and an array of angular positions) to draw uneven slot machine like (image generated with my current version of pyleecan as a proof of concept):
[image]
Then I wanted to add notches and it became quite complicated to mix both.

That's why I want to introduce a new method for Lamination objects: 
```
def get_bore_desc(self):
"""This method returns an ordered description of the elements 
that defines the bore radius of the lamination
```

    Parameters
    ----------
    self : LamSlot
        A LamSlot object

    Returns
    -------
    bore_desc : list 
        list of dictionary with key: "begin_angle", "end_angle", "obj"

For a "normal" Lamination with Zs=12 SlotW10, it would returns (T_op = tooth opening angle, S_op=slot opening angle) :
bore_desc [0] = {"begin_angle" = -T_op/2; "end_angle" = +T_op/2; "obj": Arc1}
bore_desc [1] = {"begin_angle" = T_op/2; "end_angle" = T_op/2+S_op; "obj": SlotW10}
bore_desc [2] = {"begin_angle" = T_op/2+S_op; "end_angle" = 3/2*T_op+S_op; "obj": Arc1}
...
- This bore description would then take into account the notches and the uneven bore shape. 
- The "obj" are either pointer to the corresponding Slot either the Arc to use. 
- The bore_desc can either describe the complete lamination or a symmetry part. In case of symmetry, the first angle is 0. 
- We keep the convention: The first tooth is centered on X+ axis (with or without symmetry)

The build_geometry of the "Lamination" object can then call this function to draw the bore (what ever the lamination type or the sym). The polymorphism would be on the get_bore_desc method.

I would like to take the opportunity to update the Notch definition.
In Lamination we currently have a list of Notch and then the Notch class itself contains a list of Slot object. I would like to keep the list of Notch in the Lamination and remove the list of Slot in Notch. 

The Notch class would be abstract and empty. NotchEvenDist would have a single Slot object as property and alpha as a float (with alpha=0, first notch centered on X+axis). The number of notches would be set with Zs in the Slot. 
Then we can either create a "NotchSingle" with a Slot and alpha as a property that draws only one notch centered on alpha. Either we use a NotchEvenDist with Zs=1 (which can have an impact on the geometry of some slot). 

What do you think of these modifications ?
Best regards,
Pierre

SebGue, 2020-02-11:
Hello Pierre,

your update to the Notch definition seem to keep its functionality, so from my side it's okay.

For the LamSlotMulti class, I'm not quite sure, what the benifit will be compared to a LamSlot with multiple Notches? Do you plan to have windings within the slots?

Best regards,
Sebastian

BonneelP, 2020-02-11:
Hello Sebastian,

Indeed the point of LamSlotMulti is to have winding within the slots and to have the possibility to have uneven distributed slot. For instance we can design such machine:
[image]

Best regards,
Pierre

closed 2020-02-18 by BonneelP