Eomys/pyleecan #79: [CO] Change of ROOT_DIR

SebGue, 2020-02-21:
Hello Pierre,

what do you think about a change of the "root_dir" of pyleecan to ..\XYZ\pyleecan instead of ..\XYZ ?
Therefore all imports have to be corrected, but on the otherhand its enable the user to rename the pyleecan folder without breaking the program.

Best regards, Sebastian

BonneelP, 2020-02-21:
Hello Sebastian,

Indeed that would be handy to be be able to rename the main pyleecan folder (on our side its already creating some confusion!). Maybe we can extend this thought about how pyleecan can be imported in another software. For instance, how should we organized the import if one day we go to pip ? 

Best regards,
Pierre

CedMrnl, 2020-04-08:
Hello Sebastian, hello Pierre, 

First of all, it could be great that users can install Pyleecan with pip. To do so, I think that we would need to reorganize the repository by moving the code into a pyleecan folder and to let general file such that README, LICENSE, and requirements at the root. We would also need to write some setup files as explained in this [tutorial](https://medium.com/@joel.barmettler/how-to-upload-your-python-package-to-pypi-65edc5fe9c56). To be able to run the GUI, we would need to add a *\_\_main\_\_.py* file to parse a command : 
```bash
python -m pyleecan run_GUI 
```

What do you thing about it ? 

To be able to rename pyleecan folder, I think that we could try to switch from absolute to relative imports, I can try to do it soon. 

Best regards, 
Cédric

PR #90 (2020-04-16): Relative imports
[NF] replace absolute import by relative imports to fix #79
[CC] Remove headers to avoid copyrights  on pyleecan website

Before imports in every pyleecan file were:
```python
from pyleecan.Classes.Circle import Circle
```

Now it has to be:
```python
from ....Classes.Circle import Circle
```

Theses changes allows to rename pyleecan folder. The GUI has to be regenerated to work in the new folder.
closed 2020-04-16 by BonneelP