Eomys/pyleecan #87: Change testing process by replacing unittest to pytest

CedMrnl, 2020-04-06:
Hello to everyone,

What do you think about switching from unittest to  [pytest](https://docs.pytest.org/en/latest/contents.html) for our testing process ? 

I remarked that some tests and especially validation ones can be really long and I think that it could be good to have the possibility to exclude some tests easily or to only include some of them. Pytest enables to do it easily by defining markers: 
```python
@pytest.mark.validation
def test_example):
    """Write the test here"""
    pass
```  

Pytest also handle [Pytest-cov](https://pypi.org/project/pytest-cov/) to generate coverage test and help us improving our test process. It also handles tests parametrization so we wouldn't lose any testing ability. 

Pytest can also execute unittest so the transition could be easy. We could write all new tests with pytest formalisme and progressively convert every tests and then remove ddt dependancy ? 

Best regards, 
Cédric

BonneelP, 2020-04-06:
Hello Cedric,

Great idea :) I have two remarks to go further:
- Is it possible to set several markers on a same test (for instance, "fast" "plot" "validation"...)? Whatever the answer we need to set some global marking rules to be consistent in the test.
- If I remember well there is a tool to convert automatically a unittest script to a pytest one. It may cause some issue with ddt but we can use that on all the script that don't use ddt. By the way, what is the alternative for ddt in pytest ?

Best regards,
Pierre

CedMrnl, 2020-04-07:
Hi Pierre,

You can use different markers for a single test :)
I think that we can [mark automatically tests according to their name](https://docs.pytest.org/en/latest/example/markers.html#automatically-adding-markers-based-on-test-names).

To convert automatically I will check this [project](https://github.com/pytest-dev/unittest2pytest). 

To replace ddt, there is a dedicated [marker](https://docs.pytest.org/en/latest/example/parametrize.html) : 
```Python
import pytest

from datetime import datetime, timedelta

testdata = [
    (datetime(2001, 12, 12), datetime(2001, 12, 11), timedelta(1)),
    (datetime(2001, 12, 11), datetime(2001, 12, 12), timedelta(-1)),
]

@pytest.mark.parametrize("a,b,expected", testdata)
def test_timedistance_v0(a, b, expected):
    diff = a - b
    assert diff == expected
```

Best regards,
Cédric

Superomeg4, 2020-04-07:
Hello everyone,

This could be the opportunity to set up an automated testing environment. 
This way, any pull-request would be automatically tested. 

I have several solution to propose:
    Travis CI
    GitLab CI
    Coveralls
    Codecov

Travis is already compatible with github, and it can be set up on the github repo in a few clicks. I believe  it is free for open-source project.

Best regards,

Raphaël

BonneelP, 2020-09-30:
Closed with PR #148

closed 2020-09-30 by BonneelP