Eomys/pyleecan #92: Installation / GUI issue - No module named 'PyQt5.sip'

magnetron, 2020-04-23:
Hello,

I'm very impressed by all the progress that's been made, well done!

I'm on macOS. With the latest changes, when I try to start the GUI using `python -m pyleecan`, having installed pyleecan in a python3 virtual environment using `python setup.py install` I get a ModuleNotFoundError:

` File "/Users/till/Documents/pyleecan/pyleecan/pyleecan/run_GUI.py", line 6, in <module>
    from PyQt5.QtCore import QTranslator
ModuleNotFoundError: No module named 'PyQt5.sip’`

`$pip list` shows `
PyQt5              5.14.1     
PyQt5-sip          12.7.1   ` as being installed. Uninstalling those two packages with `pip uninstall` and re-installing the required versions seems to have done the trick (I can load the GUI). Do you have any idea what could be causing this?

On another note: The current setup tries to install pyFemm, which requires pywin32, which - for obvious reasons - doesn't exist for a mac install...

Best regards,

Till

BonneelP, 2020-04-23:
Hello,
Thank you very much for your feedback, we are currently working on making sure that pyleecan is accessible for everyone so your issue is very interesting. 

Regarding the PyQt issue, can you try to install the packages on a blank python ? Another explanation could be the version of the module that are in the requirements.txt. They were set with the current version at the moment without further thought since we didn't know how we can find which is the minimum version of the module to make it work. Can you check the current version of PyQt you are using ? Do you think you can investigate these two ideas ? Unfortunately, most of us are working on Windows so for now it is complicated for us to test on other OS. 

Regarding FEMM, this is an interesting case that we were discussing few weeks ago. FEMM, gmsh and deap are non-mandatory dependencies and we were wandering if we should put them in the requirements.txt file. We chose to put them to ease the use of the full package. To solve that we have two possibilities: My colleague Cedric propose that we check if it is possible to have some OS related conditions in the requirements.txt file. The other possibility could be to provide several requirements.txt with proper names and documentation headers and to develop the "How to install" page on pyleecan.org to address this issue.

Best regards,
Pierre

magnetron, 2020-04-27:
Hi Pierre,

the installation was done in a 'fresh' python3 virtual environment. The PyQt version initially installed (and not working) was the same as the version I re-installed, which then worked. That's why I found it strange. I'll do some more searching on the issue.

Regarding your second comment, I would think that the latter option you suggested makes sense from a practical point of view (several requirements.txt files depending on what the user wants, e.g. requirements-base/common.txt, requirements-femm.txt etc.) 

Best regards,

Till

BonneelP, 2020-09-14:
Hello,

Did you found an explanation for your error ? 

At the end for the release Cédric fond the following syntax: "platform_system=="Windows" to avoid installing femm on other platform. Then FEMM and GMSH are addressed on https://pyleecan.org/get.pyleecan.html

Best regards,
Pierre

HamzaFarooq007, 2020-11-03:
Hello Everyone,
I am having a problem in launching Pyleecan GUI.
The error message is: (This application failed to start because no Qt platform plugin could be initialized. Reinstalling the program may fix this problem).
If anyone is familiar with some solution, please let me know.
Thank you.

BonneelP, 2020-11-03:
Hello, 

What version of python / pyleecan are you using ? How did you install pyleecan (pip or git clone ?) 
If you are using Anaconda, you need to set the following windows environment variables:
name: QT_QPA_PLATFORM_PLUGIN_PATH
value: "path_to_anaconda"\anaconda3\Lib\site-packages\PySide2\plugins\platforms

Best regards,
Pierre Bonneel

HamzaFarooq007, 2020-11-03:
Hello Pierre Bonneel. Thank you for your response. 
I am using Python 3.8.3 and Pyleecan 1.0.2 version. Aslo Anaconda latest version. When I try to lauch GUI using notebook it was gives this error.(This application failed to start because no Qt platform plugin could be initialized. Reinstalling the program may fix this problem).
But when I tried to launch using shell it worked. Thanks again.

I am exploring Pyleecan and I think its very good open source project. In my PhD, I am working on the design of LSPM(Line Start Permanent Magnet) Motors. And I am looking to contribute in Pyleecan for LSPM module.

HamzaFarooq007, 2020-11-03:
Hello Pierre Bonneel. Thank you for your response.
I am using Python 3.8.3 and Pyleecan 1.0.2 version. And also Anaconda latest version. When I try to lauch GUI using notebook it was gives this error.(This application failed to start because no Qt platform plugin could be initialized. Reinstalling the program may fix this problem).
But when I tried to launch using shell it worked. 

I am exploring Pyleecan and I think its very good open source project. In my PhD, I am working on the design, vibro-acoustic analysis and Optimization of LSPM(Line Start Permanent Magnet) Motors. And I am looking to contribute in Pyleecan for LSPM module. Thanks again.

BonneelP, 2020-11-03:
Great news :) Feel free to open an issue to talk about LSPM. We are looking for topologies, reference machine and publications to structure our thought. Maybe you can start by sharing pictures/schematics of the machines you want to simulate. The further the discussion will go, the easiest it will be to start contributing on LSPM ;)

Best regards,
Pierre

closed 2020-11-03 by BonneelP