Eomys/pyleecan #103: Assertion error during simulation

Rushi2497, 2020-06-14:
Hello everyone,
I have been trying to simulate an IPMSM motor defined by me using the steps mentioned in the tutorial. At the last step though, an assertion error shows up:

~\Documents\GitHub\pyleecan\pyleecan\Methods\Machine\LamHole\build_geometry.py in build_geometry(self, sym, alpha, delta, is_simplified)
    123     for hole in self.hole:
    124         Zh = hole.Zh
--> 125         assert (Zh % sym) == 0  # For now only
    126         angle = 2 * pi / Zh
    127         # Create the first hole surface(s)

AssertionError: 

Can someone explain where exactly I have gone wrong in defining the machine?
From the error it seems like something is wrong in the rotor part. But I am not sure what I have defined wrongly in the rotor part.

BonneelP, 2020-06-15:
Hello,
The error occurs when we check that the symmetry factor is correct for the rotor. For now in the simulation you have to define the symmetry "by hand". In the MagFEMM object if you set is_symmetry_a=False the symmetry will be deactivated and it should work (https://pyleecan.org/tuto_Simulation_FEMM.html#MagFEMM-configuration). You can also find the correct symmetry parameters for your machine. By the way, does machine.plot() work ? I will add an error message. 

If your machine/simulation is not confidential, please share it with us when opening an issue so we can better understand what is happening.

Best regards,
Pierre

Rushi2497, 2020-06-15:
Thank you. Yes the machine.plot() works fine.
The machine is indeed confidential and hence I haven't shared.

closed 2020-06-16 by Rushi2497