Eomys/pyleecan #104: Pyleecan under Ubuntu Linux - trying to get FEMM running on Wine

normansdog, 2020-06-15:
It would be nice to have Pyleecan running on Linux but FEMM would have to run using Wine which needs to be tried out.

magnetron, 2020-06-18:
Alternatively, you could look into xfemm, which is supposed to run natively on Linux. 
Does anyone have an idea of how feasible it is to link to xfemm instead of FEMM?

BonneelP, 2020-06-18:
Hello,

Having the pyleecan coupling with FEMM working as well on Linux would be great. As for now FEMM is the only magnetic module, pyleecan is kind of "windows only" which it should not. There is a dedicated page on FEMM about how to make it run on Linux: http://www.femm.info/wiki/LinuxSupport

Indeed XFEMM can be a good solution (we should be able to reuse most of the existing code), but it would require to find a way to call it from python (I think that there is no "pyxfemm"). The xfemm provide code for C++ and Matlab/octave. When working on MANATEE (matlab) we used xfemm and there will be also the need to "clean" the geometry (remove duplicate lines and points). Anyway we need to work on such a feature for issue #16 as well (that was for xfemm that we started the "is_simplified" work). 

Does anyone knows what kind of modifications is required to call FEMM using Wine ? 

Best regards,
Pierre

crobarcro, 2020-06-18:
I am the author of xfemm. The problem with FEMM on Linux is that it cannot use ActiveX on Linux and instead uses a very slow and less reliable file based communication method. Otherwise it works surprisingly well on Linux. I'd happily accept contributions to xfemm to add python support, but have no time to develop it myself.

BonneelP, 2020-06-18:
Thank you for your feedback. Do you have any document/information about how much work it would represent to provide a python support to xfemm? We are not using too many femm methods so maybe we can start the work by focusing on what we are needing. Although, I don't think that I will have the time to develop it myself, maybe someone from the community can be interested.

BonneelP, 2022-02-28:
It seems that there was an update in Pyfemm in July 2021 that can help on this topic:
[image]
It could be worth investigating if it solves our coupling problem.

harshasunder-1, 2022-08-31:
Hello,
I was able to merge the latest pyfemm  mentioned above with pyleecan and get FEMM drawing and simulation working on ubuntu. Its pretty slow, and parallelization wasnt working though. On long simulations it also crashed once in a while. Should I share the code in a PR so that the community can see if and improve it?

BonneelP, 2022-08-31:
Hello,

Great news :) Indeed we would be very interested by this PR. We can discuss afterwards about how to improve/make parallelization work. Thank you for having investigated this issue !

Best regards,
Pierre

amunozh, 2022-09-20:
Hello,
I'm also very interested into running FEMM on Linux. I've tried several options:
- I've executed FEMM using pyfemm, but it is still too slow because it relies on file system and parallelization does not work.
- I've interfaced FEMM with octave both under Wine. I think it is feasible running on parallel, but then I have to install also Python under Wine to be able to wrap the octave code. (I think it is too much complicated)
- I tried with xfemm directly from Octave, it works really fast in parallel using linux, but I'm getting far different results w.r.t FEMM results. 

So if you need some help to integrate FEMM (or xfemm) with pyleecan to run it parallely, let me know. I would be happy to contribute.

BonneelP, 2022-09-21:
Hello,

I'm wondering if pyleecan installer can be installed in Wine ? If yes it may enable to run some simulations in FEMM maybe even with parallelization (I'm not using Linux so I never used Wine). The installer is available here: https://pyleecan.org/get.pyleecan.html. Do you think you can check that ?

If you want to contribute, we already have an open PR on the subject: #560

Best regards,
Pierre

harshasunder-1, 2022-09-22:
Hello!
Good news - i was able to run the pyleecan installer in Wine. Once installed it was able to parallely run the simulations in FEMM also! The pyleecan installer installs without trouble. However it crashed on my machine till I installed winetricks and used it to install vcrun2015. (using the command ./winetricks vcrun2015) 
Thanks
Harsha

BonneelP, 2022-09-22:
Great news :) Do you think you could write a small text to extend the "Method 1: GUI installer" part of [this page](https://pyleecan.org/get.pyleecan.html) for other Linux user ? If I understood correctly the process is:
- install wine
- install femm
- install pyleecan exe
- install vcrun2015

The corresponding file is available here https://github.com/Eomys/pyleecan-doc/blob/master/get.pyleecan.md but I can update it myself if you prefer.

Best regards,
Pierre

amunozh, 2022-09-22:
Awesome! I think it is a good starting point.  For me it would be more interesting to run it natively on Linux. Anyone knows how to use XFEMM? Because I think it would be the best solution.

ajpina, 2022-11-05:
Hi,
I am interested in running the GUI in linux but I found that newer pyleecan versions fail due to pythoncom (pywin32) in FEMM_Handler.py. Is there any way to disable FEMM handler in order to just open the GUI in Linux with no FEMM coupling?  
Best,
Alejandro

harshasunder-1, 2022-11-06:
> 
Hi,
I think the PR 'basic linux support' in the pull requests should allow you to open the GUI. In that FEMM_Handler supports both linux and windows. Otherwise, if you use the pyleecan exe installer through wine, the GUI should open also. (See above for one winetricks command you might need to run)
harsha

crobarcro, 2023-04-04:
FYI there is a new release of xfemm, which fixes some long standing bugs, introduces the air gap element (which implements a sliding mesh scheme to avoid remeshing) and includes a program `femmcli` which can run lua scripts just like FEMM. 

Your python code could generate a lua script to run a simulation if you didn't want to directly link to the code.

It also includes a ctest based test suite to improve reliability of releases in the future.

It also includes the code to do frozen permeability simulations implemented in FEMM, for more accurate time-stepped simulations, but I confess tis has not been extensively tested. I would welcome PRs to add tests for this.