Eomys/pyleecan #105: List of index out of range

Rushi2497, 2020-06-16:
Hello,
I have two doubts regarding plotting of the simulation results:

**1** I have been trying to plot the results of the "tuto_Simu_FEMM_results.json" available online. But every time I use the command out.plot_A_time("mag.Br"), an error at the end 'List of index out of range' shows up. How do I resolve this? Attached is the error message in pdf(a) and the jupyter file(b).
a) [error.pdf](https://github.com/Eomys/pyleecan/files/4787007/error.pdf)
b) [Online FEMM results plotting.zip](https://github.com/Eomys/pyleecan/files/4787044/Online.FEMM.results.plotting.zip)

**2** How do I create the .json results file for my own simulations. I have set the savefile option - mySimu.mag.is_save_FEA = True. But when I import this file as out = load('MeshSolutionFEMM.json') for plotting, and apply the the command  'out.plot_A_space/time', it gives the error: 'MeshSolution' object has no attribute 'plot_A_space'
Attached is the jupyter file(a) and the mesh solution json file(b) 
a) [Plotting own results.zip](https://github.com/Eomys/pyleecan/files/4787053/Plotting.own.results.zip)
b) [MeshSolutionFEMM.zip](https://github.com/Eomys/pyleecan/files/4787055/MeshSolutionFEMM.zip)

Superomeg4, 2020-06-16:
Hello, 

About the point 2),  the is_save_FEA option is only to save the MeshSolution object (which can get heavy for big simulation). I am thinking to replace it by a text file because it is a bit misleading for user. 

The methods such as plot_A_space are only available for an Output object. So,  you need to save the output object to solve your issue:
```
from pyleecan.Classes.Output import Output
myResults = Output(simu=mySimu)
mySimu.run()
my_path = "myresults.json"
myResults.save(my_path)
```

Then you can load it and plot everything:
```
out = load('my_path')
out.plot_A_space("mag.Br")
```

Does that answer your point number 2) ? 

Raphaël

Rushi2497, 2020-06-16:
Thanks a lot Raphaël! Yes, the point (2) is resolved now. But now, the same problem in point (1) is happening with myResults.json as well

Trying to plot anything brings me back to the error - "List of index is out of range"
What am I doing wrong here?

helene-t, 2020-06-16:
Hello,

The bug comes from a new version of SciDataTool. The plot_A_* methods need to be updated, I will let you know as soon as the code is available.

Helene

Rushi2497, 2020-06-17:
Yes thank you.

helene-t, 2020-06-18:
Hello,

The plot commands have just been updated (#109). Please update your version of SciDataTool:
`pip unsinstall SciDataTool`
`pip install SciDataTool`
You should have version 1.0.3

Please let us know if you encounter other issues.
Helene

Rushi2497, 2020-06-18:
Hello Helene,
I have installed the new version of SciDataTool 1.03
The issue still persists - List of index out of range - when I use the ou.plot_A_* command

Is there anything I might have missed out on doing?

helene-t, 2020-06-19:
Hello,
Do you get this error with your own results only, or with the "tuto_Simu_FEMM_results.json" as well? Have you downloaded the latest version of Pyleecan? It is still a development version but we should soon make a release.

Rushi2497, 2020-06-19:
Yes, I downloaded the latest version of pyleecan and now it's working for both my results and tutorial results. Thanks a lot!

Also, I wanted to plot the "Tem" (Torque) object from the .json file. Is there any command available for it? I didn't see any command mentioned in the tutorials.
I tried out.plot_A_time("mag.Tem") but it did not work.

helene-t, 2020-06-19:
Indeed, Tem was not stored in a SciDataTool.DataND object, so that plot_A_time was not available for the torque. This has just been corrected (please download the latest version of Pyleecan).

Is it working well now?
Helene

Rushi2497, 2020-06-19:
Yes, it is working now. Thank you.

closed 2020-06-29 by Rushi2497