Eomys/pyleecan #111: [BC] GUI load machine error

SebGue, 2020-06-19:
Hello,

if I load a machine once (e.g. IPMSM_A.json), go to '8 RotorLam' and try to load a machine again I get the follow error message.

[image]

Best regrads,
Sebastian

CedMrnl, 2020-06-19:
Hello Sebastian,

Sorry for this issue it must be linked to my recent changes in the GUI. 
I can't reproduce this issue. Have you got more messages in the terminal used to start the GUI? 
Is this bug only related to 8 RotorLam?

Best regards,
Cédric

SebGue, 2020-06-22:
Hello Cédric,

I don't have any terminal messages. 
For me, the issue is reproducable on two different systems (both Win10 x64).
I figured out, it happens to all GUI pages every second time I try to load a machine.
Maybe someone else from your team can reproduce the issue. Otherwise I have to do some further research.

Best regards, Sebastian

CedMrnl, 2020-06-22:
Hello Sebastian,

I managed to reproduce the bug. It should be corrected once Pierre accepts the PR.

Best regards,
Cédric

closed 2020-06-23 by SebGue