Eomys/pyleecan #116: Torque ripple value from magnetic simualtion

Rushi2497, 2020-06-29:
Hello,
I wanted to know what is the unit of Tem_rip (Torque ripple) value in the output. Is it the torque ripple in N.m or the percentage of the maximum torque available?

Also, is there going to be a tutorial available on using the structural and acoustic module?

SebGue, 2020-06-30:
Hello Rushi2497,

the torque ripple is a normalized quantitie, i.e. it is the peak to peak torque divided by the average torque.
`output.mag.Tem_rip = abs((np_max(Tem) - np_min(Tem)) / output.mag.Tem_av)` - from solve_FEMM method of MagFEMM class

I personally dislike this normalization since it will lead to large 'torque ripple' values for small average torque values and is undefinded if average torque is exactly zero. I would prefer to have a Nm unit.

Best regards, Sebastian

Superomeg4, 2020-06-30:
Hello Rushi2497, Hello Sebastian,

A tutorial for the Force module is already available. But there are no Structural or Acoustic models. If you have the time to develop one, you are more than welcome :-)

Best Regards,

BonneelP, 2020-06-30:
Hello all,

The current torque ripple expression is limited to loaded machines with non-zero torque, and so cannot be used for cogging torque calculation. I think that we can introduce two definitions for the torque ripple (as a ratio of average and « peak to peak » Nm) by having two properties (with the proper definition) in the Output object. According to this new organization, I propose the following code for solve_FEMM:
```
output.mag.Tem_rip_pp = abs((np_max(Tem) - np_min(Tem)) # [Nm]
if output.mag.Tem_av != 0:
        output.mag.Tem_rip_norm = abs((np_max(Tem) - np_min(Tem)) / output.mag.Tem_av) # []
else:
        output.mag.Tem_rip_norm = output.mag.Tem_rip_pp # Or None ?
```
What do you think of this evolution ? 
Best regards,
Pierre

SebGue, 2020-07-01:
Hello Pierre,

IMHO, I would not change the unit of quantities by condition since it introduces uncertainity to the user what the condition was.
Therefore I would go with the alternative version of your code:
```
output.mag.Tem_rip_pp = abs((np_max(Tem) - np_min(Tem)) # [Nm]
if output.mag.Tem_av != 0:
        output.mag.Tem_rip_norm = output.mag.Tem_rip_pp / output.mag.Tem_av) # []
else:
        output.mag.Tem_rip_norm = None
```
Best regards, Sebastian

closed 2020-07-13 by Rushi2497