Eomys/pyleecan #117: [BC] ClassesRef: dict - Type

SebGue, 2020-07-01:
Hello all,

there are several classes that have a 'Type' of dict. If the respective dict contains a ndarray it is not possible to save the object, i.e. a json error occurs.
This may be 'HotFixed' by a type of <{ndarray}> if the data of the dict are known apriori.

@Superomeg4 I have fixed the issue for your Solution branch in 7a17c9de4a40f6503423667ca8e86b540db39677

Best regards, Sebastian

CedMrnl, 2020-07-01:
Hello Sebastian, 

I think that we serialize the dict directly instead of looking if its content is serializable. We can fix this issue by improving the class generator to handle this case. 

Anyway, I'm currently working on other save and load methods to store Pyleecan objects in hdf5 using [h5py](http://www.h5py.org/). This format is better compressed, it handle ndarray,  and it may overcome json limitations. 

Best regards,
Cédric

Superomeg4, 2020-07-01:
Hello Sebastian,

 I may have not understand everything, but I think we have already "fixed" this in a previous issue: #60. 
If I correctly fill the MeshMat .csv file with "{ndarray}" instead of dict in my branch, don't you think it would overcome this issue ? 

Best Regards,

Raphaël

BonneelP, 2020-07-01:
H5 save/load was introduced in the PR #118

SebGue, 2020-07-02:
Hello Raphaël,

if you fill the MeshMat .csv file with "{ndarray}", it will fix the issue, but it was filled with 'dict' in your branch. 
So I could not save the reulting Output object after the FEMM simulation.

You can also reproduce it with the following code on the 'Master':

```
from pyleecan.Classes.EEC_PMSM import EEC_PMSM
myEEC = EEC_PMSM()
myEEC.parameters["BEMF"] = array([1, -0.5, -0.5])
myEEC.save(save_path='MyEEC.json')

```

This may also applies to the FEMM_dict attribute of some classes, since it also has type of dict.

@CedMrnl Is there some Windows App, that let you view and edit hdf5 files easily as it can be done with json files?

Best regards, Sebastian

CedMrnl, 2020-07-09:
Hello Sebastian, 

You can view and edit hdf5 files with [hdfview](https://www.hdfgroup.org/downloads/hdfview/#download).

Best regards,
Cédric

BonneelP, 2020-09-28:
Hello Sebastian,

as_dict is now a bit more consistent with ndarray, {ndarray} and [ndarray] (PR #164). Can we close this issue ?

Best regards,
Pierre

closed 2020-09-29 by SebGue