Eomys/pyleecan #120: Installation of Pyleecan

Rui-papo-6, 2020-07-05:
Hello,

I am an e-motor designer and would like to test and potentially contribute to Pyleecan, I've done quite some scripting for geometry and winding configuration in the past on different motor topologies.
However I am totally new to Python, that scripting work was done in Visual Basic for commercial software (MagNet and Maxwell) and I am stuck at the very first step, how to install Pyleecan.

I downloaded the zip file and installed Python 3.7, then I cannot figure out what to do next.
I used the string "pip install -r requirements.txt" in the python command window but get a "Syntax error: invalid syntax" message, pointing to the second "L" of the word "install".
The same error message appears when using the string "pip install pre-commit"...

Can you help me solve this issues and getting started, please?

magnetron, 2020-07-05:
Hi & welcome! Those commands are meant to be executed from the terminal, not from inside the python REPL / command window. Sounds like that’s the first hurdle. What OS are you on?

Rui-papo-6, 2020-07-05:
Hi and thanks for the prompt reply. 
I am working on the usual Windows 7

BonneelP, 2020-07-06:
Hello & welcome,

On windows you can search for "cmd" that should open the following terminal:
[image]
The command "pip install -r requirements.txt" should work in this one.

Once everything installed, you will need to add the pyleecan folder to your Python Path to be able to import/use it. You can do that for instance directly in your python script: https://stackoverflow.com/questions/16114391/adding-directory-to-sys-path-pythonpath/16114586

We are currently working on the first official release for pyleecan to simplify the installation process but we have no release date yet. 

Best regards,
Pierre

BonneelP, 2020-09-14:
Hello,

Did you managed to install pyleecan or do you need any further help ?

Best regards,
Pierre

MVreemann, 2020-09-15:
The problem i'm facing with installing Pyleecan is when i execute "pip install -r requirements.txt" i get an error that there is no matching distribution found for vtk. 
I am quite certain this just means that vtk is either updated and the version in the requirements is no longer supported. Still i don't know how to fix the problem.

I have some other issues, but i'll make a seperate issue for those.

SebGue, 2020-09-15:
Hello MVreemann,

since there is no vtk in the requirements directly, it may be due to dependencies, maybe with meshio or pyvista.
You could try to install them on your own and see if there is the same error.

Best regards, Sebastian

BonneelP, 2021-02-05:
Hello all,

We just released a new version of pyleecan. I managed to install pyleecan from pip without any problem. Is there anything to solve for this issue or can we close it ? 

Best regards,
Pierre

kamelazeb, 2021-03-18:
I have tried multiple times to start Pyleecan and it did not work:

I installed everything.. 

Then, I watched this webinar:> https://www.pyleecan.org/webinar_1.html  ... it don't show how to start the pyleecan .. the lecturer skipped the most important part which is (run -m pyleecan)

I searched pyleecan folder and I paste it in the python working folder ... nothing worked

I gave up

SebGue, 2021-03-18:
Hello kamelazeb,

you shouldn't give up yet :-)
How do you try to start pyleecan and if there is some error message, what does it tell?

Best regards, Sebastian

kamelazeb, 2021-03-18:
Hi Sebastian,

I remember it says 'run' command isn't recognized or something like that.

SebGue, 2021-03-18:
Okay, so if xou installed pyleecan with pip then you should be able to run 'python -m pyleecan' from your terminal.

kamelazeb, 2021-03-18:
SebGue

but the guide [https://www.pyleecan.org/get.pyleecan.html] is saying we have to enter this command "run -m pyleecan" not 'python -m pyleecan'... Am I missing something?

SebGue, 2021-03-18:
Just give it a try.

BonneelP, 2021-03-18:
Hello,

Indeed we need to update the tutorial: 
- "run -m pyleecan" is a command for the notebook (used in the tutorial)
- "python -m pyleecan" is for the command line.

I will try to take some time to update the corresponding files.

Best regards,
Pierre

teodoralupu, 2021-04-08:
Hello,

I tried to install pyleecan using all the methods explained before.  
- First, directly in cmd using python -m pyleecan
- Second, I use PyCharm, so I installed packeges there, but, the error I get on both methods is: "INFO: This is taking longer than usual. You might need to provide the dependency resolver with stricter constraints to reduce runtime. If you want to abort this run, you can press Ctrl + C to do so. To improve how pip performs, tell us what happened here: https://pip.pypa.io/surveys/backtracking".
- I even downloaded anaconda and tried to install it in Jupyter terminal and I got this message: "ERROR: Could not find a version that satisfies the requirement vtk (from pyvista>=0.25.3->pyleecan) (from versions: none)
ERROR: No matching distribution found for vtk (from pyvista>=0.25.3->pyleecan)"
I downloaded the pip and put it in the Python folder and then "run -m pyleecan" and it does recognize the program.

Am I doing something wrong?

Best Regards,
Teodora

BonneelP, 2021-04-08:
Hello,

Thank you for your interest in pyleecan.
Which version of python are you using (We recommend >=3.6)? Normally you need to run "pip install pyleecan" before being able to call "python -m pyleecan"
Can you also check your pip version ? I heard that they were some issues with pip new backtracking feature so you may need to revert your pip version. 
If pyvista is an issue, this package shouldn't be mandatory (as long a you don't go in depth with Mesh manipulation) so you can skip its installation.

Best regards,
Pierre

teodoralupu, 2021-04-08:
Hello,

I have the latest version of Python (3.9), but I tried on 3.8 too. My pip version was 20.03 on the first try and then I updated it to 21.0.1. What version are you using? Maybe that is an issue for me.

Thank you,
Teodora

BonneelP, 2021-04-09:
Hello,

On my side I'm using python 3.8.6 and pip 20.2.1
Can you try installing the requirements directly to check what is going on ? The file is available at: https://github.com/Eomys/pyleecan/blob/master/requirements.txt
The corresponding command is:
`pip install -r path/to/requirements.txt`

Best regards,
Pierre

17350545062, 2021-06-05:
Hello, 
I run the command 'python -m pip install pyleecan' in cmd windows several times and install by the pycharm, it always stuck in this package 'importlib_metadata-0.12-py2.py3-none-any.whl'(the message say: pip try all the other version of this package and can not find the match, this version 0.12 is the last version limited by the dependency), my python version is 3.9, 32bit.
I just want report this bug and I will try install pyleecan without the help of pip.

SebGue, 2021-06-05:
Hello,

I don't know if this is related to your issue, but had also some installation issue with python 32 bit. So you maybe should try python 64 bit if that is possible.

Best regards, Sebastian

17350545062, 2021-06-07:
Thank you, Sebastian
My OS is Windows10 64bit. I use python to manipulate 32bit software, which is not compatible with the 64bit python, and the other 64bit software, which are all compatible with 32bit python. So for the compatibility, I tend to use 32bit python. 
I will try the other way around and I will issue it if I find the way.
Best regards, Shu Chun ZHENG

17350545062, 2021-06-15:
Hello,
After several tests, I find pyleecan can install via pip while the version of python is 3.8 64BIT. The reason why pyleecan is not compatible with python 32BIT is that pyleecan rely on some 64bit packages which are not compatible with 32BIT python. And do not install pyleecan with the python version big than 3.8 .

SebGue, 2021-06-15:
Hello, could you tell which package this is. Maybe there is some chance to have a substitute for it.
BR Sebastian

17350545062, 2021-06-16:
Hello, I am not sure if I am correct for I just tried to install pyleecan with python 3.9 32bit and 3.8 32bit manually via setup.py,  and it failed in the 'h5py' , and I tried other way around by puting the code in the path 'C:\Program Files\Python38\Lib\site-packages', and tried to run run_GUI.py in the pycharm, so I got the error track to the 'h5py' and after a seach in the pypi, I found the package 'h5py' only available for window 64bit. I am a amateur in python and if I am wrong, please point out.
I only did those two tests and I know it not enough to pin the real problem. Meanwhile it is not rush to adjust pyleecan for 32bit python. I am focus on the project of using pyleecan to build a model suitable for my python Taguchi Methods optimization script and I will share my experience of the modeling base on pyleecan.
Shu Chun Zheng

BonneelP, 2022-02-18:
Hello, 
Thank you for your interest in pyleecan. 
I'm personnaly on windows 10 (64 bits) using python 3.8.6 and I managed to install pyleecan 1.3.8 with pip 20.2.1
We haven't checked if pyleecan is installable on all python version (although I don't see why it wouldn't be). Is it possible for you to try another python version maybe ?

If it can help here are my current version of the requirements of pyleecan:
cloudpickle                       2.0.0
ddt                               1.4.4
deap                              1.3.1
ezdxf                             0.14.2
h5py                              3.6.0
matplotlib                        3.3.4
meshio                            4.4.6
mock                              4.0.3
nbconvert                         6.4.1
nbformat                          5.1.3
numpy                             1.21.4
pandas                            1.4.0
pyfemm                            0.1.3
PyQt5                             5.15.4
PyQt5-Qt5                         5.15.2
PyQt5-sip                         12.8.1
PySide2                           5.15.2.1
pytest                            7.0.0
pytest-excel                      1.5.0
pytest-qt                         4.0.2
pyuff                             2.0
pyvista                           0.31.3
pyvistaqt                         0.5.0
quadpy                            0.16.10
SciDataTool                       2.2.3
scipy                             1.8.0
setuptools                        60.8.1
swat-em                           0.6.3
xlrd                              2.0.1
xlwt                              1.3.0

Hope it helps,
Best regards,
Pierre

ErikViking1968, 2022-02-18:
Hi Pierre,

With your python version, everything works ok now. I have installed all the
packages manually but I had to install first numpy before I could install
matplotlib.

With pytest I had an error concerning permission error on the application
data....But I can work now with the program. I run it from the windows cmd.
I will test now the rest, first machine I put in gave some errors with
producing the femm file. But maybe I did something wrong. I am mostly
interested in simulating large HV machines (10 poles etc).

Thank you for the help.

Regards,

Erik

BonneelP, 2022-02-18:
Great to know that you managed to install pyleecan :)
Regarding your machine you can share your error messages in a dedicated issue if you need any assistance.

Best regards,
Pierre

ErikViking1968, 2022-02-21:
Hi Pierre,

I have installed Anaconda and made my own environment for Pyleecan. Works
great but I have some problems.
When I use the tuto-Machine, the Toyota_Prius json file comes up with only
one phase (A+ and A-). And ofcourse a lot of errors when I follow the
tutorials.
When I open Pyleecan and select the same file, you definitely see the 3
phases.
What is going wrong?
Then in Pyleecan when I select Plot Stator Unit MMF, the FFM harmonic is
not showing, see enclosed picture.

When I put in a high voltage machine, I get still errors (always the same
error, independent of the 2 designs tested) when I want to export to FEM.
The error code is:
"[13:26:24] Error while drawing machine 160323 EQRG560LB2 in FEMM:
unsupported operand type(s) for *: 'int' and 'NoneType'"
I think I missed something in the installation?
I have enclosed the sample json file.
Femm works with the example json files you send with the Pyleecan
installation.

Any idea's?
Kind regards,

Erik

BonneelP, 2022-02-21:
Hello,
Maybe we need to update something in the tutorial. I can check that but first the image and json file are missing. Can you upload them again (maybe directly on the github website https://github.com/Eomys/pyleecan/issues/120) ?
Best regards,
Pierre

ErikViking1968, 2022-02-21:
[image]
[image]
[image]
[image]

I cannot upload json file

BonneelP, 2022-02-21:
Regarding the mmf plot I guess it maybe link to the latest changes in SciDataTool (this plot is more powerful now)
Regarding the machine without the json it can be tricky can you send it by email to pierre.bonneel(at)eomy.com ?
I will try to investigate this at the end of the afternoon if I have the time.

BonneelP, 2022-02-22:
Hello,

So the first issue is in fact:
[image]
The idea is that in the GUI we use PySide2 as a package to handle Qt and for the winding we use swat-em which uses PyQt5 to handle Qt and it creates a conflict in the notebook. We managed to find some workaround in some cases but solving that may require some investigation (it used to work so we need to find what have changed). Sadly by waiting the notebook won't be usable for pyleecan (and I don't think that I will have some time to investigate that in the near future)

Regarding the plot mmf as expected, we need to update the code to use the new SciDataTool 2.0 GUI:
[image]
I'm currently working on a PR which will speed-up the coupling with FEMM, it will include this update inside and make a release afterwards.

Best regards,
Pierre

ErikViking1968, 2022-02-22:
Ok thanks Pierre, no problem I was just wondering if I did something wrong. I will use notebook as how it should be and simulate directly in python. this should help me to understand and play with the different objects and classes. 

Thanks for clearing out and your help!

Best regards, 

Erik

ErikViking1968, 2022-03-03:
Hi, sorry to come back. I am facing problems with version conflicts on PyQT5. 
I can use Anaconda and Spyder (4.1.4) and with PyQt5.12.1 I can work Pyleecan in Spyder (without GUI). When I am running Pyleecan Gui in the anaconda prompt I am forced to install PyQT5.15.2 version. When installed the 5.15.2 version GUI works perfect but Spyder doesn't run anymore.
I know a lot is discussed concerning Spyder and not possible to run PyQt5.15 version. Any ideas? 
Do I have to use a different IDE?
My goal is to run the GUI and to run/write own programs...

Best regards, 

Erik

BonneelP, 2022-03-03:
Hello,
We also have some trouble using the GUI with Spyder. If you are looking to run/write program then I would recommand to (this is my configuration on windows): 
- Uninstall Anaconda/Spyder to make sure that there will be only one python on your computer (to avoid conflict or path issues). 
- install python 3.8 (from python website not windows store)
- install VScode with the python plugin.
- install git/TortoiseGit to be able to clone the code (don't use pip install pyleecan otherwise you won't be able to update the code).  
- use pip install -r requirements.txt (file at the root of the project folder) to get all the required packages
- I can add a file to the pyleecan repository that help running the GUI from VScode if you need it.

Best regards,
Pierre

gforti84, 2022-03-03:
I decided to use miniconda to manage these packages within the environment. I installed using ´pip install´. 
Is there a way to keep the code updated using the miniconda? I got the python engine in these environments to work with the VS Code as well.

BonneelP, 2022-03-07:
Hello
If you installed pyleecan with pip install, you need to run "pip install -U pyleecan" after each new release of the package. For now we have some modifications within the master of pyleecan that you can't use since we haven't created a new release but we can create a new release "quickly" upon request (next release should be after the end of #498)

Best regards,
Pierre

Dionigi998, 2023-02-25:
Hello,
I am trying to install Pyleecan, but I have some troubles.
My python version is 3.11.2 (64 bit) and I tried to install Pyleecan typing  "python -m pip install pyleecan" on the cmd, but this error came out:

C:\Users\Dionigi>python -m pip install pyleecan
Collecting pyleecan
  Using cached pyleecan-1.4.2.tar.gz (11.9 MB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: cloudpickle>=1.3.0 in c:\users\dionigi\appdata\local\programs\python\python311\lib\site-packages (from pyleecan) (2.2.1)
Collecting ezdxf==0.14.2
  Using cached ezdxf-0.14.2-py3-none-any.whl (741 kB)
Requirement already satisfied: h5py>=3.2.1 in c:\users\dionigi\appdata\local\programs\python\python311\lib\site-packages (from pyleecan) (3.8.0)
Requirement already satisfied: matplotlib>=3.3.2 in c:\users\dionigi\appdata\local\programs\python\python311\lib\site-packages (from pyleecan) (3.7.0)
Collecting meshio<=4.4.6,>=4.0.15
  Using cached meshio-4.4.6-py3-none-any.whl (158 kB)
Requirement already satisfied: numpy>1.19.5 in c:\users\dionigi\appdata\local\programs\python\python311\lib\site-packages (from pyleecan) (1.24.2)
Requirement already satisfied: pandas>=1.0.3 in c:\users\dionigi\appdata\local\programs\python\python311\lib\site-packages (from pyleecan) (1.5.3)
Requirement already satisfied: pyfemm>=0.1.3 in c:\users\dionigi\appdata\local\programs\python\python311\lib\site-packages (from pyleecan) (0.1.3)
Collecting pyleecan
  Using cached pyleecan-1.4.1.tar.gz (11.9 MB)
  Preparing metadata (setup.py) ... done
  Using cached pyleecan-1.4.0.tar.gz (11.5 MB)
  Preparing metadata (setup.py) ... done
ERROR: Packages installed from PyPI cannot depend on packages which are not also hosted on PyPI.
pyleecan depends on swat-em@ git+https://gitlab.com/CedMrnl/swat-em.git

I don't know what to do, can someone help me please?

BonneelP, 2023-03-06:
Hello,

It was tricky but thanks to @CedMrnl we finally understand what is going on. After investigation, the issue is linked to Python 3.11 and PySide2. In fact PySide2 is not available for Python 3.11 (they are moving to PySide6 - Qt6). As pip can't install 1.4.2, it roll back to the previous versions 1.4.1 and 1.4.0 that have the swat-em issue (that was solved with 1.4.2). So the error message is "wrong". By typing pip install pyleecan==1.4.2 we managed to find the correct error message that is linked to PySide2 as explained above.

So the solution is to use a python version up to 3.10 (we are currently working on 3.8) or to use the GUI exe: https://pyleecan.org/get.pyleecan.html. We are about to release the next version of pyleecan once #558 will be merged (this week I think), I will add the limitation on the python version. 

In the near future, we will investigate moving to Python 3.11 and updating all the package but this is a lot of work, so I don't know exactly when it will be done.

Best regards,
Pierre