Eomys/pyleecan #124: [BC] Machine class inconsistency

SebGue, 2020-07-09:
Hello all,

I found some inconsistency in the Machine (also in MachineSync and MachineAsync) class. The plot method uses the rotor and stator property while this propery is not defined in this class.
It's not a big issue since it's an abstract class, but maybe we should correct this some day.

Best regards, Sebastian

BonneelP, 2020-07-10:
Hello Sebastian,

Indeed, now that we have moved the stator and rotor to MachineSCIM (for instance) we need to update this method. What about adding a method "get_lam_list" that return the list of all lamination (from ext to int for instance). That way the plot method of MachineUD would be moved to Machine. 

Anyway, I think that some further optimization/reorganization can be done in the plot method. In particular now we have SurfRing that enables to return the correct patches with the use of alpha for the inner radius. So maybe we don't need to take into account the order of the plot.

Best regards,
Pierre

BonneelP, 2020-07-17:
Hello Sebastian,
I have introduced the get_lam_list method. As rotor and stator are property of MachineDFIM, MachineIPMSM.... I unfortunately had to provide the method for all these machine types. This method is also a first step for the extension of the FEMM coupling to N lamination.
Best regards,
Pierre

BonneelP, 2020-11-05:
PR #192 update the plot method of machine to use get_lam_list
PR #210 remove the plot method of MachineUD since now plot use get_lam_list
To go further see issue #212

closed 2020-11-05 by BonneelP