Eomys/pyleecan #126: GUI does not start due to config_dict reorganization

mxgnsr, 2020-07-14:
Hey guys,

Unfortunately the GUI doesn't work anymore due to the changes in commit 95166146.

Hopefully that can be fixed easily.

BR
Max

BonneelP, 2020-07-15:
Hello, 

Thank you for noticing. It should be corrected now :)

Best regards,
Pierre

mxgnsr, 2020-07-16:
Hello Pierre,

There is still a minor mistake in Functions/init_environment.py line 95-96 it should be "Dara" instead of "Plot"

BR
Max

BonneelP, 2020-07-16:
Hello Max,

We have moved the pyleecan_color.json file into Data/Plot folder. In init_environment/init_user_dir (line 59) we now copy this folder into join(USER_DIR, "Plot"). The idea is that we copy some main files into APPDATA and then we don't use the join(MAIN_DIR, "Data") folder anymore. So I think lines 95 and 96 are Ok.

Best regards,
Pierre

BonneelP, 2020-07-17:
Hello Max,

I changed the code anyway: if you were using an old version of the APPDATA folder then the Plot folder doesn't exist. Now we check that all the expected folder are present for the backward compatibility. You can also delete your previous version of the folder and it will be updated. 

Best regards,
Pierre

mxgnsr, 2020-07-20:
Hello Pierre,

Yeah, the problem was on my side. Thanks for updating.

BR
Max

closed 2020-07-22 by mxgnsr