Eomys/pyleecan #127: [NF] Class Generator - List of ndarray

SebGue, 2020-07-15:
Hello Pierre,

it seems there is no type '[ndarray]' implemented yet. Maybe you could add this feature easily. (For me, it would take some time, since I'm not into the ClassGenerator.)

Best Regards, Sebastian

BonneelP, 2020-07-15:
Hello Sebastian,

For which application do you need a list of array ? Is it possible to add another dimension to your array or to use a Data object ?

Best regards,
Pierre

SebGue, 2020-07-15:
Hello Pierre,

it is for an application I told you before. (I have to discuss if we publish it if I'm done with it.)
But that may also apply to the 'MachineUD' class, i.e. I want to store multiple Br, Bt results corresponding to multiple lamination gaps for example. 
For now I use pure 'list' type. BTW: Is it possible to save lists of ndarrays?

OT: With my new classes, I see a lot of possible improvements especially to the FEMM integration. I will discuss all my findings with you in a while.

Best regards, Sebastian

helene-t, 2020-07-16:
Hello Sebastian,

We are currently working on a new SciDataTool object: VectorField will be an object to store fields with 2 or 3 components (e.g. magnetic flux B will contain Br and Bt). Components will be DataTime or DataFreq objects.
VectorField will have `get` methods similar to DataTime or DataFreq objects, with access to radial, tangential and axial components (e.g. `get_rad_along("time", "angle=0")` or `get_mag_tan_along("time=0", "wavenumber=[0,80]")`). We plan to replace Br and Bt by such an object in OutMag. We will let you know as soon as those changes are effective.

Concerning multiple results, you may want to use an additional axis in a SciDataTool object:
```
Airgap_pos = Data1D(
    name="airgap",
    values=positions_array,
)
Br = DataTime(
    axes=[Time, Angle, Lam],
    values=Br_array,
)
Br.get_along("time", "angle", "airgap")
```

Best regards,
Helene

SebGue, 2020-09-11:
Hello to all,

could anybody help me, since I'm struggling setting up a list of SciDataTool.Classes.DataND.DataND type in the ClassRef.
Or is this not possible at the moment.

BR Sebastian

CedMrnl, 2020-09-11:
Hello Sebastian,

This feature is not available yet. Maybe we could improve the class generator to handle dict and list of external type, or just SciDataTool types. 
If you define a simple list instead of _[SciDataTool.Classes.DataND.DataND]_, the class should work but you will have to use pkl format to save your object. 

Best regards,
Cédric

SebGue, 2020-09-11:
Thank you! Could you tell me what 'pkl format' is.

CedMrnl, 2020-09-14:
Hello Sebastian, 

Pyleecan can serialize objects in a file to save them using [cloudpickle](https://pypi.org/project/cloudpickle/). To do so, you just need to use the file extension *.pkl* in the *save_path* argument of the method save. 

Best regards,
Cédric

CedMrnl, 2020-09-15:
Hello Sebastian,

I modified the class generator to handle list of external type. Now you should be able to define list of DataND and list of ndarray with *[numpy.ndarray]*.

Best regards,
Cédric

SebGue, 2020-09-15:
Hello Cédric,

thank you very much for that. The only restriction now, is that such class can not be saved to json as long as the external type has no 'as_dict' method, right?

BTW: You opened and closed a parallelization branch lately. Do you work on that topic?

Best regards,
Sebastian

CedMrnl, 2020-09-15:
Hello Sebastian, 

In fact, pyleecan as_dict method serializes external type with cloudpickle before saving the object in json so that the whole object can be saved in json. However, the limit of cloudpickle is that it only supports the same python version so we cannot save such an object and load it in different python versions. Cloudpickle discourages users to store long-term data with it.

Maybe we could use [dill](https://pypi.org/project/dill/) to save such objects or we could improve the save method to store those kind of objects in seperated files to be able to load json even if unpickling external type doesn't work. Another solution would be to improve the class generator for specific cases such as *[ndarray]* to save this data without cloudpickle. 

Concerning the parallelization, we decided to keep that work private for now, that's why I deleted the branch.

Best regards, 
Cédric

closed 2020-09-29 by SebGue