Eomys/pyleecan #144: mySimu class property error

jrenedo, 2020-08-27:
Hi,
I am trying to run the same example shown in "How to define a simulation to call FEMM" but I am getting the following error message: 

runfile('C:/Users/JRA/Documents/pyleecan/call_femm.py', wdir='C:/Users/JRA/Documents/pyleecan')
Traceback (most recent call last):

  File "C:\Users\JRA\Documents\pyleecan\call_femm.py", line 36, in <module>
    mySimu.input.N0 = 2000

  File "C:\Users\JRA\anaconda3\lib\site-packages\pyleecan-0.0.0.0-py3.7.egg\pyleecan\Classes\_frozen.py", line 33, in __setattr__
    self.__class__.__name__ + ' class has no "' + key + '" ' "property"

FrozenError: InputCurrent class has no "N0" property

I tried installing again again pyleecan in different python environments (classic IDLE and in Spyder) but I get the same error. 

Maybe there is something trivial that I am missing but I would appreciate some help.

Cheers

CedMrnl, 2020-08-28:
Hello,

It seems that you have an old version (_0.0.0.0_) of pyleecan installed :
> File "C:\Users\JRA\anaconda3\lib\site-packages\pyleecan-0.0.0.0-py3.7.egg\pyleecan\Classes_frozen.py", line 33, in setattr
self.class.name + ' class has no "' + key + '" ' "property"

If you have cloned the repository to install pyleecan, you need to pull the latest modifications in your local version and reinstall pyleecan.

You can also install the first release using the command `pip install pyleecan`.

Best regards,
Cédric

SebGue, 2020-08-28:
Hello Cédric,

it seem, that the tutorials also have to be updated, since some of the attribute types have changed (e.g. input.time is of type ImportMatix now).

BR, Sebastian

BonneelP, 2020-08-28:
Hello Sebastian,
In fact we introduced a trick in the class generator. When you provide an array (or a list) as value for a ImportMatrix, the value is converted to ImportMatrixVal(value=value). cf commit 6d21c8bbf8bc637d55bf0bf02a480a5ec07d1c8a and 672e97f3caf4345dcc5ef29de20d1090d20d99b5
That way we can "hide" a part of the complexity and the tutorial are more readable. 

Best regards,
Pierre Bonneel

jrenedo, 2020-08-28:
Hi Cédric and Pierre,

Merci beaucoup! Indeed I had an old version of Pyleecan that I downloaded long time ago. Now it works.

Cheers,
Jaime

closed 2020-08-28 by jrenedo