Eomys/pyleecan #156: Optimization Module

SebGue, 2020-09-21:
Hello Eomys,

I had the chance to finally try the optimization module, i.e. the optimization tutorial. 
@CedMrnl Thank you for implementing this great feature!

My questions are:

1. Is there actually the possibility to have dependent quantities? E.g. if I would like to have a constant air gap and my design variable is the height of the (surface) magnets, some other geo. parameter had to be adapted to maintain the air gap width.

2. Can I reuse the results of a previous optimization simulation for a 'new' simulation with bigger population or a higher number of generations?

Best regrads, Sebastian

CedMrnl, 2020-09-21:
Hello Sebastian, 

Thank you for your feedback!

For your first question, I think you can use the `OptiDesignVar.setter` by defining a defining a function that takes a simulation and a value in argument. 

For instance, if you want to modify the size of all your slot parameters, you can define the following function and `OptiDesignVar`:

```python
def slot_scale(simu, scale_factor):
   """Edit stator slot size according to a percentage

   Parameters
   ----------
   simu: Simulation
     simulation to modify
   scale_factor: float
     stator slot scale factor
   """
   simu.machine.stator.slot.W0 *= scale_factor
   simu.machine.stator.slot.W1 *= scale_factor
   simu.machine.stator.slot.W2 *= scale_factor
   simu.machine.stator.slot.H0 *= scale_factor
   simu.machine.stator.slot.H1 *= scale_factor

OptiDesignVar(
    name="Stator slot scale factor",
    symbol = "S_sf",,
    type_var="interval",
    space=[
        50,
        150,
    ],
    get_value=lambda space: random.uniform(*space), # To initiate randomly the first generation
    setter=slot_scale,
```

For your second question, the feature is not available yet be it would be great to have it. 

Best regards,
Cédric

SebGue, 2020-09-21:
Hello Cédric,

that seems to be exactly what I need :-)
I missed that setter can be a function, since in the tutorial it is a string.
So for OptiDesignVar class the setter function must have 2 inputs. The first input passed by OptiDesignVar is always the coressponding simulation and the second input is the design variables value.

The attribute design_var of OptiProblem class is a list. What is the execution order of this list. Is it from first list element to last list element?
Then it would be possible to have a last 'unused' design variable (e.g. type = 'set', space = [ 1 ]) with a setter, that is doing all the work?! This should not affect optimization itself.

Maybe you could also inherit some OptiDesignVar class, that is exactly intended to do such things after all design variables are set.

Best regards, Sebastian

CedMrnl, 2020-09-21:
Hello Sebastian, 

I wanted the tutorial to be as simple as possible but it should appear in it. 

Yes the setter function must have two inputs, the simulation to modify and the design variable value. 

The execution order of the setters follow the design variables list so you can modify your last setter so that the machine keeps the same airgap. 

Your second solution should also work as you would provide only one value. It would be a kind of processing between the simulation modification and its execution.  

I agree that the two solutions aren't really clean, so I can add a *prepocessing* property to `OptiProblem` which would be a function that would only take simulation as an argument. This function would be executed right after the setters and could also generate errors to set the simulation parameters as incompatible and avoid running it.

What do you think ? 

Best regards,
Cédric

SebGue, 2020-09-22:
Hello Cédric,

your preprocessing solution seems to be a good idea. Maybe I will find some time to add an 'advanced' turorial on that feature.

Do you have some additional documentation on the Optimization module. I was going through the methods code, but in general it was really hard to follow the ideas and get to know the main points of how you process the data.
Do you use other tools besides inline code documentation to visualize your code?

Best regrads, Sebastian

CedMrnl, 2020-09-22:
Hello Sebastian,

I will the preprocessing soon. 

Unfortunately I didn't wrote any detailed documentation. I understand your feeling as this part of the code aims to fit with both DEAP and pyleecan. Furthermore I edited it last month so that the results can be saved in _XOutput_ object.  

The main method of _OptiGenAlgNsga2Deap_ is _solve_.
DEAP is based on objects called _Toolbox_, it enables to create population of your choice. The method _create\_toolbox_ creates such object which defines the _individual_ class. This class inherits from a list (containing its design varable values) and has the following attributes:

- design_var_name_list: name of the design variables
- setter_list: list of setter to edit the simulation with the right design variables values before executing it
- is_simu_valid: boolean to know if the simulation failed
- cstr_viol : number of constraints violated
- output: output that contains the simulation to run
- fitness: objective function values

The genetic algorithm manipulates _individuals_ to compute fitness values, and perform selection, crossover and mutations.

As we wanted the results to be in an _XOutput_, the design variables values are stored in _XOutput.paramexplorer_list_ and the fitnesses values are stored in the _OptiObjFunc_ objects in _XOutput.xoutput_dict_. Each _ParamExplorerSet_ and each _DataKeeper_ contains a list of values with the i-th value corresponding to the i-th individual.

If you're not comfortable with this organization we can discuss to modify it. Furthermore, it shouldn't be to hard to create an object that doesn't rely on DEAP and its toolbox. 

Best regards,
Cédric

SebGue, 2020-09-22:
Hello Cédric,

Thank you for your explanations and the time you spent therefore. I got some moe or less technical questions.
How do you calculate the combined fitness of multiple objectives?
Will the algorithym really aim for minimum values, i.e. negative values as in the tutuorial?
Are the results of previous generations also included in the 'creation' of new individuals? Or is this not nessesary with genetic optimization.
Is there a way to use DataKeepers without them being objectives?
Have you done some simple function tests on the algorithym yet, e.g. get the max. of a 2D gaussian distribution?
How do you handle the different types of OptiDesignVar.setter, i.e. string vs. funciton? (Maybe I will adapt that to the LossModels :-) )

Best regards, Sebastian

CedMrnl, 2020-09-23:
Hello Sebastian, 

I am not sure to understand your first question. Once the simulation has been run, the algorithm executes every DataKeeper.keeper and store the result in the individual fitness which is a list. 
The current generation is compared with the N best individuals to create the mating pool so that ellitism is ensured. Then selection, crossover and mutation are performed to create the next generation. Every step is well explained in the [original article](http://www.iitk.ac.in/kangal/Deb_NSGA-II.pdf). 

The algorithm really aims for negative values. We made this choice to keep the code as simple as possible. User can juste put a minus sign to its objective to maximize it or use an absolute value function to target values close of zero.

I will add the possibility to use non objective DataKeeper today.

I implemented two tests based on Binh and Korn function and zdt3 function. There is a problem with the second when running the two tests in a row but the results are good when we run them independantly (it may be a problem of allocation). However, both tests are with two objectives functions. 

We added in the class generator the possibility to overwrite a property setter. _OptiDesignVar_ inherits from _ParamExplorer_ which has _\_set\_setter_ defined as a method. This method enables to create the function if _ParamExplorer.setter_ is defined as a string:

```python
def _set_setter(self, value):
    """setter of setter"""
    # Create the function if setter is defined as a string (e.g. simu.machine.rotor.slot.W0)
    if isinstance(value, str):
        if not value.startswith("simu"):
            raise Exception("Setter string must start with 'simu'")

        *objs, attr = value.split(".")

        accessor = ".".join(objs)
        value = lambda simu, val: setattr(eval(accessor), attr, val)

    try:
        check_var("func", value, "list")
    except CheckTypeError:
        check_var("func", value, "function")
    if isinstance(value, list):  # Load function from saved dict
        self._setter = [loads(value[0].encode("ISO-8859-2")), value[1]]
    elif value is None:
        self._setter = [None, None]
    elif callable(value):
        self._setter = [value, getsource(value)]
    else:
        raise TypeError(
            "Expected function or list from a saved file, got: " + str(type(value))
        )
```

The default simulation is not put in the first generation, maybe we could add an option to do it. 
The optimization has not been used a lot so we are really interested in your tests. 

Best regards,
Cédric

SebGue, 2020-09-23:
Hello Cédric,

thank you for that expanation. The first question was about if e.g. `f1 = [1, 3]` or `f2 = [2, 2]` will be the better solution to the optimization. In terms of the sum both will be equal but in terms of the square sum 'f2' will be better. So the question is also on how to do I normalize my objectives to somehow set it's importance.

Best regards, Sebastian

CedMrnl, 2020-09-23:
Hello Sebastian,

NSGA-II is based on Pareto-dominance so objectives don't have importance order. 

As none of these individual is dominant, the algorithm puts them in the same front and sort them according to the crowding distance procedure to keep a diversity in the population. For your case I would add a third objective integrating this square sum criterion or we could create a specific issue to add a class with a different algorithm.

Best regards,
Cédric

SebGue, 2020-09-23:
Hello Cédric,

so I guess I have to take the time read something on Pareto-dominance :-)
But this is in general nothing specific to GA, right?!

BR Sebastian

SebGue, 2020-09-24:
Hello Cédric,

I got an issue loading my opitimization results :-(

```
File "...\pyleecan\Functions\Load\load_pkl.py", line 21, in load_pkl
    obj = load(file_)
AttributeError: Can't get attribute 'Individual' on <module 'deap.creator' from '...\\Python38\\lib\\site-packages\\deap\\creator.py'>
```
I hope there is a least a workaround since it seems to be related to DEAP and CloudPickle.

BR Sebastian

CedMrnl, 2020-09-24:
Hello Sebastian, 

For you issue, I think it is because the problem.toolbox is not deleted. I will do a PR to correct it.

I think that you may retrieve your data if you use this script:
```python
from deap import base, creator, tools
from pyleecan.Functions.load import load

# Create Fitness and individual
creator.create(
    "FitnessMin", base.Fitness, weights=[-1 for _ in self.problem.design_var]
)
creator.create("Individual", list, typecode="d", fitness=creator.FitnessMin)

my_xoutput = load("myfile")
```

For the Pareto-dominance, I think that's it depend of the way of comparing fitnesses. We could use a different criterion in another class.

Best regards,
Cédric

SebGue, 2020-09-24:
Hello Cédric;

you saved my data :-) thanks a lot!
(I only had to set weights to an empty list, since there is no self.)
BR Sebastian

ajpina, 2020-10-27:
Hi All,
Have anyone tried to implement an optimization where the objective functions are evaluated in parallel? I wonder if there is any way in the current code that multiples instances of FEMM can be launched while the optimization is taking place (pool).
This would be great to speed up the optimization based on Electromagnetic analyses. I understand that multiprocessing is currently supported by DEAP framework. Is there any constraint in pyleecan that prevent that from being implemented?
Best,
Alejandro

BonneelP, 2020-10-27:
Hello, 

Tomorrow morning we should merge PR #190. It includes the possibility to run several FEMM in parallel ;) As a first step we need to check if it is compatible with multi-simulation that way each simulation would be faster (if Nt_tot>1). Then we would need to investigate how we can reuse the same principal on each simulation of the multi-simu (even without FEMM)

Best regards,
Pierre

BonneelP, 2020-11-06:
Hello Alejandro,

Now that the second webinar is done, do you need any further information for on this issue or can we close it ? During the presentation I was using nb_worker > 1 and time periodicity. So the multi-simulation itself is not parallelized but the MagFEMM computation are (on time steps).

BTW the replay of the second webinar is available here: https://pyleecan.org/tutorials.html, https://pyleecan.org/webinar_2.html, https://www.youtube.com/watch?v=Kj-DJGpbBkU&feature=emb_logo

Best regards,
Pierre

ajpina, 2020-11-06:
Hi Pierre,
It's fine with me. Thank you.

closed 2020-11-06 by BonneelP