Eomys/pyleecan #157: ModuleNotFoundError: No module named 'PyQT5'

Nerdtron9000, 2020-09-21:
I keep getting this error and I am not sure why. I am running it in a virtual environment and below I have listed all the packages installed on the environment (Note pyqt5 is installed). I am just really confused.

(CS2316) D:\Varun\College\CS 2316 - Data Input and Manipulation\Participation>conda list
WARNING: The conda.compat module is deprecated and will be removed in a future release.
packages in environment at C:\Users\varun\.conda\envs\CS2316:

Name                    Version                   Build  Channel
ca-certificates           2020.7.22                     0
certifi                   2020.6.20                py38_0
icu                       58.2                 ha925a31_3
jpeg                      9b                   hb83a4c4_2
libpng                    1.6.37               h2a8f88b_0
openssl                   1.1.1g               he774522_1
pip                       20.2.2                   py38_0
pyqt                      5.9.2            py38ha925a31_4
pyqt5                     5.15.1                   pypi_0    pypi
pyqt5-sip                 12.8.1                   pypi_0    pypi
python                    3.8.5                h5fd99cc_1
qt                        5.9.7            vc14h73c81de_0
qtpy                      1.9.0                      py_0
setuptools                49.6.0                   py38_0
sip                       4.19.13          py38ha925a31_0
sqlite                    3.33.0               h2a8f88b_0
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_3
wheel                     0.35.1                     py_0
wincertstore              0.2                      py38_0
zlib                      1.2.11               h62dcd97_4

(CS2316) D:\Varun\College\CS 2316 - Data Input and Manipulation\Participation>python PE03.py
Traceback (most recent call last):
    File "PE03.py", line 3, in <module>
        from PyQT5.QtWidget import QApplicaton, QWidget
**ModuleNotFoundError: No module named 'PyQT5'**

BonneelP, 2020-09-22:
Hello,

Were you trying to run the GUI when the error occurs ? It seems that none of the pyleecan dependencies are installed. Did you run the command "pip install pyleecan" or did you download the software from Github ? More informations are available on the installation page: https://pyleecan.org/get.pyleecan.html

Can you also detail the content of your PE03.py script ? 

Best regards,
Pierre

CedMrnl, 2020-09-22:
Hello, 

It seems that you tried to import PyQt5 with an uppercase T. `from PyQt5.QtWidget import QApplicaton, QWidget` should solve this problem. 

> (CS2316) D:\Varun\College\CS 2316 - Data Input and Manipulation\Participation>python PE03.py
> Traceback (most recent call last):
> File "PE03.py", line 3, in
> from PyQT5.QtWidget import QApplicaton, QWidget
> **ModuleNotFoundError: No module named 'PyQT5'**

Best regards,
Cédric

Nerdtron9000, 2020-09-22:
Hello,
Thank you Pieere and Cedric for the responses. I had not installed pyleecan, only pyqt (that was the direction given to me by my teacher). Now I installed pycleecan from 'pip install', but that still didn't fix the issue. Additionally changing PyQT5 to PyQt5 didn't fix the issue. My environment now had the following packages:

Name                    Version                   Build  Channel
appdirs                   1.4.4                    pypi_0    pypi
ca-certificates           2020.7.22                     0
certifi                   2020.6.20                py38_0
cloudpickle               1.6.0                    pypi_0    pypi
cycler                    0.10.0                   pypi_0    pypi
deap                      1.3.1                    pypi_0    pypi
h5py                      2.10.0                   pypi_0    pypi
icu                       58.2                 ha925a31_3
imageio                   2.9.0                    pypi_0    pypi
jpeg                      9b                   hb83a4c4_2
kiwisolver                1.2.0                    pypi_0    pypi
libpng                    1.6.37               h2a8f88b_0
matplotlib                3.3.2                    pypi_0    pypi
meshio                    4.2.0                    pypi_0    pypi
numpy                     1.19.2                   pypi_0    pypi
openssl                   1.1.1g               he774522_1
pandas                    1.1.2                    pypi_0    pypi
pillow                    7.2.0                    pypi_0    pypi
pip                       20.2.2                   py38_0
pyfemm                    0.1.1                    pypi_0    pypi
pyleecan                  1.0.0                    pypi_0    pypi
pyparsing                 2.4.7                    pypi_0    pypi
pypiwin32                 223                      pypi_0    pypi
pyqt                      5.9.2            py38ha925a31_4
pyqt5                     5.15.1                   pypi_0    pypi
pyqt5-sip                 12.8.1                   pypi_0    pypi
python                    3.8.5                h5fd99cc_1
python-dateutil           2.8.1                    pypi_0    pypi
pytz                      2020.1                   pypi_0    pypi
pyvista                   0.26.0                   pypi_0    pypi
pywin32                   228                      pypi_0    pypi
qt                        5.9.7            vc14h73c81de_0
qtpy                      1.9.0                      py_0
scidatatool               1.1.2                    pypi_0    pypi
scipy                     1.5.2                    pypi_0    pypi
scooby                    0.5.6                    pypi_0    pypi
setuptools                49.6.0                   py38_0
sip                       4.19.13          py38ha925a31_0
six                       1.15.0                   pypi_0    pypi
sqlite                    3.33.0               h2a8f88b_0
vc                        14.1                 h0510ff6_4
vs2015_runtime            14.16.27012          hf0eaf9b_3
vtk                       9.0.1                    pypi_0    pypi
wheel                     0.35.1                     py_0
wincertstore              0.2                      py38_0
xlrd                      1.2.0                    pypi_0    pypi
zlib                      1.2.11               h62dcd97_4

Unfortuatly PE03.py is an assignment from a class and I am hesitant to post it online. However the issue is with this line: 
from PyQt5.QtWidget import QApplicaton, QWidget

I appreciate the help.

Varun

CedMrnl, 2020-09-22:
Hello Varun,

An "s" is missing in _PyQt5.QtWidgets_. The line `from PyQt5.QtWidgets import QApplication, QWidget` doesn't raise any error on my side.

By the way, could you give us your teacher's email ? We are really interested in the educational resources using pyleecan and it would be great to share them to help people learn how to use it.

Best regards,
Cédric

Nerdtron9000, 2020-09-23:
Hello Cedric,
That managed to fix the issue, Thanks. I can't believe that all the time of troubleshooting was due to a simple spelling error. I want to let you know that I was a little confused earlier, I downloaded pyleecan as I felt it was necessary to run pyqt. My class isn't using pyleecan, simply pyqt for a GUI. If you are interested in reaching out to professors who use pyleecan I would recommend checking out the electrical engineering branches of various colleges. I know that pyleecan is used at my college, but I am not aware of any professors (just a few forum posts). If I do find someone who uses pyleecan I will let you know. Once again, thank you so much and sorry for the confusion.
Varun

closed 2020-09-23 by Nerdtron9000