Eomys/pyleecan #158: New magnetization type for spoke type PMSM

mxgnsr, 2020-09-23:
Hello everyone,

I would like to build a machine that looks something like this.

[image]

I have already done the new rotor slots but I'm now stuck in adding a new magnetization for the flux concentration. Could someone help me out with that and give me some tips where to change things properly.

Thanks and best regards 

Max

SebGue, 2020-09-23:
Hello Max,

the magnetisation direction is set in ...\Functions\FEMM\assign_FEMM_surface.py 
Do you plan to release this new rotor topology?

Best regards, Sebastian

mxgnsr, 2020-09-23:
Hello Sebastian,

So actually the Labels are "Radial", "Parallel" and "Hallbach". Should we call its something like "Tangential"?

I'm going to push the new topology by now. 
I called the slot "SlotMFlat2"... Maybe we could find a better name.
I also will need some further help with the test function.

BR 
Max

SebGue, 2020-09-23:
Hello Max,

in my understanding, the labels "Radial" and "Parallel" are meant to be along the height of the magnet. So "Tangential" would be along the width of the magnet, which is not true. At least if the smaller edge of your magnet (according to your figure) is the magnets height.
@BonneelP, @CedMrnl, @Superomeg4 What do you think?
Regarding function tests we can maybe adapt it from some other slots after you pushed your code. 

Best regards, Sebastian

SebGue, 2020-09-23:
BTW: You can use the 'Generator\uml_generator.py' to generate an uml diagram (found under 'Classes\class_uml.gv') to visualize the classes with their inheritance to get better overview. You only need a proper viewer.

BonneelP, 2020-09-23:
Hello Max, 

The magnetization direction are defined in the ...\Functions\FEMM\assign_FEMM_surface.py function. This file would need to be optimized in a more general way (related to issue #10). For now you can extend the following code:
```
        elif "Magnet" in label:  # LamSlotMag
            if "Radial" in label and "_N_" in label:  # Radial magnetization
                mag = "theta"  # North pole magnet
            elif "Radial" in label:
                mag = "theta + 180"  # South pole magnet
            elif "Parallel" in label and "_N_" in label:
                mag = angle(point_ref) * 180 / pi  # North pole magnet
            elif "Parallel" in label:
                mag = angle(point_ref) * 180 / pi + 180  # South pole magnet
            elif "Hallbach" in label:
                Zs = lam.slot.Zs
                mag = str(-(Zs / 2 - 1)) + " * theta + 90 "
```
If I understood well the magnetization you are looking for, indeed you can add the "Tangential" magnetization with something like:
```
            elif "Tangential" in label and "_N_" in label:
                mag = angle(point_ref) * 180 / pi + 90  # North pole magnet
            elif "Tangential" in label:
                mag = angle(point_ref) * 180 / pi - 90 # South pole magnet
```
Just make sure that the build_geometry of your magnet shape set the proper label and that FEMM define the expected magnetization. I think that "Tangential" is clear enough (in 2D), but we can maybe use "Circumferential" to be more general.

Ok for the name "SlotMFlat2". We do need to define some proper guidelines about how to properly name our slot. 

Best regards,
Pierre Bonneel

SebGue, 2020-09-23:
Hello Pierre, hello Max,

as I pointed out "Tangential" is misleading in my opinion. If you have a look at HoleMagnets their 'radial' magnetization is also fixed to the magnets height, although their magnetization in the lamination might not be in radial direction.

BR Sebastain

BonneelP, 2020-09-23:
Hello Sebastian,

As I said, I may have misunderstood what the expected magnetization is. To be sure:
[image]
For me the red arrow correspond to the height of the magnet (for slot the height is always in "radial direction") and to the Parallel magnetization. Then if the expected magnetization is the blue arrow I would name it "Tangential" or "Circumferential". I think that the issue is that with this definition "Tangential" is orthogonal to the "Parallel" definition and not the "Radial" one (which change in the magnet according to the angle).

Best regards,
Pierre

SebGue, 2020-09-23:
Hello Pierre, 
this was not an offend. In the Ende it is just a naming convention. But with an additional tangential magnetization there also have to  be some more ifs and elses in the code. Also the magnet class has to be updated.
BR Sebastian

BonneelP, 2020-09-23:
Hello Sebastian,

> this was not an offend. 

Don't worry it wasn't taken as such ;) Sorry if my latest message let you though otherwise. This kind of discussion is very important for the consistency of the project :) Thank you for your messages. 

It all go down to what we call "height" for this magnet. The parallel magnetization was defined with this kind of magnet in mind:
[image]
Indeed in this case, "Parallel" means parallel to the height.

The [slot convention](https://pyleecan.org/slot.convention.html) define the height of a slot as "The slot height is defined as the difference between the bore radius and the slot bottom radius". With that in mind it makes sense for me to define the magnet as: 
[image]
Which is consistent with the other LamSlotMag magnet height definition but I agree that it can be counter intuitive to have a magnet with Hmag > Wmag. 

If we use the "intuitive" definition:
[image]
then I agree that it make sense to set the magnetization as "Parallel" by using a method similar to the HoleMag to find what is the angle of the magnet. We would then need to introduce a clean way of computing the angle of the magnet for both Hole and Slot in order to merge all definitions in FEMM (LamHole and LamSlotMag) in a single "if" (which would greatly improve the code). But in that case the height of the magnet and of the slot would have two different definitions.

Indeed, if we introduce the "Tangential" magnetization, it would require to update Magnet.csv to add "3: Tangential" (and change the max value to 3). Then all build_geometry of each Magnet and HoleMag need to be update to take 3 as a possibility (and raise an error if value>3 to remember to update if we introduce other magnetization). Maybe for some magnet kind we can just raise an error to say that this magnetization doesn't make sense (same as Radial). It should be also added to the GUI. Only then, the code I suggested earlier can be added to the "if LamSlotMag" part. 

I also agree that "Radial" magnetization doesn't make sense for this kind of magnet. 

Best regards,
Pierre

SebGue, 2020-09-23:
Hello Pierre,

there are pros and cons on both solutions as pointed out. Now, how do we get a desicion?
BTW: I thougth we already had a disscusion on faulty magentization direction and how to realize that, but I don't know the result.

OT: (Maybe worth an own issue.) I had a look at the classes graph and saw a lot of potential for improvement there. My initial idea/ question was - why don't we use an existing slot for this type of rotor. The answer is all Slots that may fit the above geometry are SlotWind, although there is no obvious to me they are restricted to windings. Also there is some inconsistency since LamSlotWind stores the winding within the lamination, while LamSlotMag stores the magnet in the slot.
(I know the above geo. is not exactly the same as the SlotWind's since it has no yoke :-) )

BR Sebastian

BonneelP, 2020-09-23:
Hello Sebastian,

For this case I would say that the best choice is to add the "Tangential" magnetization because it will introduce a new feature that can be helpful for other topologies or experimentation. Anyway I will add this file to issue #10 because we need to reorganize this part of the code.

I can't agree more, it is something that bother me for a while. Yes this will be a dedicated issue. I see three main improvement for the architecture:
- Unification of SlotMag and SlotWind by setting a "special zone" that is either a winding or a magnet
- Using the split method (once fully validated) of surface to automatically split a "special surface" into the correct winding layers (build_geometry_wind method are way too complicated) 
- Possibily to use Slot and Hole regardless of the machine type (induction machine with closed slot, or spoke PMSM). I will try to take some time to properly write all 3 issues.

Best regards, 
Pierre

BonneelP, 2021-02-05:
Hello,
With the unification of SlotMag and SlotWind, SlotMFlat2 is now SlotM16. 
Everything seems done for this issue. Can we close it or is there any further development required for spoke PMSM ? 

Best regards,
Pierre

BonneelP, 2021-03-22:
Hello, 

In PR #336 the GUI for SlotM16 was introduced. I'm closing this issue for now, feel free to reopen it if it needs further work ;)

Best regards,
Pierre

closed 2021-03-22 by BonneelP