Eomys/pyleecan #166: Class Generator SyntaxWarning

SebGue, 2020-09-29:
Hello Pierre,

with the new Class Generator, I get a lot of SyntaxWarning, e.g.
```
File "...\pyleecan\Classes\MachineIPMSM.py", line 253
    elif value is -1:  # Default constructor
SyntaxWarning: "is" with a literal. Did you mean "=="?
```
(Didn't you get these warnings? What's your python version? Mine is 3.8, i think.)

I tried to fix this with the suggested '==', but this gave me some other issues with list type attributes, e.g. ImportMatrixVal.value.

Then I recognized that the __init__ description states, that ' -1 will call the default constructor'. Following the code, that seems not to be true anymore. Further do you think '-1' is needed? If so, then the order of the __setter__, i.e.
```
def _set_value(self, value):
        """setter of value"""
        if value is -1:
            value = list()
        elif type(value) is list:
            try:
                value = array(value)
            except:
                pass
        check_var("value", value, "ndarray")
        self._value = value
```
should be corrected. Else if value is not a list `if value is -1` will result in an error.

I can do the changes to the code. I only need to know, what the correct 'behavior' should be.

BR Sebastian

BonneelP, 2020-09-29:
Hello Sebastian,

Thank you for taking the time to inspect the class generator. My latest PR change quite a few things and we need to make sure that everything is still Ok.

The "is" is to avoid issue when the value is a list or an array. Otherwise value == -1 would raise that we need to call .all( ) for ndarray but if the value is a ndarray the condition shouldn't be called (since we expect an int). value is -1 is True only if value really is -1. So to answer to the warning I do mean "is". Or maybe we can change it to:
```
if isinstance(value, int) and value == -1:
```

We introduced the "-1" as default argument because if we set:
```
__init__(value=[])
```
Then the list is unique for all instance and we have the following behaviour:
```
a = Obj()
b = Obj()
a.value.append(1)
print(b.value)
    [1]
```
Same for pyleecan objects. 

Anyway there is indeed an error when setting ndarray with -1:
```
        if value is -1:
            value = list()
        elif type(value) is list:
```
Either the "elif" need to be change to "if" or value = list() need to be changed to value = array(list())

Regarding the comment "-1 call the default constructor" it was meant for pyleecan classes:
```
    def _set_mat_type(self, value):
        """setter of mat_type"""
        if isinstance(value, str):  # Load from file
            value = load_init_dict(value)[1]
        if isinstance(value, dict) and "__class__" in value:
            class_obj = import_class(
                "pyleecan.Classes", value.get("__class__"), "mat_type"
            )
            value = class_obj(init_dict=value)
        elif value is -1:  # Default constructor
            value = Material()
        check_var("mat_type", value, "Material")
        self._mat_type = value
```
Maybe we can update this description ? 

Best regards,
Pierre

SebGue, 2020-09-29:
Hello Pierre,

I fixed the code according to your proposal. Only I use 'type' instead of 'isinstance' to be consistant with the other code. (I can also switch to 'isinstance' if you like.)

For the other part of the issue (the '-1'), I better should have read the code comments :-)
I wasn't aware that '__init__' already uses the '__setter__' methods. 

You are right, it's maybe not the best idea to have mutable class default input values :-)

Regarding unit tests, would it better to wait for #148 being merged? 

Best regards, Sebastian

BonneelP, 2020-11-06:
Hello Sebastian,

I guess that we can close this issue or is there any other work ?

Best regards,
Pierre

closed 2020-11-06 by SebGue