Eomys/pyleecan #168: [NF] Class Gen: Function serialization

BonneelP, 2020-09-29:
Hello all,

Recently the Class Generator was modified to ease save and load operation in pyleecan. Now list and dict of array are properly converted with call to lolist and SciDataTool are now longer serialized with cloudpickle (with call to as_dict). Other packages types are not that common and shouldn't be used as much as possible (again to avoid call to cloudpickle).  The next type that need to be properly handled is function (DataKeeper, PostFunction...)

Here is a proposal of how we can try to handle function to remove the need to use to cloudpickle:
- Function should always be set with string containing either code to evaluate, either a path to a python file to run.
- The string would be stored in the class as _prop_name_str
- The resulting function would be stored as _prop_name_func
- The setter will convert the string to the proper func and set both properties
- The getter will return only _prop_name_func
- as_dict would only contain the _prop_name_str

This would require further investigation to know how to properly convert string to function. Should the string to evaluate be lambda only ? If we use python script should we import a function with a standard name or evaluate the script ? 

What do you think of this new evolution ?
Best regards,
Pierre

SebGue, 2020-10-02:
Hello Pierre, 

just a short response to some aspect of this issue and #174.
How do we deal with the 'eval' function regarding security reasons?
I also read that 'You can’t evaluate an import statement using eval().' Source: [Python Eval Function](https://realpython.com/python-eval-function/)

BR, Sebastian

BonneelP, 2020-10-02:
Hello,

> How do we deal with the 'eval' function regarding security reasons?

Indeed eval can provide anyone the tools to inject some strange and unwanted code but on this topic I think that the issue is more on the fact that we use function rather than the fact that we use eval. Anyway as pyleecan is open-source I think we can trust the user code.

> I also read that 'You can’t evaluate an import statement using eval()

This is something worth investigating. For now eval is used for lambda function (that shouldn't have import statement). For more complex function we use exec. So if an import is needed the solution may be to use a file (or a function directly). Another solution would be to try dynamic import.

Best regards,
Pierre

BonneelP, 2020-10-02:
Hello Sebastian,

I have updated my test in 7e30383e803d69c755ea52e0606b31770db2c20f to use a file with import and it works :)

Is the PR Ok for you ?
Best regards,
Pierre

SebGue, 2020-10-05:
Hello Pierre,

I had a look at the code (in your PR description) and would like to recap the actual possibilities. Correct me if I'm wrong.
- One can set 'inline' functions, i.e. lambdas as a string that will be saved in the json.
- The string only option, e.g. 'simu.machine.stator.L1' seems to be removed (by the above).
- One can have a seperate file with a function of the same name (that may have imports). 
- At last one can define/reference a function directly.

If the 3rd option won't save the file content inside the json, I think the PR is okay. So the user can always check the function file before he is loading and running the json, i.e. there is no 'hidden' code inside the json files. 

BTW: Is there a possibility to e.g. do a poll among the people who forked the project, to have a broader base on such decisions?

Best regards,
Sebastian

BonneelP, 2020-10-05:
Hello Sebastian,

> One can set 'inline' functions, i.e. lambdas as a string that will be saved in the json.

Yes (then there can be code in the json)

> The string only option, e.g. 'simu.machine.stator.L1' seems to be removed (by the above).

Still available, in fact it was only for ParamExplorer class. The corresponding code is in a dedicated setter file (Methods/Simulation/ParamExplore/_set_setter). We can add this feature to any class by surcharging the setter or changing the class generator to add it for all classes (but it is very specific to ParamExplorer).

> One can have a separate file with a function of the same name (that may have imports).

Yes in this case the path to the function file is saved in the json (which can create some issue if the file is moved)

> At last one can define/reference a function directly.

Yes but it won't be saved. 

> BTW: Is there a possibility to e.g. do a poll among the people who forked the project, to have a broader base on such decisions?

I didn't found any poll option :/ But this is why I opened this issue, to take the time to discuss about it. Anyone can participate to this discussion, I just directly refer to you to answer your questions. I will continue to wait for reply on this issue before merging it (maybe by the end of the week if nothing major is discovered before). Anyway this PR doesn't break the backward compatibility, it just introduce some new standards and fix save/load issues so it shouldn't be an issue to merge.

Pyleecan architecture is not finished (and may never be). There will always have some new optimization and reorganization to do. We are working a lot on our private fork and we find new optimization and feature every week. This is why I have created the milestone 1.1 to discuss about what are the next steps for the project and why. We need to progressively structure the project and the development planning so that anyone can understand where we are going, why and how they can help. At the end, merging SlotWind and SlotMag will break backward compatibility but it will provide a significant improvement so it is worth investigating and discussing about it. The project is still rather young so we still need to find a way of moving forward by including everyone. Any idea on this topic is welcome :)

Best regards,
Pierre

closed 2020-10-08 by BonneelP