Eomys/pyleecan #169: [NF] Unification of SlotMag and SlotWind

BonneelP, 2020-09-29:
Hello all,

As mention in issue #158 we need to find a way to unify SlotMag and SlotWind. Otherwise we will have duplicate slot for magnet and winding (rectangular slot with winding or magnet for instance). 

This would require to rethink how both slot types are handled and redefined/rename all methods:
- The Slot would now have a "special area" (looking for a proper name, pocket ? filled zone ? **active area** ?)
- All method should be renamed accordingly and should be unified according to the current SlotWind methods
- SlotMag methods need to be simplified by having the "special area" dimensions as property of the slot object instead of the magnet (as we already do for HoleMag)
- The method of the abstract SlotWind class would be moved to the Slot class (then all slot need to have a "special area" that can be the full slot)
- Instead of build_geometry_wind a method "get_area_special" would return the corresponding surface object. Then the split method of issue #24 would enable to split the surface in the corresponding winding layer (or even magnet layers, why not).
- For now winding is a property of Lamination and magnet of the Slot. Both should be in the corresponding Lamination daughters. 
- The possibility to have several magnet to define a same pole should be compatible with this new organization by moving "W3" to LamSlotMag and still using a list of magnet. Maybe we can also have a LamSlotMultiMag class for that ? 
- The methods "has_winding" and "has_magnet" would be useful to add. Maybe we can use them for check in the machine instead of enforcing Lamination types in the machine classes. 

On a side note, one day we should also consider the possibility to use either Slot or Hole regardless of the machine type (induction machine with closed slot, or spoke PMSM). We won't consider this for now, but having winding and magnet in Lamination would be a first step (although HoleMag objects have magnet as properties). 

What do you think of this proposal ?
Best regards,
Pierre

BonneelP, 2020-10-19:
SlotWind classes are now easier to define thanks to the possibility to cut surfaces (by the way, this cutting method needs to be improved to return both "top" and "bottom" part in one call). 

Here is a proposal for the renaming of the slot methods:
- SlotWind.comp_height_wind => Slot.comp_height_active
- SlotWind.comp_surface_wind => Slot.comp_surface_active
- SlotWind.comp_radius_mid_wind => Slot.comp_radius_mid_active
- SlotWind.comp_angle_wind_eq and SlotMag.comp_angle_opening_magnet => Slot.comp_angle_active_eq
- SlotWind.get_surface_wind => Slot.get_surface_active
- SlotWind.plot_wind => Slot.plot_active (plot get_surface_active result)
- SlotWind.build_geometry_wind => Slot.build_geometry_active (still use Nrad and Ntan to enable to cut magnets as well - with a new magnet type ?)
- SlotMag.get_point_bottom => This method needs to be removed (use build_geometry_active)
- SlotMag.W3 => LamSlotMag.W3
- SlotMX.magnet => LamSlotMag.magnet_list
- SlotWind and SlotMag classes will be removed
- As an answer to #40 SlotX.build_geometry => SlotX.comp_edges ?

W3 and magnet_list can be moved into LamSlotWind, because they are in fact related to the build_geometry of the lamination.

As a first step, the schematics for SlotMFlat and SlotMPolar won't change and the magnet will always match the slot. Then new parameters will be added to these slots to change the magnet surface. 

Best regards,
Pierre

BonneelP, 2020-10-20:
The last issue is how to draw the magnet in the slot ? 
First, the magnet types that we have are limited to the slots that match their bottom line (rectangular or polar). So there is already a strong link between the magnets and the slots. Then I think that we can remove the Magnet daughters and handle the magnet geometry directly in the Slot (as we do for the winding). Since now the magnet is the active surface of the slot, the magnet would be now displayed on the slot schematics (with the proper parameters). But it would require:
- either to introduce a "type_magnet" property to change the magnet shape in the slot (then one slot type with several schematics). Maybe just by playing on the value of Rtop we already have one schematics. 
- either to define a Slot object for each combination Slot+Magnet. 

On a side note we would loose the possibility to add a new magnet topology that would automatically be available for all the slot but we would gain the possibility to merge SlotMag and SlotWind

Then the magnet object would contain only the meta-data of the magnet (length, materials, insolation...) and no geometry information.

Sebastian, I know that you wanted to use the Magnet types to be able to draw some more complex geometry in holes and this may remove this possibility. In the same way, what do you think about adding some complexity in each Hole schematics to draw within the active surface ? Maybe Slot 60 and 61 can be an inspiration about how we can tune the active surface. 

What do you think about this proposal ?
Best regards,
Pierre

SebGue, 2020-10-20:
Hello Pierre,

I need my days to have 48h to do all things I want to :-)

I vote against removing the magnets, since it's non conforming to object orientation (i.e. something that also bothers me with HoleMag). Also there are at least some more relevant magnet geometries that aren't implemented yet, e.g. the non-rectangular types but with parallel sides.

What do you think instead of having a reference point in the magnet or the slot and freely place the magnet in the slot (with the most likely position as a default)? This in turns would require a check, that the magnet is smaller or equally sized compared to the slot.

Best regards,
Sebastian

BonneelP, 2020-10-20:
Hello,

> I need my days to have 48h to do all things I want to :-)
That would definitively be handy :D 

I agree that losing the Magnet daughter would be a shame because it do enable to create some new magnets that would be able to automatically fit in any existing slot. But at the end the magnet slots are not that complicated. We only have two (if we exclude SlotMFlat2 that can be define "as a SlotWind") and they are strongly linked to the magnet. I realized that I did a mistake in my previous proposal: I wanted to use the slot schematics and then add the magnet which is not conform to object orientation as you said. Instead we should use the magnet schematics as the new slot schematics (by adding the slot parameters to the magnet parameter). Then MagnetType10 daughter of Magnet would become SlotM10 daughter of Slot (with some new properties to set the magnet position as for SlotW60). For instance this can be the schematics for SlotM10:
[image]

With this new organization all the Magnet daughters would still be there and to add a new magnet shape one just need to add a new Slot. If we remove all geometry consideration from magnet then it would have the same status as for the winding which is required if we want to merge SlotMag and SlotWind (to avoid having to define slots like SlotMFlat2)

I think that it is more consistent to consider that all the geometry concern are fully handled by the slot or the hole. One day we will also want to be able to "merge" HoleMag and HoleWind so we should not consider that we would call MagnetType10.build_geometry to draw the "active area" of HoleM50. Instead I think that we can introduce some new parameter to improve the drawing of the corresponding area to include insulation for instance. 

[image]

What do you think of this new proposal ?
Best regards,
Pierre

SebGue, 2020-10-21:
Hello Pierre,

I am not completely convinced yet :-) I hope we got some more time to dicuss, since this will be some kind of fundamental change to the machine structure.
I would like to do some 'checks' on your proposal and think about the needs,goals, pros and cons and what that all means in detail. 
My main concern is that your proposal could lead to a lot of new Slots that are nearly identical, while we wanted to reduce duplicates with this.
Further will this NF enable for (feel free to edit, check and uncheck)

- [ ] isolation of winding

- [ ] gap between magnet and lamination

- [ ] empty spaces above, below or between a winding  (e.g. slot divider) and ...

- [ ]  multiple winding (systems) within a slot (see e.g. #193)

- [ ] (magnetic) slot wedges

- [ ] tangential magnet division (for e.g. eddy current reduction) with optional gap

- [ ] slotless arrangements (e.g. hallbach and airgap winding)

- [ ] consistent way to define different magnets each slot (e.g. for defective magnet sim.)

BTW: What do your colleagues think on this issue?

Further, what do you think about unification of Holes as closed Slots at this point too?

Best regards,
Sebastian

BonneelP, 2020-10-22:
Hello Sebastian,

Thank you for your answer. This is indeed a major modification and we need to make sure that the new implementation is really better. At the beginning I wanted to close this issue before the 3rd webinar but maybe it wasn't a good idea. We will take as much time as needed to discuss about it so the implement is Ok for everyone and optimized as much as possible.

> My main concern is that your proposal could lead to a lot of new Slots that are nearly identical, while we wanted to reduce duplicates with this.

In my first proposal indeed but with using the magnet as schematics for the slot it won't be duplicate any more. Each slot would define a different active shape (even if the slot looks similar). For this new definition a Slot is the "gap in the bore radius" AND the active surface. But we can still draw only the slot without the active surface (for notches for instance). 

>  insulation of winding 
>  empty spaces above, below or between a winding (e.g. slot divider) and ...

For that, either we update the schematics with parameter for the insulation (possible to optimize according to each slot), either we create a generic method that would take a SurfLine and fill it with the conductor (better solution but can be really complex). For the second case, the schematics define the global special area and we draw something more real in FEMM (by calling a build_geometry from the conductor maybe?). In both case, "build_geometry_active" would return some new surfaces with special insulation label.

> gap between magnet and lamination

For that we can adapt the schematics with now properties (default at 0) and return the proper surface with the corresponding label in "build_geometry_active". We can then add a material somewhere (in Magnet object ?) if we want to consider the glue. 

> multiple winding (systems) within a slot (see e.g. #193)

"build_geometry_active" return some generic surface with the tag "SX_RX_TX". Then it is in FEMM that we say what each surface is. So we can define any rules for that. On this particular example, maybe we can reuse the current winding matrix of WindingSC and add a second independent radial layer (and double the number of phases). Or we can have two winding object and two rotor currents (which can be use as a first step toward machine with several laminations)

> (magnetic) slot wedges

Corresponding surface returned (or not according to some properties) by "build_geometry_active". Maybe the wedge material can be define in the Slot object.

> tangential magnet division (for e.g. eddy current reduction) with optional gap

By merging SlotMag and SlotWind, build_geometry_active would be available for SlotMag. Then we can automatically split the active surface of the magnet as we now do for the winding. It can also be used for Hallbach magnetization pattern if we set different magnetization direction for each layer (which is possible in FEMM since they are separated surfaces). 

> slotless arrangements (e.g. hallbach and airgap winding)

The SlotMX classes with H0 = 0 would create a lamination "without slot" (as the edges of the slot would match the bore radius) and then, since SlotMag and SlotWind are the same thing, the active surface can either be a magnet or a winding according to the Lamination type. 

> consistent way to define different magnets each slot (e.g. for defective magnet sim.)

As the Slot is the "gap in the bore radius" and the active surface: A different magnet => a different slot. Then LamSlotMulti would work for that. 

> BTW: What do your colleagues think on this issue?

I can't tell how many of your answers/issues have started some very interesting discussion in the team ;) We talk a lot about pyleecan and I try to summarize the content of our discussions in my answers. Unfortunately we don't have enough time to have everyone discussing about everything on github :/ 

> Further, what do you think about unification of Holes as closed Slots at this point too?

Unification of Hole and slot is not a machine concern (it will remain two different object family). This unification is for the models: get_Zs (that can return Zh) for instance. So I think that we can postpone this part of the discussion. 

Thank you again for taking some time to challenge this architecture proposal :)
Best regards,
Pierre

SebGue, 2020-10-22:
Hello Pierre,

Thank you for your answer. I know the faster this whole change is implemented the better it is.
Do you also have some code (or class refs) ready that outline your proposal, since I am not sure I got all your ideas and the resulting Lamination structure.
Best regards, Sebastian

BonneelP, 2020-10-22:
Hello, 
I have no code or class ref yet. I will create a branch and start some modification to see how it goes. On my side as well having some proper method and csv would help to have a better understanding. If it doesn't work, we can still remove the branch.
Best regards,
Pierre

BonneelP, 2020-11-09:
Hello all,

I just pushed a branch with the first step for this issue. The methods of SlotWind daughters were renamed and I have created "SlotM10" as a first example of the new slot (daughter of SlotWind for now) to replace MagnetType10. Magnet have lost all its methods (and daughters), LamSlotMag methods have been reorganized. For now LamSlotMag contains only one slot/magnet. Should we have a LamSlotMultiMag or just directly having magnet_list in LamSlotMag ? 

Next step would to change the other MagnetTypeX to SlotMX and remove SlotWind and SlotMag (and check that everything still works)

On a side note, this merge introduce the possibility that the active surface overlap the airgap. I have introduce a comp_radius (on the example of the Hole ones) method for SlotM10 but this method would be required for all slots (return (Rbo, Rbo+H) in the "normal cases"). comp_height still returns the distance between Rbo and the bottom of the slot to compute the yoke height.

What do you think about these first modifications ? 

Best regards,
Pierre

SebGue, 2020-11-13:
Hello Pierre,

I didn't have the time to see your branch yet. I hope to do it next week.
As for the first question I would prefer the magnet list so we could have one magnet or Zs magnets in it.

BR Sebastian

BonneelP, 2020-11-13:
Hello Sebastian,

No problem I will see if I find some time to carry on the work to have a more complete branch.

Regarding the question, it depend what we are looking for:
1) one magnet and one slot: the magnet is duplicated in all the slot evenly distributed
2) Zs magnet and one slot: The magnet list enable to define a different magnet in each slot, the slot are evenly distributed
3) N magnet + W3 + one slot: the slot is duplicated N times, each slot is separated by W3 bore radius, the N magnets correspond to only one pole. The N slot + magnet are then evenly distributed Zs time
4) Zs magnet + Zs slot + alpha_list (something like LamMultiSlot): The slot are placed according to alpha_list and each slot can have a different magnet.
5) Something else...

Note that here Magnet objects only define materials and magnetization, there is no more geometry concern in it. If you want to change the shape of only one magnet, you need to change the slot. 

All these implementation are possible and can be interesting. For now my implementation of LamSlotMag is the first one and previously we had the third one. So my question was should I use solution 3 as the default implementation of LamSlotMag or should we introduce a new lamination type for that so the default LamSlotMag is the most common one?

Best regards,
Pierre

BonneelP, 2020-12-04:
Hello all,

Today I have continue to work on the SlotMerge branch on this repository. I have SlotM10 to 14 that matches the previous plot results of the corresponding magnets from pyleecan\Tests\Plot\test_Lam_Mag_inset_plot.py. 

I still need to add SlotM15 from #246. I need to update (or merge into test_Lam_Mag_inset_plot.py) test_Lam_Mag_surface_plot.py and to make sure that all the methods (comp_height, comp_surface....) returns the proper results (which will be convenient since now I can reuse the generic SlotWind one). Once all that validated, I will remove SlotMag and SlotWind and I will need to update the GUI as well but we are getting closer :)

Best regards,
Pierre

BonneelP, 2020-12-11:
Hello all,

New important step for this issue this week: All SlotMX classes are now available and validated with the corresponding Tests (Plot and method with comparison analytical/numerical) ! All the code is accessible in the pyleecan/SlotMerge branch

@SebGue Can you check SlotM15 that would replace your MagnetType15 ? I have added a plot test in Tests\Plot\test_Lam_Mag_inset_plot.py here is how it looks:
[image]
Is it what you were looking for ? I have also update your test method now named "Tests/Methods/Slot/test_SlotM15_meth.py
By the way with the new organization it is possible to update the slot to have the edges of the slot parallel as well and W0 in [m] instead of [rad].

@mxgnsr SlotMFlat2 is now SlotM16 (which is a copy adapt of SlotW29) and it look like that:
[image]
[image]
Is it Ok for you ?

Next step: Removing SlotWind class, fixing the few remaining method bugs and most of all updating the GUI. Hoping to be able to do the PR next week !

Best regards,
Pierre

mxgnsr, 2021-01-21:
Hello Pierre,

> @mxgnsr SlotMFlat2 is now SlotM16 (which is a copy adapt of SlotW29) and it look like that:
> Is it Ok for you ?

Thanks for mentioning! Looks very good :) 

BR 
Max

closed 2021-01-22 by BonneelP