Eomys/pyleecan #181: Does anyone want to make a design about generator?

NinomiyaSakana, 2020-10-08:
Hey guys!
Does anyone want to make a design about generator? I'm new to the machine design and trying to optimize a generator by using this tool. But now I have no idea about how to do it. Does anyone have suggestions and advice?
Thanks so much.

SebGue, 2020-10-09:
Hello NinomiyaSakana,

at first, I don't want to make a design for you :-)
But I may give you some suggestions and advice. Therefore it would be nessesary to know what your actual knowledge base (especially on machine design) is. What are your design requirements? What are your expectations on pyleecan as a tool, i.e. what do you want to simulate and optimize?
Maybe you should also attend to the upcoming [webinars](https://github.com/Eomys/pyleecan#webinar) if you are completely new to all of that.

Best regards, Sebastian

SebGue, 2020-12-15:
Hello NinomiyaSakana, 
is this issue still relevant to you? Otherwise I will close it. 
If you got questions that are not worth an issue, you may also use the new discussions feature.
Best regards, Sebastian

NinomiyaSakana, 2020-12-16:
Hello Sebastian,
I'm sorry I didn't receive the notification of your message.
I'm still working on this project but I have changed to another way. Although I have learn a lot from Pyleecan!
Thanks again and you could close this discussion.

Best regards, Sakana

closed 2020-12-16 by SebGue