Eomys/pyleecan #191: why the code organization is not like the traditional modules?

tinix84, 2020-10-20:
Hi
let me say your project is amazing and I would like to know more...
I was expecting the classical code organization of python modules like https://docs.python-guide.org/writing/structure/, but you put all the method in a external folder like the tests.
why did you go for this structure?
thanks
Riccardo

BonneelP, 2020-10-20:
Hello, 

Thank you very much for your interest in pyleecan. Indeed the organization of pyleecan is quite original: it is linked to "the class generator". We have a dedicated page on the website to introduce the tool https://pyleecan.org/class.generation.html and it was explain in more details in the architecture proposal for pyleecan (page 15 of https://eomys.com/IMG/pdf/pyleecan_architecture.pdf - this document is more than 2 years old, lot's of thing have changed!)

Basically the idea is that pyleecan have almost 200 classes (and many more will be added) so we wanted to have a convenient, quick and efficient way of creating the classes. We have csv files that fully describe every single classes and the code of each class is automatically generated to match the csv. One of the constraint is to put all the method of the class in a dedicated folder to be imported in the generated code. Having all the methods code in the class file directly would have created some very long files and errors. The architecture enforce one method one file which is better to read and maintain in our opinion. If you have any further question on the class generator we would be happy to discuss/debate about it. 

For now to use pyleecan, one needs to import the Classes from pyleecan.Classes.X but I think that now that pyleecan is available as a package we can maybe think of a more convenient way of importing the classes. Any feedback on how we can improve the organization of pyleecan is welcome.

Best regards,
Pierre

SebGue, 2020-12-23:
Hello Pierre,
I was going through the issues and had a further idea/question.
Wouldn't it be possible to have all class related files (i.e the generated class and the class methods and maybe a csv copy) in one folder for each class?
For the user it doesn't matter.  I on the other hand find myself switching between class def. and methods folders quite often. So this could be an improvement.
What do you (all) think?
BR Sebastian

BonneelP, 2020-12-29:
Hello,
The class generator is split in several steps and use several global variables for the path. So instead of scanning the ClassRef folder we could easily scan the method folder to get all the csv files and change where we save the class file. We could then merge the classes and methods folder to have Classes/<package>/<class name> that would contains the class, the methods and the csv. 
It would also be handy to investigate how we can use "from pyleecan import MachineSCIM" instead of "from pyleecan.Classes.MachineSCIM import MachineSCIM". Finally we can also take the opportunity to create some new main packages. The Simulation one is getting overcrowded. 

Best regards,
Pierre