Eomys/pyleecan #193: Designing a double cage rotor for SCIM

MVreemann, 2020-10-21:
Since we are working on the simulation of SCIMs (#165) right now I think it would be a good point to start thinking about implementing a double cage rotor for machines.

For the design i suggest the following layout:
[image]

This is in good agreement with many commercially available laminations. The dimensions as given are the dimensions which are often provided with a rotor.

BonneelP, 2020-10-22:
Hello,

Thank you for this schematics :) 
To add this new slot you will need to define a new SlotWind daughter, you can take inspiration from the following tutorial: https://pyleecan.org/tuto.add.slot.html
Note that PR #180 change how the build_geometry_wind method is defined (we will soon update this tutorial). Now you only need to define build_geometry and get_surface_wind which is an easier method. By the way, can you add on the schematics where the "winding surface" is ? Maybe you will have to cheat as in SlotW60 and SlotW61 build_geometry_wind if the "W2 part" is empty. 

Best regards,
Pierre

MVreemann, 2020-10-23:
The typical rotor for an induction motor is made of cast aluminium, so all voids will be filled. So the two slots will be part of the winding surface. So the winding surface would be:
[image]

I did already take a look at the tutorial but today i have the time to actually start working on it. Thanks for the heads up on the new convention in build_geometry.

all the best,
Menno

MVreemann, 2020-10-23:
I just made the csv file for the new Hole, called HoleDC. I don't see any problem with this file. However it is not recognized by the run_generate_classes function somehow. The file is in pyleecan/Generator/ClassesRef/Slot.
When i run the generator every file is updated except for HoleDC. So i tried recreating the file, but nothing seems to work. Somehow when i change a file which is updated nothing changes either.
Am i missing something?

[HoleDC.zip](https://github.com/Eomys/pyleecan/files/5427876/HoleDC.zip)

In the mean time i will just try to start writing the build_geometry code, which should be possible without generating the class first

SebGue, 2020-10-23:
??? I tried to generate and it worked for me. Have you checked the generator terminal output?
Regarding the name 'Hole' I think it better should be 'Slot' since it is opened.

MVreemann, 2020-10-23:
Ah, slot is indeed the better name. I will change it, but what do you mean by terminal output? If you mean the output in command prompt: 

The output seems to be alphabetically ordered, however the name HoleDC (which i will change) is mentioned nowhere in the list.
Than at the end it sais "1716 files left unchanged".

The class generator script has not changed in the last couple of days right? So it should work

BonneelP, 2020-10-23:
Can you make sure that the generator scan the correct folder ? Maybe if you have install pyleecan from pip it can be called on another folder.

BonneelP, 2020-10-28:
Hello,

The last webinar is all about "how to contribute to pyleecan". Did you finished the implementation of this slot ? If not do you think that we can use it to create the "live" equivalent of the "how to add a slot tutorial". Then I would create the schematics and push the code on the 6th November. 

Otherwise does anyone have a slot that we can use ? 

Best regards,
Pierre

MVreemann, 2020-10-29:
Hey pierre,

So far i created the points and lines, i dont know how much more there is to do, but i think there still is some way to go. But feel free to use this for your webinar. I wont mind at least.

All the best,

Menno

BonneelP, 2020-10-29:
Hello, 

Normally "points and lines" should be enough to make it work but we can add some other methods to optimize it. Can you share your code with us ? I will carry on the work for the other methods then.

Best regards,
Pierre

BonneelP, 2020-11-03:
Hello @MVreemann ,

I should start working on the slot tomorrow or on Thursday. Can you share your code with me so I can save some development time ?

Best regards,
Pierre

MVreemann, 2020-11-05:
Hey Pierre, Sorry for the long delay, i am still struggling with internet at home, but luckily i currently can finally access internet. so here are the files you asked for, i hope its not too late
[SlotDC.zip](https://github.com/Eomys/pyleecan/files/5492639/SlotDC.zip)

BonneelP, 2020-11-05:
Hello,

Thank you for your code, I will use it to create all the methods live tomorrow :)
I will be using this schematics:
[image]
The winding area is not the one you intended but it will be better for the explanation in the webinar. We can then easily change it to your definition.

Best regards,
Pierre

BonneelP, 2020-11-13:
Hello,

Do you think we can close this issue or do you need further work on the slot . For instance should we introduce a new property "type_active_surface" to switch from the surface I have defined and an active surface that would overlap the full slot ?

Best regards,
Pierre

MVreemann, 2020-11-13:
For the validation of #165 we have data on a double squirrel cage rotor with a slightly different rotor design. I am currently working on the implementation of that. I'll put the documentation in this issue. After that i think we can close it.

All the best,

Menno

SebGue, 2020-11-13:
Hello,

what do you think of a general property for all slots that is None (for default active surface) or has a user specified height for the surface starting from slot bottom.

Best regards, Sebastian