Eomys/pyleecan #195: MagFEMM naming conventions

SebGue, 2020-10-23:
Hello,

is there a documented naming convention on the labels used with MagFEMM class?
Further it would be good to have some generel program flow chart so (new) contributors get to know the organization of some class specifics. 

BTW: 
I setting up rotor bar circuits and just recognized that stator circuits were only set correctly by 'accident' :-)
I.e. with 'create_FEMM_materials' labels are tested e.g. for 'WindS' or 'WindR', then 'create_FEMM_circuit_material' check for 'Rotor' in the label to set 'Circr' (else 'Circs') while there actually is no 'Rotor' (nor 'Stator') in the label. Instead the label is 'WindR_xxx' or 'WindS_xxx'. It only works correctly because rotor is tested first.

Best regards, Sebastian

BonneelP, 2020-10-23:
Hello,

Indeed there is no documentation yet on surface and line label in build_geometry. Such documentation would be very helpful to create some new coupling or some new topologies. There are only some explanation on the https://pyleecan.org/winding.convention.html but we should do better and create a full page on the subject.
Anyway a clean up and unification of the label won't hurt (we shouldn't use R nor S in label but Rotor and Stator and even Rotor_X or something like that because one day there will be several of these). What about writing and validating the new standard/documentation and then update the code ? Do you think you would have time to work on this issue ? 

Best regards,
Pierre

BonneelP, 2021-09-14:
Done in #412, the documentation itself will be soon available on the website pyleecan.org (cf https://github.com/Eomys/pyleecan-doc/pull/5)

closed 2021-09-14 by BonneelP