Eomys/pyleecan #197: pytest import error

SebGue, 2020-10-23:
Hello @BenjaminGabet ,

since you seem to be into pytest, you maybe could help me with the follow error.
I try to run pytest (even with the -m "not GUI" parameter) and get an import error related to PySide2. 
I have installed PySide2 and I also can import it, but nevertheless I get the error.

```
___________________________________________ ERROR collecting Tests/GUI/DMatLib/test_DMatSetup.py ___________________________________________
ImportError while importing test module 'D:\GitHub\PyleecanLokal\master_clean\pyleecan\Tests\GUI\DMatLib\test_DMatSetup.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
c:\program files\python38\lib\importlib\__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
Tests\GUI\DMatLib\test_DMatSetup.py:9: in <module>
    from PySide2 import QtWidgets
C:\Program Files\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py:146: in _import
    return original_import(name, *args, **kwargs)
E   ImportError: could not import module 'PySide2.QtGui'
```

BenjaminGabet, 2020-10-23:
Hi @SebGue 

Do you use python <File> or python3 <File> to launch your file ? And also, did you do pip install PySide2 or pip3 install PySide2 ? Because there can be an error if you install PySide2 with pip install and running it with python3 <File>. Can you show me your imports please ?

SebGue, 2020-11-02:
Hello Benjamin,

I'm not quiet sure what you mean with 

> Can you show me your imports please ?

I think I'm using 'python' all the time. Also I installed PySide with python, i.e. `python -m pip install PySide2`
(If I try to run `python3` in terminal nothing happens except Microsoft store opens with Python3.8 site :-) )

If I run 'python' and e.g. `import PySide2.QtGui` there is no error. 

BR,Sebastian

BenjaminGabet, 2020-11-02:
Hi Seb,

Have you modified the file ? Try to import like that : "from PySide2.QtWidgets import *". I don't have the error on my side. Have you uninstalled PyQt5 ?

SebGue, 2020-11-02:
Due to your suggestion, I uninstalled PyQt5 and now it works, although I don't understand why. Thank you for that.
I guess, now I have no excuse to not extensively unit test my code anymore :-(

closed 2020-11-02 by SebGue

BenjaminGabet, 2020-11-02:
PyQt5 and PySide2 are both using Qt-libs. In pyleecan we are now using PySide2 instead of PyQt5. I think some errors occured when both are installed :).
Unit testing is the best way to make good code :)