Eomys/pyleecan #200: [CO] Clean import and saving of Data objects in MagFEMM module

EmileDvs, 2020-10-27:
Hello,

This issue deals with improving current import from electrical output inside the FEMM model and put all savings of Data objects (B, Tem, Phi_wind_stator etc.) on a higher level (preferably in run() method of Magnetics)

At the moment currents are set in set_FEMM_circuit_prop by doing "I.values" which is not a robust way to get values from Data object, because it implies that time values in electrical output are exactly the same as those in the magnetic model. Besides all physical quantities are saved in solve_FEMM and solve_FEMM_parallel which duplicates code that could also be useful for other magnetic models (for example, calculate bemf or torque ripple).

This issue can be an important step to demonstrate how the run() method of a physical model can be organized:

1. First define axes to store in the Output and to  (here in MagFEMM, compute Time and Angle using comp_time_angle method). This method can be renamed "comp_axes_dict" for more genericity. axes_dict contains axes that are used for calculations
2. Import output from previous physics (here currents from electrical output) on the axes calculated in first step
3. Run calculations (here comp_flux_airgap)
4. Save outputs in Data Objects (all should be put in run())

Let me know if you have any feedback on this.

Emile

SebGue, 2020-10-27:
Hello Emile, 
I think I got some addition. Since MagFEMM is steady state only, BEMF for one current only differ by time/rotor angle axes. So it would be good to have the possibility to 'recalculate' for different speeds without the actual FEMM calculation.
Therefor it may be a good idea to have a is_steady_state property in Magnetics class to determin if a recalculation is suitable.

Best regards, Sebastian

EmileDvs, 2020-11-02:
Hello Sebastian,

Thanks for your feedback.

I am going to work on this pull request for the last Webinar, to have a clean demonstration on how a physical model in Pyleecan can be implemented. Besides, we now have a new Magnetic model (MagElmer) so we'll make sure that all of these changes work well for both FEMM and Elmer models. For your information, I am going to do it on the FEMM branch of EOMYS-Public.

Concerning your point, I completely agree to add the solver type to the Magnetics class since it is compulsory for flexible solvers such as Elmer. Regarding FEMM, FEMM can solve both magnetostatic and magnetoharmonic problems. Even if we only have static solver at the moment, the solver type is also relevant for FEMM. 

For more genericity, I suggest to define a "type_solver" attribute in Magnetics class that is a string among "static", "transient" (or "dynamic" as you wish), and "harmonic". I am going to open a new issue to deal with the "transient" workflow (no need to tackle "harmonic" workflow yet I presume).

Concerning the bemf calculation, I plan to put it in the Magnetics.run() method to have it at the higher level, if Phi_wind_stator is not None (i.e. if it has been calculated). 

Best regards,
Emile