Eomys/pyleecan #218: [CO] Defining groups

Superomeg4, 2020-11-06:
Hello everyone,

The goal of this issue is to discuss how to properly define groups in MeshSolution objects. 
Until now, the idea was to store the indices of cells (or "elements") in a dict which is an attribute of the MeshSolution objects. 
For example:
`meshsolution.group = {["stator", [1,2,3,4,5,...]]} `

However, in this state, it is not possible to define groups based on nodes id. 

Two possibilities are discussed:
-> Insert "_node" in the group name when it is defined with nodes id.
-> Add a new attribute meshsolution.group_node, which would be specific to node, and rename the previous one as meshsolution.group_cell. 

Feedback is welcome :-)

BR

Raphaël

SebGue, 2020-11-06:
Hello Raphaël, 

I would go with the second solution, i.e. having node and cell groups, so there is no messing around with the group names itself.

Another solution would be to have the indeces contained in another list, e.g. 
`meshsolution.group = {"stator": [[node_ids], [cell_ids]]}`

BTW: The group property description "Dict sorted by groups name with cells indices" is somehow misleading, since Dict is not sorted.

BR, Sebastian

Superomeg4, 2020-11-06:
Hello Sebastian,

Your proposition is very interesting ! Because it enables to easily switch between the two type (it is already a problem we have faced).  @helene-t @EmileDvs @BonneelP  What do you think ? 

> The group property description "Dict sorted by groups name with cells indices" is somehow misleading, since Dict is not sorted.

That's exactly why I want to switch to list ^^

BR

Superomeg4, 2021-03-22:
Just a small update on this task:
The MeshSolution object obtained from get_group has now the same indices as in the original MeshSolution object.
The "renum()" method is called when necessary (for example, to perform plots with pyvista). 

The main task is now to defined a group from nodes. 
Right now, the group are defined from cells indices, and the new MeshSolution is obtained by finding the nodes included (based on connectivity) and "clearing" the other nodes (to save memory space). 
On the same idea, the group defined from node will allow to find all the cells which have ALL its nodes included in the group.

SebGue, 2021-03-23:
Hello Raphaël,

when do you need a feedback, since I may need some time to get into this issue again?
What does poke mean? Are you going to implement your code in basic? :-)

Best regards, Sebastian

Superomeg4, 2021-03-23:
Hello @SebGue,

"Poke" means that I want to send a notification specifically to you so you don't miss this message ;-)
For the moment, I don't need this feature for EOMYS work, so there is no hurry. 

BR

SebGue, 2021-03-23:
Ah okay. I thought the @ notation should be sufficient to do that.