Eomys/pyleecan #220: [NF] Coupling with Ansys (Maxwell, Simplorer, etc.)

EmileDvs, 2020-11-06:
Hello,

I am opening this issue to start discussion around Ansys coupling: 

- What would be the objectives of the coupling (import results from Ansys, build models in Maxwell/Simplorer, launch simulations etc.) ?

- Is there already any Python package that we could use to interface with Ansys ?

- How could we implement the Ansys coupling eventually ?

Best regards,
Emile

tinix84, 2020-11-17:
I can help for that, there are some possible implementation but we need to build an abstraction layer between pyleecan and FEMM

BonneelP, 2020-11-17:
Hello,

Thank you for your interest in pyleecan. If you can help us on this issue that would be great :) 

The coupling with FEMM already use a geometry abstraction layer. We have a method known as "build_geometry" that returns a generic representation of the machine "as a list of surface". All the topology complexity including symmetries are handled by this method and the coupling with FEMM (and gmsh) is mostly:

for surface in surface_list:
    for line in surface.lines:
        line.draw_FEMM()
    assign_surface(surface.label, surface.point_ref)

We talk more about it in our ICEM 2020 publication:
- https://pyleecan.org/icem.2020.html
- https://pyleecan.org/videos.html

What kind of simulation (machine / physics / software...) do you want to run with this coupling ? Maybe a first step can be to draw and mesh a machine from pyleecan within Ansys. Maybe we can reuse some work from the coupling with GMSH for this purpose. 

Best regards,
Pierre