Eomys/pyleecan #221: [NF] Coupling with Motor-CAD

EmileDvs, 2020-11-06:
Hello,

I am opening this issue to start discussion around Motor-CAD coupling:

-     What would be the objectives of the coupling (import results from Motor-CAD, make compatibility between Motor-CAD and pyleecan templates, launch simulations etc.)?

-     Is there already any Python package that we could use to interface with Motor-CAD?

-     How could we implement the Motor-CAD coupling eventually?

Best regards,
Emile

enjoyneer87, 2022-01-19:
Hi

- What would be the objectives of the coupling:
 ->  My first goal is, try inverse problem solving as an optimization to minimize the errors(or sigma) in a deterministic view. Then, probably,  the code can be reused on the expanded task of a way to quantify the uncertainty of my model or further detailed model with a Bayesian point of view. 

- Is there already any Python package that we could use to interface with Motor-CAD?
-> It is not precisely Python package, but MotorCAD provides ActiveX command and ActiveX parameter to link with python.  It was quite comfortable for me although I don't know how useful for further combining in the object-oriented programming frame or pyleecan it is( as I'm not familiar with  OOP).

I am working on calculating the error and correlation between the tested data (efficiency map) and the result provided by the MotorCAD LAB module.
 MotorCAD LAB can compose the loss model and saturation model based on FEA and several hybrid methods, and it is quite convenient to recalculate the efficiency map and performance curve by changing some factors regarding temperature, iron loss, AC copper loss, bearing loss, current and voltage usage.  (Since the LUT has been added, it will be possible soon in pyleecan as well.)

As explained above, since the analysis has already been performed in motorcad and pyleecan is only used as a tool for post-processing, the geometry is not defined and only the analysis data is imported in the middle to get the optimization code.
I just started take a look the Xouput Class and Opti~~ Class, Seek help with comments or criticism of starting point and any kind and.
Thank you.

Figured out things
- [ ] controllabe factor on motorcad LAB (python with ActiveX)
- [ ] I/O the result(Input(LUT) and Effi map) -> Export mat file from motorcad, import as dic type (using scipy)
- [ ] Norm for objective function (with some interpolation)

BonneelP, 2022-01-19:
Hello,
Thank you for your interest in pyleecan!
Regarding Bayesian optimization, there is an ongoing work on the topic: https://github.com/Eomys/pyleecan/issues/475
We have contacted the corresponding contributors and they should post a message about their progress soon. Maybe you can join their work ?

Normally the Optimization classes were organized with the following ideas:
- The problem definition is split from the solver to allow to chose between different solver
- Whatever the solver the output are organized in the same way in XOutput to gain access to all post-processing
- The problem doesn't require to call pyleecan cf https://github.com/Eomys/pyleecan/blob/master/Tests/Validation/Optimization/test_Binh_and_Korn.py

Best regards,
Pierre