Eomys/pyleecan #225: [BC] Initialize the mesh from MeshVTK with StructuredGrid

Superomeg4, 2020-11-06:
Hello everyone,

I have met an issue with the current MeshVTK class. In order to initialize the attribute "mesh" by hand, the only type possible is "pyvista.core.pointset.UnstructuredGrid". Unfortunately, I am working with StructuredGrid object from pyvista. 

The error can be reproduced using this code:
`    meshpv = pv.StructuredGrid()
    meshvtk = MeshVTK(is_pyvista_mesh=True, mesh=meshpv)
`

I don't know if it is possible to find a parent class to both UnstructuredGrid and  StructuredGrid.

BR

Raphaël

helene-t, 2020-11-09:
Hi Raphaël, looks like there is a `Common` class which gathers all classes. I will update the class generator accordingly. In the meantime, you can save the mesh as a vtk file and deactivate `is_pyvista_mesh`.

Hélène

Superomeg4, 2020-11-09:
I replaced "pyvista.core.pointset.UnstructuredGrid" by "pyvista.core.pointset.Common" in the MeshVTK class. 
Unfortunately, I did not work, as I get the following error:
`CheckTypeError: For mesh: Common expected, StructuredGrid given`.

We are looking for a type of pyvista or vtk object which would work.

helene-t, 2020-11-10:
Actually it looks that the common class is the vtkPointSet class from the vtk package. You should try to replace by "vtk.vtkPointSet".

Superomeg4, 2021-02-10:
Hi everyone,

It seems that the last changes of pyvista have created again a bug: Trying to init a MeshVTK object with MeshMat.get_mesh_pv() will return the following error: CheckTypeError: For mesh : PointGrid expected, UnstructuredGrid given

closed 2021-03-22 by Superomeg4