Eomys/pyleecan #227: Can't launch pyleecan: Could not load the Qt platform plugin "windows"

jonjox, 2020-11-07:
Hello,

I have installed pyleecan in conda in two laptops. In the first one I have no problem, but in the second one, which has a better processor, when I start with the tutorial of the Webminar of October the 16th in the second "block" of the Jupyter Notebook, when it launchs pyleecan I obtain the next error in my terminal:

qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

The same if I launch pyleecan from terminal with python command.

I would thank any clue about why it is hapenning me in the 2ndd laptop.

Thanks in advance.

Juanjo Ugartemendia

EmileDvs, 2020-11-08:
Hello Juanjo,

If you use Windows, have you tried to setup the following path variable in your system path ? 
"QT_QPA_PLATFORM_PLUGIN_PATH : path\\to\\anaconda3\\Lib\\site-packages\\PySide2\\plugins\\platforms"

Best regards,
Emile

BonneelP, 2020-11-08:
Hello Juanjo,

Please note that we have changed pyleecan during the webinar. The plot command in the first notebook are not used any more and will raise error if you are using pyleecan 1.0.3. The more efficient command that we introduced are detailed in the second webinar. 

Best regards,
Pierre

jonjox, 2020-11-08:
Thank you!

That was the problem. I was trying with the first webminar's notebook, where I saw that the line where pyleecan was called was commented:
# %run -m pyleecan
But I removed the # symbol.
Thank you!

Juanjo

jonjox, 2020-11-08:
Hello again,

sorry, but if I try to laucn pyleecan from a jupyter notebook or from my terminal, the kernel is restarted and it reports the error that I told to you. I have added that variable to the system path, but, it didn't fix the error. I can't launch pyleecan.

Juanjo Ugartemendia

BonneelP, 2020-11-08:
Hello,

Which version of pyleecan are you using ? If it is not 1.0.3 please run "pip install -U pyleecan". 
The error occurs when running the line to open the GUI ? Are you able to plot the machine at the beginning of the notebook ?

Best regards,
Pierre

jonjox, 2020-11-09:
Hello,
yes I have 1.03 version installed. And yes the error occurs when try to run the GUI: %run -m pyleecan.
The machine is plotted at the beginning of the notebook.
To say the truth, I usually work with Linux and I am not used to work with Windows. Perhaps I didn't set up correctly the variable you told me to the system path, thus:
"QT_QPA_PLATFORM_PLUGIN_PATH : path\\to\\anaconda3\\Lib\\site-packages\\PySide2\\plugins\\platforms"
I added the path (path\\to\\anaconda3\\Lib\\site-packages\\PySide2\\plugins\\platforms), to the system path. I didn't add any variable (I mean: QT_QPA_PLATFORM_PLUGIN_PATH)

Another question. I suppose it is not possible to work with Linux, because FEMM doesn't work in Linux, isn't it?

Thanks in advance for your help!

Juanjo Ugartemendia

BonneelP, 2020-11-09:
Hello,

By waiting that the GUI issue is solved, you can start to use pyleecan in the notebook. The GUI only enables to define the "simple machine" in a more efficient way. 

Just to be sure, did you update "path\to\anaconda3" to your own path (on my computer it is C:\Anaconda3)? We need to make a proper documentation on this issue, I will soon update the page on pyleecan.org. 
I don't think that adding the path should work (but I may be wrong), you need to create the variable. This post seems to explain how: https://superuser.com/questions/949560/how-do-i-set-system-environment-variables-in-windows-10

Yes for now the only Magnetic model we have is FEMM so we are sadly windows only. If you are interested to help us on this topic, there is an issue to use FEMM on Linux #104 and one to add a new magnetic model with Elmer that should work on Linux #202. 

Best regards,
Pierre

jonjox, 2020-11-12:
Hello Pierre,

Fixed! The problem was the path variable. I followed the instructions of your link and now it works! Thank you very much, now I must train with the help of your tutorials.
Regarding your propostion, in this moment what I want is to use your software. I knew FEMM, and I think it is a good software, but, I have been looking Elmer in the web and it looks great also and it is open soucde, it would be wonderful to integrate it in pyleecan. This way it would be 100% open source and by the way it would be possible to have a Linux version. In any case, I will continue using your software, we'll be in touch.

Congratulations to the whole the team!

Juanjo

BonneelP, 2021-02-05:
Hello @jonjox,

I'm reviewing the opened issue and I wanted to tell you that since our last exchange there were several PR/discussions regarding Elmer #272 #291 #202 ;)
Can I close this issue or do you have further questions ? 

Best regards,
Pierre

jonjox, 2021-02-05:
Hello Pierre, yes, you can close it, thank you very much!

Regards,

Juanjo

Pierre Bonneel <notifications@github.com> igorleak hau idatzi zuen (2021
ots. 5, or. 22:37):

closed 2021-02-06 by BonneelP