Eomys/pyleecan #235: [CO] Introduicing star of slot method for winding / New organisation for the windings

BonneelP, 2020-11-13:
Hello all,

We want to introduce the star of slot to compute the winding matrix but by doing so the current winding classes would be obsolete. So here is my proposition for the new organization:

- Winding is no longer abstract and is the default class to use.
- Winding would have two new properties Nrad and Ntan to tune the number of layer
- comp_connection_mat method would be added to Winding and return the star of slot matrix according to Nrad and Ntan (and qs)
- get_dim_wind method would be added to Winding and return (Nrad, Ntan)
- All the daughters of Winding would be removed except WindingUD and WindingSC (I propose to keep WindingSC because I expect it to evolve one day). 
- WindingUD would have some new properties like init_as_DW1L, init_as_CW2LR... that would set the user_wind_mat according to the result of the comp_connection_mat method from the removed classes (that way we can still use the old matrices).

Note that one day we can also introduce WindingUD.load_from_koil that would work as mentioned in #56 

What do you think of this new organization proposal ?
Best regards,
Pierre

EmileDvs, 2021-04-27:
Hello everyone,

We are digging up this topic as we would like to clean the winding user interface in the next release. 

We found a very powerful python package on gitlab (https://gitlab.com/martinbaun/swat-em/) that provides a python APÏ to generate any kind of winding using Star of Slots method (same as in Koil in fact). It also provides tools to import and export winding configurations, and algorithms to handle end-windings.

First I am going to implement the modifications that Pierre detailed above a while ago using the interface with swat-em to generate any kind of winding based on number of slots/poles/phases, coil span, number of turns per coil etc. 

Please let me know if you have any remarks,
Best regards,
Emile

SebGue, 2021-04-28:
Hello Emile,

I'm fine, thank you. Unfortunatly I couldn't contribute due to lack of time in the last weeks.

This is a very nice tool you found.
Is there a dedicated branch for integration of it? When do you plan to implement this package (to pyleecan/ to the GUI)?

Regarding end winding, it seems swat-em 'only' calculate the end winding connections?! So my end winding length calculation should be a complementation to it. 

Best regards, 
Sebastian

EmileDvs, 2021-04-28:
Hello Sebastian,

Thanks for the precision, I must say I didn't look much into the end-windings feature and it's good to know that it should be complementary to what you already did.

We started the work on EOMYS-Public with a new PR (#378) so you can have a look at all the details if you like.

Best regards,
Emile

BonneelP, 2021-05-17:
Solved with #378

closed 2021-05-17 by BonneelP