Eomys/pyleecan #238: myResults = mySimu.run()  Error

hanafy4, 2020-11-16:
I have run one of your tutorial examples and got an error when running the simulation at myResults = mySimu.run() 

The Error  message is as following: 

TypeError: __init__() got an unexpected keyword argument 'symmetries' 

This error was within  
File "C:\ProgramData\..........\Machine\LamSlotWind\comp_mmf_unit.py", line 86, in comp_mmf_unit
    MMF = DataTime(

Could you, please, support us to fix this error to be able to continue investigating pyleecan.

Thank you in advance
Kind Regards

BonneelP, 2020-11-16:
Hello,

I think that your issue is linked to an outdated version of SciDataTool and/or pyleecan. Can you check your versions of both package ? To get the latest version of the packages you can use:
pip install -U SciDataTool
pip install -U pyleecan

Best regards,
Pierre

hanafy4, 2020-11-16:
Thank you dear Bonneel for your valuable reply. I have installed them and it skip that error and ended up with another error regarding the mesh, as following:

Exception: error: problem loading mesh

Could you, please direct us on how to fix that error too.

Kind Regards

BonneelP, 2020-11-16:
Hello again,

To make sure that everything is clean, I've released a new version of pyleecan with the latest development and I have updated the tutorials machine, FEMM, OP and Elec.
[Tutorials.zip](https://github.com/Eomys/pyleecan/files/5550184/Tutorials.zip)

All tutorials and tests are correct on my side with this new version. Please run again pip install -U pyleecan and hopefully it should work now.

Best regards,
Pierre

hanafy4, 2020-11-17:
Hi Bonneel, 

I have installed the updated version of Pyleecan as you mention. The version I have now is 1.0.4 and still I have the same error.

I have saved the .fem file to check the mesh loading problem, I tried to run the model after remeshing it, but it gives an error as shown in the following photo. 

[image]

I think there is an error, which I can not follow it as I am new to pyleecan. 

I am looking for your support in fixing this. 
Kind Regards

BonneelP, 2020-11-17:
Hello, 

Can you send me the script / notebook you are using so I can be sure to call the same methods as you do ? By the way which version of femm / pyfemm are you using ? 

Best regards,
Pierre

hanafy4, 2020-11-17:
Hello, 
I have pyfemm 0.1.1 installed. 
Please, find the attached python file.
[test_1.zip](https://github.com/Eomys/pyleecan/files/5552741/test_1.zip)

Thank you for your prompt reply.
Kind Regards

BonneelP, 2020-11-17:
Hello,

I've tried your script on my side and it worked :/

I'm using FEMM 4.2 21Apr2019 (c:/femm42/read.me), windows 10 and python 3.8.3 as part of Anaconda3.

It seems that the error happens when femm try to read a file. Maybe it is a right problem, you can set the path to save the fem file with simu.path_result or you can try to run your python with administration right. Normally pyleecan should automatically save the .fem file in the path_result folder. Can you check that the file is correctly created ? 

Best regards,
Pierre

hanafy4, 2020-11-17:
Thank you dear Bonneel,  Now it works, as I have reinstall the newer Femm version. 

FYI, I think there is a pug regarding the drawing, as I have got these message when I was trying to draw the contour of the flux density, it works only for the first time running the simulation and does not work after that, besides the drawing the flux density versus time and angle does not show any figures. 

Please, find the message I have got, as followes:

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Functions\Plot\plot_2D.py:235: UserWarning:

Matplotlib is currently using module://ipykernel.pylab.backend_inline, which is a non-GUI backend, so cannot show the figure.

Again, Thank you for your valuable support.

Kind Regards

BonneelP, 2020-11-17:
Great :) 
For this error, I think it is linked to the context of how you are calling the plot. In a notebook, try to add "%matplotlib notebook" at the beginning of the cell with the plot. 

Does it work when calling the method directly in a script ? Maybe you need to add "import matplotlib.pyplot as plt" and after the call to plot "plt.show()"

Best regards,
Pierre

hanafy4, 2020-11-17:
Yes, I have added that line to my script, but it gives the same warning.

I have attached the script to you below.

BonneelP, 2020-11-17:
I think the issue is maybe not linked to pyleecan. If you are running it from a script can you try to add the following at the beginning:

from matplotlib import use
use("Qt5Agg")  # Use PySide2 backend

In your test_3 files on which line do you get the error ? Does the plot machine display something ? 

Best regards,
Pierre

BonneelP, 2021-02-05:
Hello,

Is everything working correctly on your side ? Can we close this issue ? 

Best regards,
Pierre

BonneelP, 2021-03-15:
Hello,

I'm closing this issue, if the error happens again we can reopen it ;)

Best regards,
Pierre