Eomys/pyleecan #242: Simulation magnetics like inductors, transformers, coupled inductors

tinix84, 2020-11-17:
Hello,

I am opening this issue to start discussion around simulation of magnetics

What would be the best approach ?

Is there already any Python package that we could use to design it ?

How could we implement the actual code?

Best regards,
Riccardo

EmileDvs, 2020-11-18:
Hello Riccardo,

Nice to meet you, and thank you for starting this very important topic about inductor/transformer modelling in pyleecan.

For simulation aspect, I think we first should determine what are the inputs: voltage / current, pure sine/including harmonics, etc. and what outputs do we expect from the simulation: power, losses, ... The answer to this key question will guide us through the following steps. 

Then, another piece of work is to define inductors and transformers as pyleecan objects, which are luckily not very far from pyleecan machine objects as they also have a laminated core and windings inside/around. In fact, we can see inductors as a static motor that only have one lamination which includes one windings and transformers have one lamination and two windings (primary/secondary). 

We need to define templates of laminations and windings for inductors and transformers to actually draw them in any FE software. Maybe we can also adapt the dxf import tool to get lamination and winding objects from already existing geometries.

Finally, about how to actually perform the simulation, there is already the possibility to use the MagFEMM model (coupling with FEMM) to run 2D magnetostatics of electrical motors. The MagElmer model (coupling with GMSH/Elmer) is currently under development for 3D magnetodynamic. 

Depending on the required outputs, if we can stay in 2D at first, I think the best is to enable magnetoharmonic in MagFEMM (we already discussed about this there: #205 ). If we have to switch to 3D simulations we'll go to Elmer, or investigate GetDp coupling (I know there are already tutorials on this). Besides, there is always the possibility to implement analytical models as well.

Best regards,
Emile

tinix84, 2020-11-23:
Hi emile,
regarding the input can you use voltages? or just current in FEMM?

I think the obj must be different in case of inductor and transformer (at least from the external point of view because it needs different electrical interfaces and different equivalent circuit as output).

regarding the simulation we need AC simulation to get the AC resistance at different harmonics, this can be run in parallel?

the intersting point is the thermal model such that the design will be optimized for power density

EmileDvs, 2020-11-24:
Hello Riccardo,

I think FEMM only has currents as input, if you need input voltage you can setup an electric circuit in pyleecan and couple it with MagFEMM. This is what we did for example to calculate the dq inductances of the PMSM and deduce voltage function of current using an electric equivalent circuit: https://www.pyleecan.org/tuto_Elec.html

If you refer to FEMM parallelization (MagFEMM.solve_FEMM_parallel), this is only available on time steps, of course you can adapt it to your case by parallelizing on frequencies.

Concerning thermal model we have openned an issue lately on this topic, in which we would be very glad to have your insights if you have experience in such simulation: #215 . 

Best regards,
Emile